ZoneMinder/zoneminder #3264: Cannot export events

ASoTNetworks, 2021-05-28:
**Describe Your Environment**
- Version of ZoneMinder: `v1.36.2`
- How you installed: Deb package via do_debian_package.sh
- Full name and version of OS: Debian Buster
- Browser name and version (if this is an issue with the web interface): Firefox 88.0.1 and Chromium 90.0.4430.212

**Describe the bug**
Cannot export selected events from web interface stuck at `Exporting`.

**Describe the bug**
After issue #3250 is fixed I see the `Exporting` status at the bottom of the page and see that tar or zip is running on the system when they are selected.

After I see the process end I do not get prompted to download the file. This time there is nothing in the log about the issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Go select some events you want to export.
2. Click on `export` button
3. Select `Tar` or `Zip` for ` Export File Format`
4. Click Export

**Expected behavior**
Should prompt you to download the file after it is archived.

**Debug Logs**
none

j-marz, 2021-05-30:
I'm experiencing the same issue on v1.37.0

ASoTNetworks, 2021-06-20:
I selected 7 events and still have the issue.
In the apache error.log I see:

```
[Sun Jun 20 13:25:10.329038 2021] [php7:warn] [pid 31567] [client 2607:f2c0:f00f:xxxx:xxxx:xxxx:xxxx:xxxx:35024] PHP Warning:  mkdir(): File exists in /usr/share/zoneminder/www/skins/classic/includes/export_functions.php on line 889, referer: https://HOSTNAME/zm/index.php?view=export&eids[]=636669&eids[]=636671&eids[]=636674&eids[]=636678&eids[]=636681&eids[]=636684&eids[]=636689
tar: You must specify one of the '-Acdtrux', '--delete' or '--test-label' options
Try 'tar --help' or 'tar --usage' for more information.
```

connortechnology, 2021-06-22:
There is not enough information here to figure anything out.  Would it be possible to get access to this system?

ASoTNetworks, 2021-07-03:
The system is behind a firewall but I have test on another system but it have the same issue.

When the export job is running am I suppose to see files in `/var/tmp/zm`?
There are no files inside there.

Are there some things you want me to look at?

ASoTNetworks, 2021-07-22:
After looking at it more looks like the archive is being created in `/var/tmp/systemd-private-87e265b7f888fb36975401dc90f5afab-apache2.service-LTACze/tmp/zm` but after creation it does not prompt me to download.

connortechnology, 2021-09-13:
Please update to 1.36.7 and let us know if the problem is fixed..

ASoTNetworks, 2021-09-13:
Still same issue with 1.36.7

tommysnello, 2021-09-23:
1.36.7 same problem for me

tdragonite, 2022-04-29:
Hi. Same issue here with 1.36.14. Like I've writed in the ZoneMinder forum (https://forums.zoneminder.com/viewtopic.php?t=31700), I can see in the Chrome developer tools a failed request (invalid response, no data found) to the url  "https://myhost/zm/index.php?view=request&request=event&action=download&__csrf_magic=key%3A56e6ab335958d8bf684b1548211ddbd98369d9ff%2C1650914870&connkey=971011&exportVideo=1&eids%5B%5D=376807&eids%5B%5D=376703&eids%5B%5D=376776&eids%5B%5D=376790&eids%5B%5D=376717&eids%5B%5D=376742&exportFormat=zip"

commit 0e92eace (2022-05-03): Remove timeout on ajax call to generate the zip. Fixes #3264
commit d2ef6b99 (2022-05-03): Remove timeout on ajax call to generate the zip. Fixes #3264
closed 2022-05-03 by connortechnology