ZoneMinder/zoneminder #3269: Log spam in zmc with high FPS report interval

henryptung, 2021-05-31:
https://github.com/ZoneMinder/zoneminder/commit/046e352304a84059d02014b2d3d16cf850eaadce#diff-6bbad3965ada94cfabbe05ea66d0c4a7a8ad39a38f636ece89bef5743bc76540R1680

This commit adds a pretty weird reporting condition to the fps_report_interval check - if the frame count is less than the interval, it reports every 10 frames instead.

I have a 10FPS camera with the interval configured to every hour (interval 36000), so this ends up spamming the log 3600 times before it quiets back down to the configured cadence.

@connortechnology, not sure if this is intended? I'm not sure what the intent of the per-10-frames report is, but if more frequent updates pre-first-interval are needed for some reason, maybe it could do exponential backoff instead.

Example logs:

```
May 30 23:40:24 mediaserver zmc_m1[1463]: INF [zmc_m1] [Starting Capture version 1.36.2]
May 30 23:40:27 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 10 - Capturing at 2.86 fps, capturing bandwidth 10890bytes/sec]
May 30 23:40:29 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 40 - Capturing at 25.20 fps, capturing bandwidth 54006bytes/sec]
May 30 23:40:30 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 50 - Capturing at 9.55 fps, capturing bandwidth 36332bytes/sec]
May 30 23:40:32 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 70 - Capturing at 9.96 fps, capturing bandwidth 25405bytes/sec]
May 30 23:40:34 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 90 - Capturing at 10.18 fps, capturing bandwidth 25776bytes/sec]
May 30 23:40:35 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 100 - Capturing at 9.93 fps, capturing bandwidth 12826bytes/sec]
May 30 23:40:37 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 120 - Capturing at 9.92 fps, capturing bandwidth 25341bytes/sec]
May 30 23:40:39 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 140 - Capturing at 10.08 fps, capturing bandwidth 25730bytes/sec]
May 30 23:40:40 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 150 - Capturing at 9.89 fps, capturing bandwidth 37579bytes/sec]
May 30 23:40:41 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 160 - Capturing at 9.91 fps, capturing bandwidth 12698bytes/sec]
May 30 23:40:43 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 180 - Capturing at 10.00 fps, capturing bandwidth 25686bytes/sec]
May 30 23:40:45 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 200 - Capturing at 10.00 fps, capturing bandwidth 25400bytes/sec]
May 30 23:40:47 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 220 - Capturing at 9.95 fps, capturing bandwidth 25411bytes/sec]
May 30 23:40:48 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 230 - Capturing at 9.87 fps, capturing bandwidth 37756bytes/sec]
May 30 23:40:50 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 250 - Capturing at 9.95 fps, capturing bandwidth 25460bytes/sec]
May 30 23:40:52 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 270 - Capturing at 9.93 fps, capturing bandwidth 25271bytes/sec]
May 30 23:40:54 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 290 - Capturing at 10.02 fps, capturing bandwidth 25839bytes/sec]
May 30 23:40:56 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 310 - Capturing at 10.02 fps, capturing bandwidth 25665bytes/sec]
May 30 23:40:57 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 320 - Capturing at 10.00 fps, capturing bandwidth 12979bytes/sec]
May 30 23:40:58 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 330 - Capturing at 9.95 fps, capturing bandwidth 37549bytes/sec]
May 30 23:41:00 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 350 - Capturing at 10.02 fps, capturing bandwidth 25520bytes/sec]
May 30 23:41:02 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 370 - Capturing at 10.01 fps, capturing bandwidth 25423bytes/sec]
May 30 23:41:04 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 390 - Capturing at 10.00 fps, capturing bandwidth 25476bytes/sec]
May 30 23:41:05 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 400 - Capturing at 10.00 fps, capturing bandwidth 12726bytes/sec]
May 30 23:41:06 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 410 - Capturing at 9.75 fps, capturing bandwidth 36878bytes/sec]
May 30 23:41:08 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 430 - Capturing at 10.13 fps, capturing bandwidth 26139bytes/sec]
May 30 23:41:09 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 440 - Capturing at 10.00 fps, capturing bandwidth 12981bytes/sec]
May 30 23:41:10 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 450 - Capturing at 9.96 fps, capturing bandwidth 37622bytes/sec]
May 30 23:41:12 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 470 - Capturing at 10.02 fps, capturing bandwidth 25522bytes/sec]
May 30 23:41:13 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 480 - Capturing at 10.00 fps, capturing bandwidth 12908bytes/sec]
May 30 23:41:14 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 490 - Capturing at 10.00 fps, capturing bandwidth 37806bytes/sec]
May 30 23:41:15 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 500 - Capturing at 9.98 fps, capturing bandwidth 13067bytes/sec]
May 30 23:41:16 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 510 - Capturing at 9.97 fps, capturing bandwidth 37675bytes/sec]
May 30 23:41:18 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 530 - Capturing at 9.75 fps, capturing bandwidth 25066bytes/sec]
May 30 23:41:20 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 550 - Capturing at 10.04 fps, capturing bandwidth 26135bytes/sec]
May 30 23:41:22 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 570 - Capturing at 10.00 fps, capturing bandwidth 25528bytes/sec]
May 30 23:41:23 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 580 - Capturing at 9.99 fps, capturing bandwidth 12922bytes/sec]
May 30 23:41:25 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 600 - Capturing at 10.00 fps, capturing bandwidth 25726bytes/sec]
May 30 23:41:26 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 610 - Capturing at 9.99 fps, capturing bandwidth 37929bytes/sec]
May 30 23:41:27 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 620 - Capturing at 9.98 fps, capturing bandwidth 13259bytes/sec]
May 30 23:41:28 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 630 - Capturing at 9.93 fps, capturing bandwidth 37764bytes/sec]
May 30 23:41:30 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 650 - Capturing at 10.04 fps, capturing bandwidth 25832bytes/sec]
May 30 23:41:31 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 660 - Capturing at 9.96 fps, capturing bandwidth 12933bytes/sec]
May 30 23:41:33 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 680 - Capturing at 9.96 fps, capturing bandwidth 25429bytes/sec]
May 30 23:41:35 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 700 - Capturing at 10.01 fps, capturing bandwidth 26085bytes/sec]
May 30 23:41:37 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 720 - Capturing at 10.05 fps, capturing bandwidth 25474bytes/sec]
May 30 23:41:38 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 730 - Capturing at 9.98 fps, capturing bandwidth 37673bytes/sec]
May 30 23:41:40 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 750 - Capturing at 10.01 fps, capturing bandwidth 25653bytes/sec]
May 30 23:41:42 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 770 - Capturing at 9.89 fps, capturing bandwidth 24999bytes/sec]
May 30 23:41:44 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 790 - Capturing at 10.11 fps, capturing bandwidth 25693bytes/sec]
May 30 23:41:45 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 800 - Capturing at 9.96 fps, capturing bandwidth 12779bytes/sec]
May 30 23:41:46 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 810 - Capturing at 9.54 fps, capturing bandwidth 36397bytes/sec]
May 30 23:41:48 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 830 - Capturing at 10.01 fps, capturing bandwidth 25491bytes/sec]
May 30 23:41:50 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 850 - Capturing at 9.99 fps, capturing bandwidth 25436bytes/sec]
May 30 23:41:52 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 870 - Capturing at 10.01 fps, capturing bandwidth 25583bytes/sec]
May 30 23:41:53 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 880 - Capturing at 10.00 fps, capturing bandwidth 12825bytes/sec]
May 30 23:41:54 mediaserver zmc_m1[1463]: INF [zmc_m1] [camera-1: 890 - Capturing at 10.00 fps, capturing bandwidth 38029bytes/sec]
```

connortechnology, 2021-05-31:
The intention was to get an initial quick reporting after starting up.  Clearly it is junk.
The test should probably just be image_count == 10.

connortechnology, 2021-05-31:
This code updates the entry in the db that gets used to display the fps on the web ui console.  When a camera starts up the inital fps values can vary a bit.  So for the purposes of good UI we update a little quicker.
The idea is that fps interval should be a useful value.  Having a 1 hour report time is almost entirely useless.  Any changes in value will be missed. Perhaps it should report on significant change of value.

Still that's your choice.  I'll have to think on what this code should be.

closed 2026-11-15 by connortechnology