ZoneMinder/zoneminder #3271: Capture daemon being restarted by zmwatch

ASoTNetworks, 2021-06-01:
**Describe Your Environment**
- Version of ZoneMinder: `v1.36.3`
- How you installed: Deb package via do_debian_package.sh
- Full name and version of OS: Debian Buster

**Describe the bug**
Ever since updating to v1.36.0 and above I see that `zmwatch` is restarting capture daemon randomly.

**To Reproduce**
Steps to reproduce the behavior:
1. Have monitor mode in any modes such as mocord, monitor, modect and so on.
2. You will message in the log like below.

**Expected behavior**
Capture daemon should not stop.

**Debug Logs**
```
2021-06-01 01:04:57	zmc_m18		8299	INF	Starting Capture version 1.36.3	zmc.cpp	220
2021-06-01 01:04:57	zmdc		8299	INF	'zmc -m 18' started at 21/06/01 01:04:57	zmdc.pl	-
2021-06-01 01:04:57	zmdc		26197	INF	'zmc -m 18' starting at 21/06/01 01:04:57, pid = 8299	zmdc.pl	-
2021-06-01 01:04:57	zmdc		26197	INF	Starting pending process, zmc -m 18	zmdc.pl	-
2021-06-01 01:04:56	zmdc		26197	INF	'zmc -m 18' exited normally	zmdc.pl	-
2021-06-01 01:04:56	zmc_m18		6868	INF	Camera01: image_count:51499 - Closing event 622849, shutting down	zm_monitor.cpp	3107
2021-06-01 01:04:55	zmdc		26197	INF	'zmc -m 18' sending stop to pid 6868 at 21/06/01 01:04:55	zmdc.pl	-
2021-06-01 01:04:55	zmwatch		26275	WAR	Restarting capture daemon for Camera01, time since last capture 7 seconds (1622523895-1622523888)-
```

connortechnology, 2021-06-08:
The problem is that capture has halted for 7 seconds. Why?  You can increase this limit by changing Options->System->WATCH_MAX_DELAY
I find that a value of 45 is more practical.

connortechnology, 2021-06-09:
There are a lot of potential reasons.  We are going to need debug level 3 logs to figure it out.
Our design has been to prioritize capturing, removing possibility of blocking or waiting but there are still places where it can hang. For example: writing to database.

ASoTNetworks, 2021-06-10:
Here is the log with level 3 set

```
Jun 10 02:17:25 cctv zmc_m18[18862]: INF [zmc_m18] [Camera01: 51000 - Capturing at 15.00 fps, capturing bandwidth 360276bytes/sec]
Jun 10 02:17:25 cctv zmc_m18[18862]: INF [zmc_m18] [Camera01: 51000 - Analysing at 15.00 fps from 50999 - 49999=1000 / 1623305845.526966 - 1623305778.867344 = 66.659622]
Jun 10 02:18:54 cctv zmwatch[16016]: WAR [Restarting capture daemon for Camera01, time since last capture 51 seconds (1623305934-1623305883)]
Jun 10 02:18:54 cctv zmdc[15935]: INF ['zmc -m 18' sending stop to pid 18862 at 21/06/10 02:18:54]
Jun 10 02:18:54 cctv zmc_m18[18862]: INF [zmc_m18] [Camera01: image_count:51568 - Closing event 629879, shutting down]
Jun 10 02:18:54 cctv zmc_m18[18862]: INF [zmc_m18] [Estimating the duration of the last packet in a fragment, consider setting the duration field in AVPacket instead.]
Jun 10 02:18:54 cctv zmdc[15935]: INF ['zmc -m 18' exited normally]
Jun 10 02:18:55 cctv zmdc[15935]: INF [Starting pending process, zmc -m 18]
Jun 10 02:18:56 cctv zmdc[15935]: INF ['zmc -m 18' starting at 21/06/10 02:18:56, pid = 21416]
Jun 10 02:18:56 cctv zmdc[21416]: INF ['zmc -m 18' started at 21/06/10 02:18:56]
Jun 10 02:18:56 cctv zmc_m18[21416]: INF [zmc_m18] [Enabling ffmpeg logs, as LOG_DEBUG+LOG_FFMPEG are enabled in options]
Jun 10 02:18:56 cctv zmc_m18[21416]: INF [zmc_m18] [Starting Capture version 1.36.4]
Jun 10 02:18:59 cctv zmc_m18[21416]: INF [zmc_m18] [deprecated pixel format used, make sure you did set range correctly]
Jun 10 02:18:59 cctv zmc_m18[21416]: INF [zmc_m18] [Camera01: 10 - Capturing at 3.69 fps, capturing bandwidth 97080bytes/sec]
Jun 10 02:18:59 cctv zmc_m18[21416]: INF [zmc_m18] [Camera01: 000 - Opened new event 629880, section start]
Jun 10 02:18:59 cctv zmc_m18[21416]: INF [zmc_m18] [Camera01: 10 - Analysing at 2.95 fps from 9 - 0=9 / 1623305939.551569 - 1623305936.497256 = 3.054313]
Jun 10 02:19:00 cctv zmc_m18[21416]: INF [zmc_m18] [Camera01: 50 - Capturing at 33.33 fps, capturing bandwidth 741127bytes/sec]
Jun 10 02:19:00 cctv zmc_m18[21416]: INF [zmc_m18] [Camera01: 50 - Analysing at 38.40 fps from 49 - 9=40 / 1623305940.593123 - 1623305939.551569 = 1.041554]
Jun 10 02:19:01 cctv zmc_m18[21416]: INF [zmc_m18] [Camera01: 70 - Capturing at 14.93 fps, capturing bandwidth 373556bytes/sec]
Jun 10 02:19:01 cctv zmc_m18[21416]: INF [zmc_m18] [Camera01: 70 - Analysing at 14.71 fps from 69 - 49=20 / 1623305941.952865 - 1623305940.593123 = 1.359742]
Jun 10 02:19:03 cctv zmc_m18[21416]: INF [zmc_m18] [Camera01: 90 - Capturing at 15.15 fps, capturing bandwidth 300671bytes/sec]
Jun 10 02:19:03 cctv zmc_m18[21416]: INF [zmc_m18] [Camera01: 90 - Analysing at 15.35 fps from 89 - 69=20 / 1623305943.255503 - 1623305941.952865 = 1.302638]
Jun 10 02:19:04 cctv zmc_m18[21416]: INF [zmc_m18] [Camera01: 110 - Capturing at 14.92 fps, capturing bandwidth 371428bytes/sec]
Jun 10 02:19:04 cctv zmc_m18[21416]: INF [zmc_m18] [Camera01: 110 - Analysing at 14.94 fps from 109 - 89=20 / 1623305944.594352 - 1623305943.255503 = 1.338849]
```

connortechnology, 2021-06-10:
You have NOT set debug level to 3.  If you had, there would be thousands of lines of debug logs in there.

ASoTNetworks, 2021-06-12:
Weird I have turned on the options in logging.
[image]

SteveGilvarry, 2021-06-12:
Also need a target check the help ? On that option but I think you need _zmc_m18 for you monitor 18 that seems to be having the issue. Will generate a separate file in logs/zm.

ASoTNetworks, 2021-06-15:
Thanks, I got the debug log going now and this is what I see before and after where zmwatch restart the hanged stream.
Before it happen I see an event lock message at `14:54:59.159466`

[Debug log here](https://paste.aseriesoftubez.com/p/1623784535-NO5FAQIkPWS.txt)

connortechnology, 2021-06-15:
please update to 1.36.4

ASoTNetworks, 2021-06-15:
Sorry I didn't update the version above but I am already on 1.36.4

connortechnology, 2021-06-15:
So there is a slight problem with the locking, but what is happening to the capture thread?  It appears to never capture another frame, which means it can't signal the decode thread to wake up and notice that it needs to decode a packet.  The capture thread specifically does not lock for this very reason.  Do you maybe have a value in your maxFps or AlarmMaxFps fields?

connortechnology, 2021-06-15:
The only other thing I can think of is contention on the Monitor_Status table.  mysql queries timeout after like 200 seconds.. so that could be it.

ASoTNetworks, 2021-06-15:
I have maxFps set to 40.000 and no AlarmMaxFps set. I think I have it set a long time ago on versions that was saving jpegs. Should I remove it?

connortechnology, 2021-06-15:
Definitely remove it. I'm not convinced it's the culprit, but it will starve ffmpeg for cpu.

ASoTNetworks, 2021-06-20:
I see an improvement and doesn't happen too many times now (was a few times every few hours) after removing the maxFps it only happens a couple times a day.

mgeisselreiter, 2021-07-01:
I can comfirm this problem. After upgrading from 1.36.3 to 1.36.4 zmwatch log shows restarting capture daemon every 40 to 60 seconds. Now I cannot use Live watch - I see video just 2-3 seconds and than picture stops, after a while there is black screen and it repeats every minute. The same problem is with recording - there are no records, only 2-3 seconds videos every minute.
Now I upgraded to 1.36.5 but the problem remains. There must be some bad change between versions 1.36.3 and later. It is frustrating because I need to watch daily records every day which are now missing. I use CentOS 7 and there is no simple way to roll back to version 1.36.3. Any help will be appreciate.

mgeisselreiter, 2021-07-01:
After a long playing with changing configuration parameters I manage to find some solution to solve this issue. I changed storage parameters for my camera for Video Writer from Camera Passthrough to Encode, OutputCodec to Auto, Encoder Auto, Optional Encoder Parameters crf=23 and Montage - Live video is fluent without breaks. Tomorrow I will see how recording works. Still wonder why Camera Passthrough stopped working since version 1.36.4. Any ideas?

joe7rocks, 2021-08-14:
I'm having the very same issue with various camera models after the update, changing to Encode from Passthrough has also 'fixed' it temporarily

mgeisselreiter, 2021-10-17:
After upgrade from 1.36.5 to 1.36.8 the problem with Camera Passthrough was not solved. With this setting zm crashes and restarts every minute as I mentioned on July 1. I have to use Encode and the x264 codec which is more CPU consumpting.

mgeisselreiter, 2021-10-17:
Yes. Options - Logging - Debug - Level 3
[log.zip](https://github.com/ZoneMinder/zoneminder/files/7359545/log.zip)
If you need more logs or higher level please let me know or insctruct me.

mgeisselreiter, 2021-10-17:
Hi Steve, 
thanks for your help. I have configured 2 cameras but using and testing only one with IP 10.1.0.36. So you are right that the camera with IP 10.1.0.17 is not connected. That is probably confusing. So debug logs should be OK. Do you need higher level debug logs?

connortechnology, 2021-10-17:
I would suggest trying master.  I am trying something there that fixes this problem. 

Or: you need to set MaxImageBUffer count > 2*keyframe interval + 1.  What is happening is that the queue cleanup process hit a locked packet before it could reach the next keyframe so it couldn't clear them out.  Your keyframe interval appears to be 50.  So try increasing MaxImage buffer to 100 or 101.

closed 2022-04-17 by connortechnology