ZoneMinder/zoneminder #3272: PTZ failure

alabamatoy, 2021-06-01:
**Describe Your Environment**
- Version of ZoneMinder: 1.36.0 and 1.36.3, happens on both
- How you installed ZoneMinder [e.g. PPA, RPMFusion, from-source, etc]: PPA
- Full name and version of OS1.36.0 is on Ubuntu 16.04 and 1.36.3 is on 18.04
- Browser name and version (if this is an issue with the web interface): Believe this is N/A

**If the issue concerns a camera**
- Amcrest, any PTZ version
- frame rate
- resolution
- ZoneMinder Source Type: RTSP

**Describe the bug**
PTZ stops working until ZM is restarted.  It will randomly cease PTZ function.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to monitor view
2. Click on any PTZ function, it may work, it may not.

**Expected behavior**
PTZ should be reliable over time without restarting ZM so frequently

**Debug Logs**

Here is a log from a correctly functioning 1.36.0 PTZ action:
```
06/01/2021 13:14:34.246949 zmcontrol_17[28583].DB1 [ZoneMinder::Logger:321] [LogOpts: level=DB5/DB5, screen=OFF, database=WAR, logfile=DB5->/var/log/zm/zm_debug17.log, syslog=WAR]
06/01/2021 13:14:34.247067 zmcontrol_17[28583].DB1 [main:73] [zmcontrol: arg string: --id 17 sock file /var/run/zm/zmcontrol-17.sock]
06/01/2021 13:14:34.248032 zmcontrol_17[28583].INF [main:134] [Starting control server 17/Amcrest_HTTP]
06/01/2021 13:14:34.274343 zmcontrol_17[28583].INF [main:149] [Control server 17/Amcrest_HTTP starting at 21/06/01 13:14:34]
06/01/2021 13:14:34.278447 zmcontrol_17[28583].DB1 [ZoneMinder::Control::Amcrest_HTTP:74] [Using initial credentials for 10.1.1.173:80, Login to AMC044326A3F886AA6, admin, admin1234, base_url: http://login:password@10.1.1.173/ auth:login:password@10.1.1.173:80]
06/01/2021 13:15:06.786373 zmcontrol_17[28583].DB1 [main:188] [$VAR1 = {
          'autostop' => 1,
          'command' => 'moveConDownLeft'
        };]
06/01/2021 13:15:06.786443 zmcontrol_17[28583].DB1 [ZoneMinder::Control::Amcrest_HTTP:245] [Move Diagonally Down Left]
06/01/2021 13:15:06.786628 zmcontrol_17[28583].DB3 [ZoneMinder::Memory:309] [Shared data appears valid for monitor 17: 1]
06/01/2021 13:15:06.786661 zmcontrol_17[28583].DB1 [ZoneMinder::Monitor:283] [Suspending motion detection]
06/01/2021 13:15:06.786698 zmcontrol_17[28583].DB3 [ZoneMinder::Memory::Mapped:84] [zmMemAttach already attached at 140335048302592 for 17]
06/01/2021 13:15:06.786734 zmcontrol_17[28583].DB3 [ZoneMinder::Memory:309] [Shared data appears valid for monitor 17: 1]
06/01/2021 13:15:06.886950 zmcontrol_17[28583].DB1 [ZoneMinder::Monitor:287] [0]
06/01/2021 13:15:06.887054 zmcontrol_17[28583].DB1 [ZoneMinder::Control:144] [cgi-bin/ptz.cgi?action=start&code=LeftDown&channel=0&arg1=0&arg2=1&arg3=0[73]]
06/01/2021 13:15:07.129053 zmcontrol_17[28583].INF [ZoneMinder::Control::Amcrest_HTTP:155] [Camera control 2: '200 OK' for URL http://login:password@10.1.1.173/cgi-bin/ptz.cgi?action=start&code=LeftDown&channel=0&arg1=0&arg2=1&arg3=0]
06/01/2021 13:15:09.758783 zmcontrol_17[28583].DB1 [main:188] [$VAR1 = {
          'command' => 'moveStop'
        };]
06/01/2021 13:15:09.758870 zmcontrol_17[28583].DB1 [ZoneMinder::Control::Amcrest_HTTP:258] [Move Stop code=LeftDown&channel=0&arg1=0&arg2=1&arg3=0]
06/01/2021 13:15:09.758919 zmcontrol_17[28583].DB1 [ZoneMinder::Control:144] [cgi-bin/ptz.cgi?action=stop&code=LeftDown&channel=0&arg1=0&arg2=1&arg3=0[72]]
06/01/2021 13:15:09.846932 zmcontrol_17[28583].INF [ZoneMinder::Control::Amcrest_HTTP:155] [Camera control 2: '200 OK' for URL http://login:password@10.1.1.173/cgi-bin/ptz.cgi?action=stop&code=LeftDown&channel=0&arg1=0&arg2=1&arg3=0]
06/01/2021 13:15:09.847159 zmcontrol_17[28583].DB3 [ZoneMinder::Memory::Mapped:84] [zmMemAttach already attached at 140335048302592 for 17]
06/01/2021 13:15:09.847391 zmcontrol_17[28583].DB3 [ZoneMinder::Memory:309] [Shared data appears valid for monitor 17: 1]
06/01/2021 13:15:09.847477 zmcontrol_17[28583].DB1 [ZoneMinder::Monitor:294] [Resuming motion detection]
06/01/2021 13:15:09.847617 zmcontrol_17[28583].DB3 [ZoneMinder::Memory::Mapped:84] [zmMemAttach already attached at 140335048302592 for 17]
06/01/2021 13:15:09.847785 zmcontrol_17[28583].DB3 [ZoneMinder::Memory:309] [Shared data appears valid for monitor 17: 1]
06/01/2021 13:15:39.276660 zmcontrol_17[28583].INF [main:143] [Received TERM, exiting]
06/01/2021 13:15:39.276767 zmcontrol_17[28583].DB1 [main:204] [Can't select: Interrupted system call]
06/01/2021 13:15:39.276791 zmcontrol_17[28583].INF [main:214] [Control server 17/Amcrest_HTTP exiting]

```
and here is where it fails:

```
06/01/2021 12:52:56.189461 zmcontrol_20[26037].DB1 [ZoneMinder::Logger:321] [LogOpts: level=DB5/DB5, screen=OFF, database=WAR, logfile=DB5->/var/log/zm/zm_debug.log, syslog=WAR]
06/01/2021 12:52:56.189568 zmcontrol_20[26037].DB1 [main:73] [zmcontrol: arg string: --id 20 sock file /var/run/zm/zmcontrol-20.sock]
06/01/2021 12:52:56.190385 zmcontrol_20[26037].INF [main:134] [Starting control server 20/Amcrest_HTTP]
06/01/2021 12:52:56.206995 zmcontrol_20[26037].INF [main:149] [Control server 20/Amcrest_HTTP starting at 21/06/01 12:52:56]
06/01/2021 12:52:56.211307 zmcontrol_20[26037].DB1 [ZoneMinder::Control::Amcrest_HTTP:74] [Using initial credentials for 10.1.1.158:80, Login to AMC04424E4D49DAF44, admin, admin1234, base_url: http://login:password@10.1.1.158/ auth:login:password@10.1.1.158:80]
06/01/2021 12:53:33.469985 zmcontrol_20[26037].DB1 [main:188] [$VAR1 = {
          'command' => 'moveConUpRight',
          'autostop' => 1
        };]
06/01/2021 12:53:33.470048 zmcontrol_20[26037].DB1 [ZoneMinder::Control::Amcrest_HTTP:221] [Move Diagonally Up Right]
06/01/2021 12:53:33.470253 zmcontrol_20[26037].DB3 [ZoneMinder::Memory:309] [Shared data appears valid for monitor 20: 1]
06/01/2021 12:53:33.470291 zmcontrol_20[26037].DB1 [ZoneMinder::Monitor:283] [Suspending motion detection]
06/01/2021 12:53:33.470334 zmcontrol_20[26037].DB3 [ZoneMinder::Memory::Mapped:84] [zmMemAttach already attached at 140396391923712 for 20]
06/01/2021 12:53:33.470376 zmcontrol_20[26037].DB3 [ZoneMinder::Memory:309] [Shared data appears valid for monitor 20: 1]
06/01/2021 12:53:33.570631 zmcontrol_20[26037].DB1 [ZoneMinder::Monitor:283] [Suspending motion detection]
06/01/2021 12:53:33.570730 zmcontrol_20[26037].DB3 [ZoneMinder::Memory::Mapped:84] [zmMemAttach already attached at 140396391923712 for 20]
06/01/2021 12:53:33.570783 zmcontrol_20[26037].DB3 [ZoneMinder::Memory:309] [Shared data appears valid for monitor 20: 1]
06/01/2021 12:53:33.670972 zmcontrol_20[26037].DB1 [ZoneMinder::Monitor:283] [Suspending motion detection]
06/01/2021 12:53:33.671128 zmcontrol_20[26037].DB3 [ZoneMinder::Memory::Mapped:84] [zmMemAttach already attached at 140396391923712 for 20]
06/01/2021 12:53:33.671232 zmcontrol_20[26037].DB3 [ZoneMinder::Memory:309] [Shared data appears valid for monitor 20: 1]
06/01/2021 12:53:33.771434 zmcontrol_20[26037].DB1 [ZoneMinder::Monitor:283] [Suspending motion detection]
06/01/2021 12:53:33.771564 zmcontrol_20[26037].DB3 [ZoneMinder::Memory::Mapped:84] [zmMemAttach already attached at 140396391923712 for 20]
06/01/2021 12:53:33.771647 zmcontrol_20[26037].DB3 [ZoneMinder::Memory:309] [Shared data appears valid for monitor 20: 1]
06/01/2021 12:53:33.872024 zmcontrol_20[26037].DB1 [ZoneMinder::Monitor:283] [Suspending motion detection]
06/01/2021 12:53:33.872111 zmcontrol_20[26037].DB3 [ZoneMinder::Memory::Mapped:84] [zmMemAttach already attached at 140396391923712 for 20]
06/01/2021 12:53:33.872159 zmcontrol_20[26037].DB3 [ZoneMinder::Memory:309] [Shared data appears valid for monitor 20: 1]
06/01/2021 12:53:33.972323 zmcontrol_20[26037].DB1 [ZoneMinder::Monitor:283] [Suspending motion detection]
06/01/2021 12:53:33.972436 zmcontrol_20[26037].DB3 [ZoneMinder::Memory::Mapped:84] [zmMemAttach already attached at 140396391923712 for 20]
06/01/2021 12:53:33.972513 zmcontrol_20[26037].DB3 [ZoneMinder::Memory:309] [Shared data appears valid for monitor 20: 1]
06/01/2021 12:53:34.072756 zmcontrol_20[26037].DB1 [ZoneMinder::Monitor:283] [Suspending motion detection]
06/01/2021 12:53:34.072848 zmcontrol_20[26037].DB3 [ZoneMinder::Memory::Mapped:84] [zmMemAttach already attached at 140396391923712 for 20]
06/01/2021 12:53:34.072893 zmcontrol_20[26037].DB3 [ZoneMinder::Memory:309] [Shared data appears valid for monitor 20: 1]
06/01/2021 12:53:34.173063 zmcontrol_20[26037].DB1 [ZoneMinder::Monitor:283] [Suspending motion detection]
06/01/2021 12:53:34.173180 zmcontrol_20[26037].DB3 [ZoneMinder::Memory::Mapped:84] [zmMemAttach already attached at 140396391923712 for 20]
06/01/2021 12:53:34.173245 zmcontrol_20[26037].DB3 [ZoneMinder::Memory:309] [Shared data appears valid for monitor 20: 1]
06/01/2021 12:53:34.273499 zmcontrol_20[26037].DB1 [ZoneMinder::Monitor:283] [Suspending motion detection]
```
This will repeat forever until either the process crashes, or ZM is restarted.

connortechnology, 2021-06-14:
I have added some code to limit the infinite loop already.  Will merge to 1.36.5 soon.  My suspicion is that zmc has crashed and restarted and zmcontrol is still reading the old mmap file.So I have added some code to deal with this.

Please update to master and let's see if I'm right.

alabamatoy, 2021-06-15:
I installed 1.37 (master) and have been unable to reproduce the PTZ failure.  I will let it run for a few days and see for sure.

I killed the appropriate zmc process, and PTZ seemed to work fine regardless.  Is that sufficient verification?

Thanks for your efforts!

alabamatoy, 2021-07-02:
This issue has reappeared.  Its far less common, but its back.  I will attempt to collect some detailed debug logs....

connortechnology, 2021-09-13:
ping. I believe we have resolved this, yes?

alabamatoy, 2021-09-13:
Yes, sorry, thanks for following up.

Herndon Elliott
"If the truth shall kill them, let them die."  -- Ayn Rand
ΜΟΛΩΝ ΛΑΒΕ  "molon labe"

closed 2021-09-14 by Carbenium