ZoneMinder/zoneminder #3274: Montage Review: No video to playback

DanwithArduino, 2021-06-02:
**Describe Your Environment**
- Version of ZoneMinder [1.37]
- How you installed ZoneMinder [PPA - master]
- Full name and version of OS: Ubuntu 20.04
- Firefox 88.0.1 and Chrome 91.0.4472.77 (Official Build) (64-bit)

**Describe the bug**
Opening the Montage Review page yields no video to review. The camera box is empty and displays "evaluating fps" 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'montage review'

**Debug Logs**
```
Opening the console in chrome yields: 

`/zm/?view=montagereview:1 GET http://192.168.1.118/zm/?view=montagereview 500 (Internal Server Error)`

`DevTools failed to load source map: Could not load content for http://192.168.1.118/zm/cache/bootstrap.bundle.js.map: HTTP error: status code 404, net::ERR_HTTP_RESPONSE_CODE_FAILURE`

`?view=montagereview:255 Uncaught SyntaxError: Unexpected token '<'`

```

**console log**

**Expected behavior**
this is different than v1.34 where I can see video as soon as the page is loaded. 

[image]

DanwithArduino, 2021-06-02:
This should be closed... The timezone was never set in zoneminder. Hopefully this issue helps someone else.

connortechnology, 2021-06-02:
Good for you for figuring it out.  However not having timezone set SHOULDN'T break montagereview.  So we will fix it and then close this.

commit 9a03c2c1 (2021-06-02): Default to UTC when no timezone set so that montagereview continues working.  Fixes #3274
commit 959107f3 (2021-06-08): Default to UTC when no timezone set so that montagereview continues working.  Fixes #3274
closed 2021-06-02 by connortechnology