ZoneMinder/zoneminder #3287: Spurious warnings regarding linked monitors

hgregorian, 2021-06-10:
**Describe Your Environment**
- 1.36.4
- PPA
- Ubuntu 20.04.2
- Firefox

**Describe the bug**
Most of my cams are represented as two monitors; one for high res in 'nodect' and one for low res, where analysis is performed which can be in mode 'modect' or 'monitor'.  The high res monitor lists the low res monitor as a linked monitor.  The mode (modect/monitor) for the low res monitor is dependent on run state (some cams idle when home, all cams "active" when away).  Since updating from 1.34.x to 1.36.x, the following warning messages continuously occur:
```

2021-06-09 23:07:09	zms_m12		10967	WAR	Can't link to monitor 6, invalid id, function or not enabled	zm_monitor.cpp	2399
2021-06-09 23:07:09	zms_m13		10963	WAR	Can't link to monitor 5, invalid id, function or not enabled	zm_monitor.cpp	2399
2021-06-09 23:07:08	zms_m14		10958	WAR	Can't link to monitor 4, invalid id, function or not enabled	zm_monitor.cpp	2399
2021-06-09 23:07:08	zms_m16		10948	WAR	Can't link to monitor 11, invalid id, function or not enabled	zm_monitor.cpp	2399
...
...
2021-06-09 23:07:00	zmu		10860	WAR	Can't link to monitor 11, invalid id, function or not enabled	zm_monitor.cpp	2399
2021-06-09 23:06:59	zmu		10840	WAR	Can't link to monitor 4, invalid id, function or not enabled	zm_monitor.cpp	2399
2021-06-09 23:06:59	zmu		10831	WAR	Can't link to monitor 5, invalid id, function or not enabled	zm_monitor.cpp	2399
2021-06-09 23:06:58	zmu		10822	WAR	Can't link to monitor 6, invalid id, function or not enabled	zm_monitor.cpp	2399
```

**To Reproduce**
Steps to reproduce the behavior:
1. Create 2 monitors (1 and 2)
2. Add monitor 1 as a linked monitor of monitor 2
3. Set monitor 1 to mode "monitor" and monitor 2 to mode "nodect" 
4. Observe warning(s)
5. Set monitor 1 to mode "modect"
6. Warning(s) no longe present

**Expected behavior**
Seems to be a new warning as it did not appear at all under the same circumstances in 1.34.x.  So I'd expect no warnings.

connortechnology, 2021-06-10:
The warning is doing it's job. Telling you that Monitor is an invalid function for this setup.  

Monitor 1 (the high res) should be NoDect
Monitor 2 (the low res) should be MoDect

hgregorian, 2021-06-10:
Yes, and this is how it is when my "away" running state is set - all low res monitors are MoDect and all high res monitors are NoDect.

I only mentioned it because it's not something that was noted before.  I suppose I can adjust my "home" running state for low and high res monitors to "monitor" for those cameras I don't want to be recording in any way.

> The warning is doing it's job. Telling you that Monitor is an invalid function for this setup.
> 
> Monitor 1 (the high res) should be NoDect
> Monitor 2 (the low res) should be MoDect

hgregorian, 2021-06-10:
Hmmm...my suggestion of setting both to "monitor" doesn't alleviate the warnings as they're rooted in the fact that there's a linked monitor.  Any suggestions on cleaning this up in my use case?