ZoneMinder/zoneminder #3288: Storage 100%

roddone, 2021-06-10:
**Describe Your Environment**
- 1.36.x
- installed from PPA iconnor
- Ubuntu Server 20.04

**Describe the bug**
I have a strange behaviours since the upgrade from latest 1.34 to the 1.36.x versions. all the events from my cameras disappeared and my storage is always 98-100% full.
[image]
[image]

Normaly I use the "PurgeWhenFull" filter to delete oldest events when storage is greater than 75%, so I imagine that's why my events disappeared, but I don't know what occupies so much space.
[image]

Logs page never shows up so I can't see if there are any problems 
[image]

Can you help me to figure out the problem ? is there something I can do to help debugging ?

SteveGilvarry, 2021-06-10:
I think you need to Add StorageArea eq NewStorage to the filter, by default it is looking at the default storage area.

roddone, 2021-06-11:
Hi, 

I already tried this with no success :( 
[image]

connortechnology, 2021-06-14:
perhaps it is something else filling it up?

Maybe try running sudo zmaudit.pl -i
Maybe you have Options->System->OPT_FAST_DELETE checked?

roddone, 2021-06-15:
I found what was the problem, it seems there was too much logs (7 Millions) and it was filling all disk space.

I did a truncate on table Logs and my storage was back to 38% and after a few hours it seems everything is back to normal.

Thank your for your help ;)

closed 2021-06-15 by roddone

roddone, 2021-07-04:
In case you wanted to dig deeper on this, or if it can help someone ...  

The storage reached 100% again, it appears there was (once again) too much logs. On way or another, it seems having too much logs fills the storage and once the storage is full, the system can't write events anymore, which cause errors and writes more and more logs in database.
I don't know why, but it also prevents the system to remove events even if "PurgeWhenFull" is correctly configured.

In my case, the solution was to set 'LOG_DATABASE_LIMIT' to 50000 and temporary disable analysis on my cameras. After a few hours the logs were purged, the extra events were deleted and everything went back to normal.