ZoneMinder/zoneminder #3289: ZM_PATH_ARP not configurable

ProjectPatatoe, 2021-06-10:
**Describe Your Environment**
- 1.36.4
- from ppa
- ubuntu 20.04
- firefox

**Describe the bug**
Get a error in the log `ARP compatible binary not found or not executable by the web user account. Verify ZM_PATH_ARP points to a valid arp tool.` but I don't see this option to change anywhere

**To Reproduce**
Steps to reproduce the behavior:
1. Go through all options pages
2. ctrl+f 'arp' doesnt show anything

**Expected behavior**
Ability to change ZM_PATH_ARP. Probably in 'Network'

knight-of-ni, 2021-06-11:
Paths for executables needed by ZoneMinder are set under the conf.d folder, typically found under /etc/zm/conf.d. See: https://github.com/ZoneMinder/zoneminder/blob/2e22cd5c3b489ee05a2cf6ff749df8995f7e6d38/conf.d/01-system-paths.conf.in

ProjectPatatoe, 2021-06-12:
Nice! I actually didn't know there were more options in /etc/zm since there were so many in the web interface. Maybe there should be a note in the web interface? I'll close this issue since its not really a problem. 

Turns out arp wasn't installed. Didn't know that's no longer a default in ubuntu

closed 2021-06-12 by ProjectPatatoe