ZoneMinder/zoneminder #3293: No image attached in the email by the filter (No image for EIM at ...)

spyjo, 2021-06-12:
**Describe Your Environment**
- Version of ZoneMinder : 1.36.4
- How you installed ZoneMinder : from source
- Full name and version of OS : Debian stable 10.9

**If the issue concerns a camera**
- ZoneMinder Source Type: HTTP simple getting a remote jpg

**Describe the bug**
When a new event is created, I have a filter "UploadAndEmail" which is triggered, and which send me an email with "%EIM%" so with the event image with high score attached.
After upgrading from 1.34.X to 1.36.X, most of the time and, only for this particular source (for now), I have no image attached to the email I receive. I never had this problem with previous version.
In the event, there is of course multiple frames and images which are ok.
In the logs, there is an zmfilter log when sending the email:
`No image for EIM at /var/lib/zoneminder/events/9/2021-06-12/65500/00005-capture.jpg`
But when I look at this folder, the file exists:
```
# ls -l /var/lib/zoneminder/events/9/2021-06-12/65500/00005-capture.jpg
-rw-r--r-- 1 www-data www-data 29796 juin  12 22:27 /var/lib/zoneminder/events/9/2021-06-12/65500/00005-capture.jpg
```
May be the zmfilter try to send the email too early when the file is not really stored ?
Because for this particular event, I see the warning log "No image for EIM" at 22:20:09, when the event was from 22:19:57 to 22:20:28. So it sounds early.

Thank you

connortechnology, 2021-06-14:
You need to add an EndTime IS NOT NULL rule to your filter.  In the past filters had this implicitly but that meant that a filter could never match an in progress event. So now you have to add it yourself.

spyjo, 2021-06-14:
Thanks I modify the filter.

With the choice "End Time" is not NULL, I just had this error in logs:
`Can't execute filter 'SELECT E.*, unix_timestamp(E.StartDateTime) as Time FROM Events as E WHERE ( E.AlarmFrames > 0 and E.EndTime IS NOT NULL ) AND ( E.Uploaded = 0 or E.Emailed = 0 ) ORDER BY E.StartDateTime DESC LIMIT 0,100', ignoring: Unknown column 'E.EndTime' in 'where clause'	`

I changed for "End Date/Time" and now it seems to be ok. I wait for the next event.

connortechnology, 2021-06-14:
Uh oh.  Ok first off I should have said EndDateTime.  I'll have to go looking for why it is giving that error though.

spyjo, 2021-06-14:
I can confirm the image is now always attached in the latests emails received, with the new EndDateTime rule added in the filter.
Thanks !

closed 2021-06-15 by connortechnology