ZoneMinder/zoneminder #3300: PTZ issue (fixed!)

alabamatoy, 2021-06-16:
Discovered a minor issue with the Amcrest PTZ...applies to file Amcrest_HTTP.pm  I know this is not the right way to document this code change, so I apologize in advance.  Currently, the diagonal Pan/Tilt movement commands are ignored by the cameras.

Apparently if you send the camera an arg1 and/or arg2 values of zero for a diagonal movement, it just ignores the command.  If you set the arg1 and arg2 values to 1, it works.  According to the API documentation, arg1 is "Vertical speed, range is [1-8]" and arg 2 is "Horizontal speed, range is [1-8]"  So....I guess it makes sense to ignore the command, in effect the arg1=0 is telling it no vertical speed.

lines 223, 231, 239, and 247 need to be modified as follows:

Change "&channel=0&arg1=0&arg2=1&arg3=0" to "&channel=0&arg1=1&arg2=1&arg3=0"

Working code:

sub moveConUpRight {
  my $self = shift;
  Debug('Move Diagonally Up Right');
  $$self{Monitor}->suspendMotionDetection() if !$self->{Monitor}->{ModectDuringPTZ};
  $$self{LastCmd} = 'code=RightUp&channel=0&arg1=1&arg2=1&arg3=0';
  $self->sendCmd('cgi-bin/ptz.cgi?action=start&'.$$self{LastCmd});
}

sub moveConDownRight {
  my $self = shift;
  Debug('Move Diagonally Down Right');
  $$self{LastCmd} = 'code=RightDown&channel=0&arg1=1&arg2=1&arg3=0';
  $$self{Monitor}->suspendMotionDetection() if !$self->{Monitor}->{ModectDuringPTZ};
  $self->sendCmd('cgi-bin/ptz.cgi?action=start&'.$$self{LastCmd});
}

sub moveConUpLeft {
  my $self = shift;
  Debug('Move Diagonally Up Left');
  $$self{Monitor}->suspendMotionDetection() if !$self->{Monitor}->{ModectDuringPTZ};
  $$self{LastCmd} = 'code=LeftUp&channel=0&arg1=1&arg2=1&arg3=0';
  $self->sendCmd('cgi-bin/ptz.cgi?action=start&'.$$self{LastCmd});
}

sub moveConDownLeft {
  my $self = shift;
  Debug('Move Diagonally Down Left');
  $$self{Monitor}->suspendMotionDetection() if !$self->{Monitor}->{ModectDuringPTZ};
  $$self{LastCmd} = 'code=LeftDown&channel=0&arg1=1&arg2=1&arg3=0';
  $self->sendCmd('cgi-bin/ptz.cgi?action=start&'.$$self{LastCmd});
}

alabamatoy, 2021-06-27:
Respectfully request re-open.  This fix did not make it into 1.36.5 latest update.

connortechnology, 2021-09-13:
This made it in now though right?  So we can close again?

alabamatoy, 2021-09-14:
It is working in latest master.  Recommend close.

commit 9438445d (2021-06-16): Fix PTZ Diagonal cmds. Fixes #3300
closed 2021-09-14 by Carbenium