ZoneMinder/zoneminder #3302: Reverse proxy configuration with Traefik2

MacGyver27, 2021-06-22:
Hello,

I am having quite troubles with configuring reverse proxy for Zoenminder (v1.36.4).

I am using `addPrefix` to add `/zm/` in background so user can directly access Zoneminder on its subdomain cam.example.com. This partially works - I am getting the web page, but some styles are missing and stream is not working. After some time I get redirected to `cam.example.com/zm/index.php` - which does not exist on subdomain context.. If I tries to load `cam.example.com/index.php` it succeeds.

### Direct access:
[image]

vs

### Subdomain access:
[image]

### For the streams its the same scenario:
[image]

Traefik2 configuration for the reverse proxy zoneminder:
```yml
http:

  routers:
    zoneminder.example.com:
      priority: 1
      entryPoints:
        - websecure
      rule: Host(`cam.example.com`)
      service: zoneminder.example.com
      middlewares:
        - zoneminder.example.com+addPrefix
        #- zoneminder.example.com+replacePathRegex

  services:
    zoneminder.example.com:
      loadBalancer:
        passHostHeader: true
        servers:
          - url: 'http://192.168.255.12:50191'

  middlewares:
    zoneminder.example.com+redirectRegex:
      redirectRegex:
        regex: "^https?://cam.example.com/(.*)"
        replacement: "https://cam.example.com/zm/"
        permanent: true
    zoneminder.example.com+addPrefix:
      addPrefix:
        prefix: /zm
    zoneminder.example.com+replacePathRegex:
      replacePathRegex:
        regex: "/zm/index(.+)"
        replacement: "/index$1"
```

Any directions will be appreciated. Thanks, Michal

zombielinux, 2021-06-22:
I have been having the same problem for quite some time, but to a greater extent. 

I have traefik2 set up properly (works with other services) and multiple ZM servers in a multi-server configuration on top of docker swarm. Live streams do not work because of some redirecting that is done to a host that is not accessible from outside the docker network. Likewise, the API access also fails. 

I am inclined to believe at this point that unless you are running a single instance of ZM on non-swarm docker, you are unlikely to be successful.

MacGyver27, 2021-06-22:
@zombielinux actually this is my scenario. Just single instance on non-swarm docker.

Similar deployment has also the Guacamole project. Where you have apache server hosting the app on the prefix `/guacamole`. Here adding just `addPrefix: /guacamole` works 100%..

michael-robbins, 2021-06-30:
So I tried this as well a while back and it never worked for me, instead I used a redirect to solve this, and it's been fine for me (users going to the parent page are nicely redirected automatically):
```
      traefik.http.routers.zoneminder.middlewares: "zoneminder-redir"
      traefik.http.middlewares.zoneminder-redir.redirectregex.regex: "^https:\\/\\/([^\\/]+)\\/?$"
      traefik.http.middlewares.zoneminder-redir.redirectregex.replacement: "https://$1/zm/"
```

MacGyver27, 2021-06-30:
@michael-robbins Tried it as well using different pattern and didnt work:
```yml
        regex: "^https?://cam.example.com/(.*)"
        replacement: "https://cam.example.com/zm/"
```

Yours works for me:

```yml
        regex: "^https?://([^/]+)/?$"
        replacement: "https://$1/zm/"
```
Mind explaining me this part: `([^/]+)/?` please? Appreciate! Michal

michael-robbins, 2021-06-30:
Not gonna lie, I just copied it off another GitHub comment somewhere lol

But yes it's just another way to match the domain!