ZoneMinder/zoneminder #3304: Wrong zip filename with event uploaded on FTP server

spyjo, 2021-06-22:
**Describe Your Environment**
Version of ZoneMinder : 1.36.5
How you installed ZoneMinder : from source
Full name and version of OS : Debian stable 10.9

**Describe the bug**
I have a "UploadAndEmail" filter which upload events on an FTP server. Upload archive format is set as "zip".
In previous versions 1.34.X , the file uploaded was named "Monitorname-IDevent.zip". Since the upgrade to 1.36.X , the file uploaded has the name "-IDevent.zip" , the Monitorname is missing.
I have looked everywhere in the settings but I don't see a setting for the filename, so I assume it's a bug.

Thank you

connortechnology, 2021-06-23:
Definitely a bug. A fix has been committed and will be included in 1.36.6. cb3dd78c6

connortechnology, 2021-06-23:
cb3dd78c61518ca0ab20c05246b6133c7c379423

commit e9845d96 (2021-06-23): Fix Archive FTP .zip filename not including monitor name. Fixes #3304
closed 2021-06-23 by connortechnology