ZoneMinder/zoneminder #3306: bug when adding a new camera

tolik1203, 2021-06-25:
**Describe Your Environment**
- Version of ZoneMinder [release version, development version, or commit] 1.6.008
- How you installed ZoneMinder [e.g. PPA, RPMFusion, from-source, etc] app-store and google play
- Full name and version of OS  Apple IOS 14.4.2 and Android 10
- Browser name and version (if this is an issue with the web interface)

**Describe the bug**
When adding a camera, the ios application hides it, when you make it visible, the application crashes, freezes, the added camera flies far down and the cameras stop loading, if you re-enter the application, the camera becomes hidden again and the video stream does not load.

**To Reproduce**
Steps to reproduce the behavior:
1. Adding a camera via the web interface
2. I log in to the official app from my mobile device
3. I click on the peephole to display the hidden camera
4. I make the camera visible
5. a bug appears

**Expected behavior**
the camera is displayed in the camera list, the broadcast is working

**Debug Logs**
```
[zmNinjaLog.txt](https://github.com/ZoneMinder/zoneminder/files/6715511/zmNinjaLog.txt)
[zmNinjaLog.txt](https://github.com/ZoneMinder/zoneminder/files/6715525/zmNinjaLog.txt)

```

SteveGilvarry, 2021-06-25:
Sounds like this is an issue in zmNinja not Zoneminder.
https://github.com/pliablepixels/zmNinja

closed 2021-06-25 by SteveGilvarry

tolik1203, 2021-06-27:
> Sounds like this is an issue in zmNinja not Zoneminder.
> https://github.com/pliablepixels/zmNinja

Good afternoon. Yes, I also think that this is a problem in the application itself, or rather in the latest update of the application