ZoneMinder/zoneminder #3310: Unable to install ZondeMinder on Debian 10 Buster

stehof2, 2021-07-02:
Hi,

I use an VM to run my Debian 10, this system:   
Static hostname: ZoneMinder
         Icon name: computer-vm
           Chassis: vm
        Machine ID: 2f4c2854c79e4b50a6b229deb257739f
           Boot ID: 490ed5a6dfe94b17843eeed9be1d306c
    Virtualization: vmware
  Operating System: Debian GNU/Linux 10 (buster)
            Kernel: Linux 4.19.0-17-amd64
      Architecture: x86-64

I followed this instruction: https://wiki.zoneminder.com/Debian_10_Buster_with_Zoneminder_1.34.x_from_ZM_Repo
Everything worked fine in the instruction with our error's, till "sudo apt update", this is the output:

Hit:1 http://security.debian.org/debian-security buster/updates InRelease
Hit:2 http://deb.debian.org/debian buster InRelease                            
Hit:3 http://deb.debian.org/debian buster-updates InRelease                    
Err:4 https://zmrepo.zoneminder.com/debian/release-1.34 buster/ InRelease      
  403  Forbidden [IP: 158.69.226.113 443]
Reading package lists... Done
E: Failed to fetch https://zmrepo.zoneminder.com/debian/release-1.34/buster/InRelease  403  Forbidden [IP: 158.69.226.113 443]
E: The repository 'https://zmrepo.zoneminder.com/debian/release-1.34 buster/ InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
W: Target Packages (Packages) is configured multiple times in /etc/apt/sources.list:23 and /etc/apt/sources.list.d/zoneminder.list:1
W: Target Translations (en) is configured multiple times in /etc/apt/sources.list:23 and /etc/apt/sources.list.d/zoneminder.list:1

How to fix it so I'll be able to install ZoneMinder?
Thanks in advance.

usmanatron, 2021-07-03:
+1
If I go to https://zmrepo.zoneminder.com/debian/release-1.36/buster/ in a browser, I get this error text:

`Server unable to read htaccess file, denying access to be safe`

Assuming it's a configuration issue on the server?

SteveGilvarry, 2021-07-03:
It is being relocated currently.

closed 2021-07-03 by SteveGilvarry