ZoneMinder/zoneminder #3313: zmfilter exited abnormally, exit status 9

spyjo, 2021-07-05:
**Describe Your Environment**
Version of ZoneMinder : 1.36.4
How you installed ZoneMinder : from source
Full name and version of OS : Debian stable 10.9

**Describe the bug**
Hello,
I had this error in logs today (in loop every X minutes):
`'zmfilter.pl --filter_id=4 --daemon' exited abnormally, exit status 9
`

This filter upload to FTP the zip file and send an email.

So I started the command manually to see the output:
```
# /usr/bin/perl -wT /usr/local/bin/zmfilter.pl --filter_id=4 --daemon
05/07/2021 18:55:52.846593 zmfilter_4[19934].INF [main:182] [Scanning for events using filter id '4']
Use of uninitialized value in concatenation (.) or string at /usr/local/bin/zmfilter.pl line 530.
05/07/2021 18:55:52.850791 zmfilter_4[19934].INF [main:547] [Creating upload file '/var/tmp/zm/-65906.zip', 3 files]
Can't call method "desiredCompressionLevel" on an undefined value at /usr/share/perl5/Archive/Zip/Archive.pm line 261.
```

The first error "use of uninitialized value in concatenation...." is from issue https://github.com/ZoneMinder/zoneminder/issues/3293 , so it's not relevant here.

After viewing the the event at /zm/index.php?view=event&eid=65906 , the error disappear and the filter is working correctly for this event.
So I assume that viewing the event is creating a file which was missing, but I'm not sure how and why.
I had this bug only for 2 events in the last days, all the others was ok.

Thanks

connortechnology, 2021-07-08:
It's interesting.  So the code just zips up every file in the event directory.  If you turn on debug logging it should at least tell us which file is causing the problem. 

In reality this is an upstream problem in Archive::Zip.   It should test for failure but it doesn't.

spyjo, 2021-07-08:
Here are mode informations with debug logs:

Debug logs of zmdc:

```
08/07/2021 16:45:33.736901 zmdc[6126].INF [ZMServer:411] ['zmfilter.pl --filter_id=4 --daemon' starting at 21/07/08 16:45:33, pid = 6271]
08/07/2021 16:45:33.736903 zmdc[6271].INF [ZMServer:411] ['zmfilter.pl --filter_id=4 --daemon' started at 21/07/08 16:45:33]
08/07/2021 16:45:33.968095 zmdc[6126].ERR [ZMServer:708] ['zmfilter.pl --filter_id=4 --daemon' exited abnormally, exit status 9]
```

Debug logs of zmfilter pid 6271:

```
# cat zm_debug.log.6271
08/07/2021 16:45:33.899865 zmfilter_4[6271].DB1 [ZoneMinder::Logger:321] [LogOpts: level=DB9/DB9, screen=OFF, database=WAR, logfile=DB9->/var/log/zm/zm_debug.log.6271, syslog=OFF]
08/07/2021 16:45:33.959060 zmfilter_4[6271].INF [main:182] [Scanning for events using filter id '4']
08/07/2021 16:45:33.959115 zmfilter_4[6271].DB1 [main:201] [Reloading filters]
08/07/2021 16:45:33.959672 zmfilter_4[6271].DB1 [main:258] [Found filter 'UploadAndEmail']
08/07/2021 16:45:33.961841 zmfilter_4[6271].DB1 [main:272] [Got 1 filters]
08/07/2021 16:45:33.961970 zmfilter_4[6271].DB1 [ZoneMinder::Filter:111] [Filter::Execute SQL (SELECT E.*, unix_timestamp(E.StartDateTime) as Time
         FROM Events as E WHERE (  E.AlarmFrames > 0
 and E.EndDateTime IS NOT NULL
 ) AND ( E.Uploaded = 0 or E.Emailed = 0 ) ORDER BY E.StartDateTime DESC LIMIT 0,100)]
08/07/2021 16:45:33.962542 zmfilter_4[6271].DB1 [ZoneMinder::Filter:124] [Loaded 3 events for filter UploadAndEmail using query (SELECT E.*, unix_timestamp(E.StartDateTime) as Time
         FROM Events as E WHERE (  E.AlarmFrames > 0
 and E.EndDateTime IS NOT NULL
 ) AND ( E.Uploaded = 0 or E.Emailed = 0 ) ORDER BY E.StartDateTime DESC LIMIT 0,100)"]
08/07/2021 16:45:33.962564 zmfilter_4[6271].DB1 [main:297] [Checking filter UploadAndEmail upload, email returned 3 events ]
08/07/2021 16:45:33.962600 zmfilter_4[6271].DB1 [main:307] [Checking event 65934]
08/07/2021 16:45:33.962868 zmfilter_4[6271].INF [main:547] [Creating upload file '/var/tmp/zm/-65934.zip', 3 files]
08/07/2021 16:45:33.962888 zmfilter_4[6271].DB1 [main:551] [Adding /var/lib/zoneminder/events/10/2021-07-06/65934/00157-capture.jpg]
08/07/2021 16:45:33.963050 zmfilter_4[6271].DB1 [main:551] [Adding /var/lib/zoneminder/events/10/2021-07-06/65934/snapshot.jpg]
# 
```

Content of event directory:

```
# ls -la /var/lib/zoneminder/events/10/2021-07-06/65934/
total 27196
drwxr-xr-x 2 www-data www-data     4096 juil.  6 06:12 .
drwxr-xr-x 9 www-data www-data     4096 juil.  6 10:18 ..
-rw-r--r-- 1 www-data www-data   254226 juil.  6 06:12 00157-capture.jpg
-rw-r--r-- 1 www-data www-data 26839203 juil.  6 06:02 65934-video.mp4
-rw-r--r-- 1 www-data www-data   740507 juil.  6 06:02 alarm.jpg
# 
```

So it sounds like it tries to add "snapshot.jpg" in the zip file, but this file doesn't exists in the event directory.

connortechnology, 2021-07-08:
So... apparently our glob will return snapshot.jpg even though it doesn't exist.  I have added a test for file existence before adding it to the zip to work around this.

commit 14047bc8 (2021-07-08): Test for existence of file before adding it to zip.  Apparently our glob can return the snapshot.jpg string even if it doesn't exist. This is a workaround to Archive::Zip not testing for failure and crashing. Fixes #3313
commit 3e59ad9c (2021-07-08): Test for existence of file before adding it to zip.  Apparently our glob can return the snapshot.jpg string even if it doesn't exist. This is a workaround to Archive::Zip not testing for failure and crashing. Fixes #3313
closed 2021-07-08 by connortechnology