ZoneMinder/zoneminder #3316: Deleting monitor results in a backtrace

externo6, 2021-07-08:
**Describe Your Environment**
- Version of ZoneMinder [release version, development version, or commit] release 1.36.5
- How you installed ZoneMinder [e.g. PPA, RPMFusion, from-source, etc] apt
- Full name and version of OS Ubuntu 20.04
- Browser name and version (if this is an issue with the web interface) Edge 

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Delete a monitor
2. Check logs

**Expected behavior**
Monitor deleted without issue

I had to restore the DB to get the backtraces to stop. 

**Debug Logs**
```

Jul 08 14:16:54 zoneminder zms_m7[3151]: ERR [zms_m7] [Bogus number of lines return from query, 0 returned for query SELECT `Id`, `Name`, `ServerId`, `StorageId`, `Type`, `Function`+0, `Enabled`, `DecodingEnabled`, `LinkedMonitors`, `AnalysisFPSLimit`, `AnalysisUpdateDelay`, `MaxFPS`, `AlarmMaxFPS`,`Device`, `Channel`, `Format`, `V4LMultiBuffer`, `V4LCapturesPerFrame`, `Protocol`, `Method`, `Options`, `User`, `Pass`, `Host`, `Port`, `Path`, `SecondPath`, `Width`, `Height`, `Colours`, `Palette`, `Orientation`+0, `Deinterlacing`, `DecoderHWAccelName`, `DecoderHWAccelDevice`, `RTSPDescribe`, `SaveJPEGs`, `VideoWriter`, `EncoderParameters`, `OutputCodec`, `Encoder`, `OutputContainer`, `RecordAudio`, `Brightness`, `Contrast`, `Hue`, `Colour`, `EventPrefix`, `LabelFormat`, `LabelX`, `LabelY`, `LabelSize`,`ImageBufferCount`, `MaxImageBufferCount`, `WarmupCount`, `PreEventCount`, `PostEventCount`, `StreamReplayBuffer`, `AlarmFrameCount`, `SectionLength`, `MinSectionLength`, `FrameSkip`, `MotionFrameSkip`, `FPSReportInterval`, `RefBlendPerc`, `AlarmRefBlendPerc`, `TrackMotion`, `Exif`,`RTSPServer`, `RTSPStreamName`,`SignalCheckPoints`, `SignalCheckColour`, `Importance`-2 FROM `Monitors` WHERE Id=7.]
Jul 08 14:16:54 zoneminder zms_m7[3151]: ERR [zms_m7] [Can't use query result: ]
Jul 08 14:16:54 zoneminder zms_m7[3151]: ERR [zms_m7] [Unable to load monitor id 7 for streaming]
Jul 08 14:16:54 zoneminder zms_m7[3151]: ERR [zms_m7] [Unable set start stream for monitor 7]
Jul 08 14:16:54 zoneminder zms_m7[3151]: ERR [zms_m7] [Got signal 11 (Segmentation fault), crashing]
Jul 08 14:16:54 zoneminder zms_m7[3151]: ERR [zms_m7] [Signal address is 0x450, from 0x560f46e9f414]
Jul 08 14:16:54 zoneminder zms_m7[3151]: ERR [zms_m7] [Backtrace 0: /usr/lib/zoneminder/cgi-bin/nph-zms(+0xe64ca) [0x560f46f0f4ca]]
Jul 08 14:16:54 zoneminder zms_m7[3151]: ERR [zms_m7] [Backtrace 1: /lib/x86_64-linux-gnu/libpthread.so.0(+0x153c0) [0x7f5a7f4253c0]]
Jul 08 14:16:54 zoneminder zms_m7[3151]: ERR [zms_m7] [Backtrace 2: /usr/lib/zoneminder/cgi-bin/nph-zms(+0x76414) [0x560f46e9f414]]
Jul 08 14:16:54 zoneminder zms_m7[3151]: ERR [zms_m7] [Backtrace 3: /usr/lib/zoneminder/cgi-bin/nph-zms(+0xe7794) [0x560f46f10794]]
Jul 08 14:16:54 zoneminder zms_m7[3151]: ERR [zms_m7] [Backtrace 4: /usr/lib/zoneminder/cgi-bin/nph-zms(+0x41c61) [0x560f46e6ac61]]
Jul 08 14:16:54 zoneminder zms_m7[3151]: ERR [zms_m7] [Backtrace 5: /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf3) [0x7f5a7cc520b3]]
Jul 08 14:16:54 zoneminder zms_m7[3151]: ERR [zms_m7] [Backtrace 6: /usr/lib/zoneminder/cgi-bin/nph-zms(+0x42f8e) [0x560f46e6bf8e]]
Jul 08 14:16:54 zoneminder zms_m7[3151]: INF [zms_m7] [Backtrace complete, please execute the following command for more information: addr2line -e /usr/lib/zoneminder/cgi-bin/nph-zms 0x560f46f0f4ca 0x7f5a7f4253c0 0x560f46e9f414 0x560f46f10794 0x560f46e6ac61 0x7f5a7cc520b3 0x560f46e6bf8e]
```

connortechnology, 2021-07-08:
Please note that your logs will still be filled with errors because someone is trying to stream a non-existent monitor.

closed 2021-07-08 by connortechnology

externo6, 2021-07-08:
> Please note that your logs will still be filled with errors because someone is trying to stream a non-existent monitor.

Great thanks for quick feedback!

commit c06cda6a (2021-07-08): Handle the case when the monitor doesn't exist more gracefully and not crash. Fixes #3316
commit 1766bbdf (2021-07-08): Handle the case when the monitor doesn't exist more gracefully and not crash. Fixes #3316