ZoneMinder/zoneminder #3317: Black window when viewing events but got sound. When cursor is placed on thumbnail video plays, download video fine!

TheBudda, 2021-07-09:
Describe Your Environment

    Version 1.36.5
    Installed via deb http://ppa.launchpad.net/iconnor/zoneminder-1.36/ubuntu focal InRelease
    Running on Ubuntu 20.4.2
    Attempted using Firefox, Chrome

If the issue concerns a camera
The cameras are not the problem

Describe the bug
3 cams 3 Monitors appear, all monitors are green and capture is fine, but when trying to view an event, a black window player opens with sound only! No video! But can download the video. I can place cursor over event thumbnail and video plays

To Reproduce
Steps to reproduce the behaviour:

    Go on Console
    Click on any monitor
    Click on any event
    See black window player with sound

Expected behaviour
The event player should show the video with the sound of the event in question. The sound is fine! Thumbnail play is fine! 
Video Download is OK great, playback in VLC with sound is great!

TheBudda, 2021-07-09:
[image]

TheBudda, 2021-07-09:
[image]

TheBudda, 2021-07-09:
[image]

TheBudda, 2021-07-12:
I forgot to say that I did not have this issue with v1.34.25

SteveGilvarry, 2021-07-12:
Any errors in the browser console under developer tools?

TheBudda, 2021-07-13:
DevTools failed to load source map: Could not load content for https://zm.tenpills.me/zm/cache/bootstrap.bundle.js.map: HTTP error: status code 404, net::ERR_HTTP_RESPONSE_CODE_FAILURE

[image]

TheBudda, 2021-07-13:
I played an event while developer tools were running, in the console I got this is the output

zoomrotate: Start
videojs.zoomrotate.js:5 zoomrotate: Init defaults
videojs.zoomrotate.js:10 zoomrotate: Init Extend
videojs.zoomrotate.js:72 zoomrotate: End
videojs.zoomrotate.js:34 zoomrotate: Register init
videojs.zoomrotate.js:43 zoomrotate: [object CSSStyleDeclaration]
videojs.zoomrotate.js:44 zoomrotate: [object CSSStyleDeclaration]
videojs.zoomrotate.js:45 zoomrotate: undefined
videojs.zoomrotate.js:46 zoomrotate: 1
videojs.zoomrotate.js:68 zoomrotate: Register end
skins_classic_views_js_event-base-1624387356.js:98 No cue frames for event
skins_classic_js_skin-base-1624387356.js:828 Nothing found for #downloadBtn
DevTools failed to load source map: Could not load content for https://zm.tenpills.me/zm/cache/bootstrap.bundle.js.map: HTTP error: status code 404, net::ERR_HTTP_RESPONSE_CODE_FAILURE

[zm.tenpills.me-1626181966280.log](https://github.com/ZoneMinder/zoneminder/files/6809075/zm.tenpills.me-1626181966280.log)

[image]

TheBudda, 2021-07-16:
Has anyone got any ideas on this, please? I still can view any events. I am thinking of downgrading to 1.34.22 :(

SteveGilvarry, 2021-07-16:
Anything stand out in Apache logs?

TheBudda, 2021-07-16:
> Anything stand out in Apache logs?

I am using Nginx

TheBudda, 2021-07-16:
> Anything stand out in Apache logs?

If I downgrade to 1.34.22 everything works how its shoud

SteveGilvarry, 2021-07-16:
The GUI has a lot of changes in 1.36, it works for me in Apache. But maybe something in your nginx config, luckily I believe it also has logs for you to check.

connortechnology, 2021-07-16:
I suspect that you are encoding or doing passthrough as h265.  You can select mjpeg from the codec drop down at top right of event view or reconfigure your monitor to do something else.

TheBudda, 2021-07-16:
> I suspect that you are encoding or doing passthrough as h265. You can select mjpeg from the codec drop down at top right of event view or reconfigure your monitor to do something else.

Isaac Connor, you are a lifesaver. You were 100% right!! I was encoding. I thought I must. I disabled encoding and turned on mjpeg from the codec dropdown and playback is now working super!! Thank you so much!! I am so chuffed \o/

closed 2021-07-16 by TheBudda