ZoneMinder/zoneminder #3318: RHEL 8 nightly builds not available

dodexahedron, 2021-07-12:
Looks like the RHEL 8 nightly builds are not being pushed to zmrepo, and the zmrepo folders for RHEL 8 are all empty.

knight-of-ni, 2021-07-13:
Isaac has moved along with his servers. He is still getting his infrastructure back up. We need to wait for that to complete.

dodexahedron, 2021-08-16:
Just an update, here. The repo is populated, again, but seems to not be receiving new builds, specifically for RHEL 8.
Newest there is from June 4th. All other distros are up to date.

knight-of-ni, 2021-08-16:
Yeah, I noticed that yesterday. Sorry for the delay. There are three problems. Two are resolved. The third I saw last night, but I can't duplicate it outside of packpack/docker, which is used in our build engine. I have not had the time to properly diagnose this latest issue.

knight-of-ni, 2021-08-16:
Don't know if this helps you, but the master branch currently mock builds just fine on el8.

knight-of-ni, 2021-08-16:
el8 builds are populating now. Dnf should show new packages after clearing your dnf cache.

dodexahedron, 2021-08-16:
Confirmed I can see the new packages.
Upgrade succeeds and zoneminder starts as expected.
Thanks

closed 2021-08-16 by dodexahedron