ZoneMinder/zoneminder #3320: Zoneminder memory issue

beatles1235, 2021-07-22:
**I'm knew to github in terms of doing bug reports, so please excuse my ignorance. I'll try to supply as much info/correct my mistakes**

- Version of ZoneMinder  (1.37.x)
- How you installed ZoneMinder (from iconnor ppa zoneminder-master)
- Full name and version of OS (Ubuntu 20.04.2)
- Browser name and version (N/A)

**Describe the bug**
A clear and concise description of what the bug is.

I was running version 1.34.x, and recently upgraded using the master branch to 1.37.x, and I now have issues with zoneminder consuming 100% of my memory. I was receiving errors of the packet queue being full, so I kept increasing the Maximum Image Buffer Size, but it eventually fills up, and I have to keep increasing it. To test, I went ahead and just left it blank (so it's unlimited) and my 1 camera keeps consuming memory until it maxes out (I currently have ~94GB of RAM), then it goes to swap until that fills up. Eventually, it crashes and starts over. If i sent a limit on the Max Image buffer size, I begin losing frames and it starts distorting the images. (This did not happen in 1.34, i tested 1.36 before going with 1.34, and it appears to be in 1.36 too)

I've read a few other people describing this exact issue, but I haven't seen any solutions to this. I went ahead and installed valgard and did as instructed (install valgrind, sudo zmdc.pl stop zmwatch.pl sudo zmdc.pl stop zmc -m whateverthemonitoridis sudo su -s/bin/bash -c"valgrind -v --leak-check=full --show-leak-kinds=all zmc -m monitorid" www-data) but oddly enough, my 1 monitor only consumes around 3-4GB at most, and never runs out of memory when ran through valgrind...

This bug only happens when I let zoneminder start up it's capturing process as it normally would.

**Debug Logs**
I don't have any debug logs to go on right now, as nothing shows up that seems to be pertinent. I will gladly add logs if requested.

connortechnology, 2021-07-22:
This happens because the consumer thread is slow for some reason.  For example if we are doing mp4 encoding  or disk are slow, or there is a report where someone was doing deinterlacing and it was causing the problem.  So let's start with some screen shots of your monitor settings.

beatles1235, 2021-07-23:
Thanks for the quick reply! I don't think I have anything too unordinary but let me know! I also have 16 cores, and 15K RPM SAS drives, so I think my performance should be okay.

[image]
[image]
[image]
[image]
[image]

mosh5382, 2021-07-23:
I have the same issue as @beatles1235.  My Zoneminder installation ran without issue for years, most recently on 1.34.  This issue showed up starting with 1.36.  I store captured video to SSD so storage should not be slow.  

This thread seems related:
https://forums.zoneminder.com/viewtopic.php?f=43&t=30846

However disabling "save jpegs" didn't solve my issue.  My camera setup is very similar to OP using H264 Pass-thru and modect.

Please let me know if you need additional information about my configuration.

mosh5382, 2021-08-08:
Is there anyway to revert to 1.34?  I attempted to install 1.34 but the database has changed and there were issues.

My system has been unusable for a couple months and I'd rather revert to 1.34 to get it running again.  Seems like I may have to rebuild.  

The lesson here is take database backups before every upgrade.

cpuetzer, 2021-09-09:
I am running 1.36.5 and am experiencing the same issue. Even if I stop and restart the application it will still consume the same amount of memory unless I reboot the machine.

connortechnology, 2021-09-09:
If you are doing 4k video, you machine likely can't keep up.  ffmpeg libs will silently drop frames, but our new design allows us to keep capturing even when decoding slows down which then consumes all ram.  So we HAVE to set a MaxImageBuffer so that we wait.

bitcoinlizard, 2021-09-11:
So this is a behavior change between 1.34 and 1.36?   I ask because my hardware ran fine on 1.34

My cameras are are 1080 and indeed, if I only run 2 of the 4 the system is stable and the memory doesn't quickly increase as it does when running 4 cameras.  So this is a CPU constraint?  If my CPU is able to keep up then I won't see the fast memory usage increase and eventually running out?

I found setting MaxImageBuffer not to be that helpful.  I would quickly hit the buffer and then the zmc process would restart so my video capture would be cut off.

mosh5382, 2021-09-11:
So this is a behavior change between 1.34 and 1.36? I ask because my hardware ran fine on 1.34

My cameras are are 1080 and indeed, if I only run 2 of the 4 the system is stable and the memory doesn't quickly increase as it does when running 4 cameras. So this is a CPU constraint? If my CPU is able to keep up then I won't see the fast memory usage increase and eventually running out?

I found setting MaxImageBuffer not to be that helpful. I would quickly hit the buffer and then the zmc process would restart so my video capture would be cut off.

cpuetzer, 2021-09-13:
When the system begins to run out of memory restart apache and it will solve the issue. I have added to crontab to restart every night at 2:30AM Sun-Fri and am hoping that this solves the problem for me temporarily until there is a code fix in a later version of the software.

connortechnology, 2021-09-13:
Please update to 1.36.7.  We fixed an issue that reduces cpu by a lot, which allows zm to keep up and free memory. It MAY fix this issue.

beatles1235, 2021-09-14:
I just updated whatever was on the master PPA last night, and the performance is 1000x better. It seems my memory usage was sitting pretty stable around 14GB for my two 8MP cameras. I'll monitor for another couple days, but looking good! What was the fix out of curiousity?

Carbenium, 2021-09-14:
@beatles1235 See 58f933ba1e88fa79916ac1a42b3a308190a4d894 (credits go to @SudoMike for finding and fixing this)

The problem was that the slower (but more universally applicable) versions of the motion detection algorithms were used, even if your CPU supports the optimized versions. Since it took too long to process the frames, the backlog grew and filled the RAM.

cpuetzer, 2021-09-14:
Hi Connor I updated to 1.36.7 and will let you know how if it solves the issue.

mosh5382, 2021-09-14:
This is a major improvement.  I'm now able to run all 4 of my cameras versus only being able to run 2 prior to 1.36.7.  However my system is right on the boarder.  If I use the web interface I will push over 100% CPU and ultimately start using up system RAM until Zoneminder crashes.  

Setting MaxImageBuffer will probably be a livable solution now as the monitors will only restart when I'm performing other activities with the system.  However, I did not have these issues when running 1.34.X.  I could probably consider a CPU upgrade as I'm currently running a lower end dual core CPU.

beatles1235, 2021-09-17:
So overall, I'm very happy with this update. The only noticed is performance is great on modect. I made two changes (added camera passthrough, and changed it to mocord) and my issue came back with 100% memory usage, crash, rinse/repeat. I don't necessarily need Mocord but I figured it would be nice initially to record everything until I can fine-tune my motion detection. Any ideas why mocord would be doing this? Anyways thanks and amazing work guys :)

connortechnology, 2021-09-17:
There is still one more issue that I believe I have fixed in master.  If decode/analysis thread bounce around each other enough they can lock up.  Current master is showing pretty rock solid.  I'll release 1.36.8 soon.

connortechnology, 2021-12-14:
I believe current releases have this pretty much fixed.  Closing.

closed 2021-12-14 by connortechnology

eliaspizarro, 2023-09-04:
:~$ free -h
               total        used        free      shared  buff/cache   available
Mem:           7.8Gi       5.6Gi       127Mi        24Mi       2.3Gi       2.1Gi
Swap:          974Mi       974Mi          **0B**

Why?