ZoneMinder/zoneminder #3324: Exporting video leads to empty tar/zip file

maretodoric, 2021-08-02:
**Describe Your Environment**
- Version of ZoneMinder [release version, development version, or commit]: 1.36.5
- How you installed ZoneMinder [e.g. PPA, RPMFusion, from-source, etc]:  RPMFusion
- Full name and version of OS: CentOS 8 (CentOS Linux release 8.2.2004 (Core))
- Browser name and version (if this is an issue with the web interface): Google Chrome 92.0.4515.107

**Describe the bug**
I have an archived video(s) that i would like to download. I go to Filters -> Search for archived videos -> Select -> Click on download then click on "Generate Download"
I will briefly wait for data to be exported then i get an almost empty zip file.
It will contain folder named "zmExport_207327"
zmFileList.txt file containing:

```
zmExport207327/zmEventImagesMaster.html
/var/lib/zoneminder/temp/zmExport_207327/zmFileList.txt
```

And empty folders named with event ID (for example 65896)
No video files anywhere.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to main page
2. Click on 'Filters '
3. Archive Status equal to Archived Only
3. Click on 'List Matches'
4. Select video to export
5. Click on "Download Video"
6. Click on "GENERATE DOWNLOAD"

**Expected behavior**
A zip file containing video should be downloaded

**Debug Logs**
Please advise how to extract debug logs

connortechnology, 2021-09-13:
Please update to 1.36.7 and let us know if the issue is fixed.  In 1.36.6 we updated a lot of the export/downloading code.

I was never able to reproduce this bug, but may have fixed it anyways.

tommysnello, 2021-09-27:
i have the same issue with zoneminder 1.36.7 and ubuntu 20.04

jorioux, 2021-10-21:
Same issue here with 1.36.7 on ubuntu 21.10.
I've enabled `OPT_FFMPEG`, and `ffmpeg` is installed on ubuntu.

Vic-Y, 2021-11-11:
{Thank you to developers who contribute to ZM!}

I found the same/similar problem running ZM 1.36.10 on Ubuntu 20.04 LTS. Downloading video generates a ZIP or TAR file containing folders and files as described by OP and missing the video. I was not filtering for archived events though, just listed events for past day.

If I view a single event, then I the download video button works. However, I would like to download multiple events, which is where the ZIP file is broken.

connortechnology, 2023-09-15:
Update, as you can see above it was fixed a long time ago.

PX03AFK, 2023-09-15:
I have just checked the script which was updated for the fix and I can see that this is the update applied in November 2021.  The history for this script indicates that there have been a few more updates since that one but nothing to suggest any further fix for the problem as described.  If it really is fixed can you help me establish why I have the problem.  What evidence can I collect?

connortechnology, 2023-09-15:
We need the web_php.log from doing the export.  Perhaps turn on debugging first.

PX03AFK, 2023-09-15:
I've turned on Log_Debug and changed Log_Debug_Level to 9 with no component specified and nothing in the web_php.log. Then I changed Log_Level_Weblog to debug but still nothing.  I even tried specifying _web as a component.  I've also restarted ZM with debug set on.

I did, however, sort out the problem trying export as a zip file.  zip wasn't installed!

commit a0b60aa4 (2021-11-17): Restore inclusion of video files in export when not including images. Fixes #3324
commit 4945a016 (2021-11-17): Restore inclusion of video files in export when not including images. Fixes #3324