ZoneMinder/zoneminder #3327: Please start using docker version tags

SaturnusDJ, 2021-08-08:
One of the benefits of Docker is that you can keep configurations and data apart. Then any version of the program can used with the configuration and data without extra effort. This is particularly useful when a certain version has bugs, then you can easily downgrade.

Versions are not being used by https://hub.docker.com/r/zoneminderhq/zoneminder, which I assume is maintained by someone on the ZoneMinder team?

Can you please start using version tags?
For example: `zoneminderhq/zoneminder:e17-1.36.5`

SaturnusDJ, 2021-08-08:
The right place for this of course is https://github.com/ZoneMinder/zmdockerfiles. Strange to have missed that. Closing issue.

closed 2021-08-08 by SaturnusDJ

SteveGilvarry, 2021-08-08:
Sorry Zoneminderhq is not Zoneminder team.

SaturnusDJ, 2021-08-08:
@SteveGilvarry 
Thanks, ok even no relation with https://github.com/ZoneMinder/zmdockerfiles?

SteveGilvarry, 2021-08-08:
I did confirm Isaac saying we are not in control of Zoneminderhq, but yet I see us referring to it in the repo. I will clarify with @connortechnology 

Dlandon is most common one people were using. And bkjaya has one and is active in forums. Also we would look at any PRs to update it.