ZoneMinder/zoneminder #3336: Cannot view live stream with mp4 format

cyrildtm, 2021-08-21:
**Describe Your Environment**
- Version of ZoneMinder [release version, development version, or commit]  1.36.5
- How you installed ZoneMinder [e.g. PPA, RPMFusion, from-source, etc]    deb https://zmrepo.zoneminder.com/debian/release-1.36/buster/
- Full name and version of OS    Debian 10
- Browser name and version (if this is an issue with the web interface)    Chrome 92.0.4515.131 (Official Build) (64-bit)

**If the issue concerns a camera**
- Make and Model    Raspberry pi 4
- frame rate  30 fps
- resolution    1920x1080
- ZoneMinder Source Type:    ffmpeg, RTSP/TCP, cuda

**Describe the bug**
A clear and concise description of what the bug is.

I cannot view live stream from monitor in MP4 format. 
I can view live stream in JPEG format, but my apache server's hard drive constantly read/write during the time I watch live stream, and with this constant rate of I/O my HDD will fail very shortly IF I keep watching live stream. 
iotop says > 5M/s total disk write even when there are no events being recorded, from journal process (jbd)
I can view recorded events in MP4 format and download mp4 files with no problems. 

- The same issue has been reported a year ago: 
- https://forums.zoneminder.com/viewtopic.php?t=30021

I took the same procedure and got the same problem. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

1. Set the following parameters in "Options": 
Web: 
WEB_USE_OBJECT_TAGS: Tried both "yes" and "no"
Images: 
MPEG_LIVE_FORMAT: mp4 [note: if I leave this field empty, then jpeg is used, can watch live stream, but constant disk I/O]
MPEG_REPLAY_FORMAT: mp4
High B/W:
WEB_H_STREAM_METHOD: mpeg
WEB_H_VIDEO_BITRATE: 150000
WEB_H_VIDEO_MAXFPS: 30

2. Watch recorded events. It works. 
3. Watch Monitor Live Stream. In Chrome, a media player interface shows up, black / blank content with spinning circle for a few seconds, and then the spinning circle disappears leaving the blank player behind. 

**Expected behavior**
A clear and concise description of what you expected to happen.

Just being able to watch live stream in mp4 format. 
If mp4 is too hard to generate in real time, any other live streaming format rather than JPEG is acceptable, as long as creating such content does not involve constant and excessive disk I/O operations. 

**Debug Logs**
```

2021-08-21 12:43:49 | zms_m1 |   | 25636 | ERR | Backtrace 5: /lib/x86_64-linux-gnu/libc.so.6(clone+0x3f) [0x7f8a983974cf] | zm_signal.cpp | 100
2021-08-21 12:43:49 | zms_m1 |   | 25636 | ERR | Backtrace 4: /lib/x86_64-linux-gnu/libpthread.so.0(+0x7fa3) [0x7f8a9a1e3fa3] | zm_signal.cpp | 100
2021-08-21 12:43:49 | zms_m1 |   | 25636 | ERR | Backtrace 3: /usr/lib/zoneminder/cgi-bin/nph-zms(+0xa7930) [0x55560d18b930] | zm_signal.cpp | 100
2021-08-21 12:43:49 | zms_m1 |   | 25636 | ERR | Backtrace 2: /usr/lib/zoneminder/cgi-bin/nph-zms(+0xa71c0) [0x55560d18b1c0] | zm_signal.cpp | 100
2021-08-21 12:43:49 | zms_m1 |   | 25636 | ERR | Backtrace 1: /lib/x86_64-linux-gnu/libpthread.so.0(+0x12730) [0x7f8a9a1ee730] | zm_signal.cpp | 100
2021-08-21 12:43:49 | zms_m1 |   | 25636 | ERR | Backtrace 0: /usr/lib/zoneminder/cgi-bin/nph-zms(+0xee8b6) [0x55560d1d28b6] | zm_signal.cpp | 100
2021-08-21 12:43:49 | zms_m1 |   | 25636 | ERR | Signal address is (nil), from 0x55560d18b1c0 | zm_signal.cpp | 79
2021-08-21 12:43:49 | zms_m1 |   | 25636 | ERR | Got signal 11 (Segmentation fault), crashing | zm_signal.cpp | 49
2021-08-21 12:43:49 | zms_m1 |   | 25634 | ERR | Could not open codec. Error code -542398533 "Generic error in an external library" | zm_mpeg.cpp | 301

```

AndrewwHummer, 2021-09-15:
Do you stream the Video from the PI as h264 or jpegs (via Motion or otherwise)?
It is advisiable to use the PIs GPU to encode the Video Stream to save on CPU Power on the server side.

I use a Pi Zero with MotionEyeOS to stream h264 to zoneminder, works like a charm in liveview and otherwise.
Can you share how you stream from the Pi to Zoneminder?

Cheers :)

cyrildtm, 2021-09-15:
> Do you stream the Video from the PI as h264 or jpegs (via Motion or otherwise)?
> It is advisiable to use the PIs GPU to encode the Video Stream to save on CPU Power on the server side.
>
> I use a Pi [...]

Thanks for the suggestion. 

My Pi 4 gives me a H264 stream. 
I decode the H264 stream before recording. I don't want to use H264 passthrough since I add features like time stamp and private zones. 
My problem is with the live view from the ZoneMinder web page. It appears that it stores the decoded images as jpegs on physical disk before passing it to apache which then shows it on the web page. 
I would like to see the decoded data gets encoded before being sent to user client. 
I have NVENC and a 4-core multithread CPU. I don't care about extra power. I just need to see the video stream that is actually getting recorded on storage (not the raw stream, see above) without wearing out my disk holding apache server excessively.

connortechnology, 2021-09-15:
Change StreamReplayBuffer in Monitor->Misc tab to 0.

cyrildtm, 2021-09-15:
> Change StreamReplayBuffer in Monitor->Misc tab to 0.

Thanks! That stopped the disk writing. 
Just wanted to point out, the option was in `Monitor -> Buffers`, and it used to be 5 in my case. 
The video feed is still a bit laggy since it's a sequence of JPEG files, but I can live with it. The CPU usage from `nph-zms` is about 20%-30% from `top`, which is also fine.

closed 2021-09-15 by cyrildtm