ZoneMinder/zoneminder #3337: Event was not found when using previous or next event

MoHf7f4, 2021-08-22:
**Describe Your Environment**
- Version of ZoneMinder [release version, development version, or commit]
1.36.5
- How you installed ZoneMinder [e.g. PPA, RPMFusion, from-source, etc]
PPA
- Full name and version of OS
4.15.0-154-generic #161-Ubuntu SMP Fri Jul 30 13:04:17 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
Ubuntu 18.04.5 LTS \n \l
- Browser name and version (if this is an issue with the web interface)
Chrome Version 92.0.4515.159 (Official Build) (64-bit)

**Describe the bug**
When using the Camera playback view I get "Event was not found." when moving to previous or next event.
View url part is: 
/zm/?view=event&eid=1793381&filter[Query][terms][0][attr]=MonitorId&filter[Query][terms][0][op]=%3d&filter[Query][terms][0][val]=17
Then trying to move to previous I get this url:
/zm/index.php?view=event&eid=0&filter%5BQuery%5D%5Bterms%5D%5B0%5D%5Battr%5D=MonitorId&filter%5BQuery%5D%5Bterms%5D%5B0%5D%5Bop%5D=%3D&filter%5BQuery%5D%5Bterms%5D%5B0%5D%5Bval%5D=17&sort_field=StartDateTime&sort_asc=1
Used to work great in 1.34, now it makes going through events not feasible because I need to go to the event list for a camera and click each one if I want to keep my place.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Console'
2. Click on 'any camera'
3. Scroll down to 'select an event'
4. Click 'previous' or 'next' event
5. immediately or after a few it'll throw the error and need to use the back button because there is no new window open in 1.36.

**Expected behavior**
A clear and concise description of what you expected to happen.
Consistenly show the next or previous event video.

**Debug Logs**
```

<html>
<body>
<!--StartFragment-->

2021-08-22 14:15:07 | web_php |   | 17982 | ERR | No data to read from socket | /usr/share/zoneminder/www/includes/functions.php | 1902
-- | -- | -- | -- | -- | -- | -- | --
2021-08-22 14:15:07 | web_php |   | 17982 | ERR | Timed out waiting for msg /run/zm/zms-125721s.sock | /usr/share/zoneminder/www/ajax/stream.php | 95
2021-08-22 14:15:05 | web_js |   | 17985 | ERR | getStreamCmdResponse stream error: No data to read from socket - checkStreamForErrors() | ?view=watch&mid=17 | -
2021-08-22 14:15:05 | web_php |   | 17982 | ERR | No data to read from socket | /usr/share/zoneminder/www/includes/functions.php | 1902
2021-08-22 14:15:05 | web_php |   | 17982 | ERR | Timed out waiting for msg /run/zm/zms-125721s.sock | /usr/share/zoneminder/www/ajax/stream.php | 95
2021-08-22 14:15:04 | zms_m17 |   | 26938 | ERR | Backtrace 7: /usr/lib/zoneminder/cgi-bin/nph-zms(+0x1948a) [0x558aa742c48a] | zm_signal.cpp | 100
2021-08-22 14:15:04 | zms_m17 |   | 26938 | ERR | Backtrace 6: /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xe7) [0x7f62c3658bf7] | zm_signal.cpp | 100
2021-08-22 14:15:04 | zms_m17 |   | 26938 | ERR | Backtrace 5: /usr/lib/zoneminder/cgi-bin/nph-zms(+0x17df5) [0x558aa742adf5] | zm_signal.cpp | 100
2021-08-22 14:15:04 | zms_m17 |   | 26938 | ERR | Backtrace 4: /usr/lib/zoneminder/cgi-bin/nph-zms(+0x68b40) [0x558aa747bb40] | zm_signal.cpp | 100
2021-08-22 14:15:04 | zms_m17 |   | 26938 | ERR | Backtrace 3: /usr/lib/zoneminder/cgi-bin/nph-zms(+0xd1c2a) [0x558aa74e4c2a] | zm_signal.cpp | 100
2021-08-22 14:15:04 | zms_m17 |   | 26938 | ERR | Backtrace 2: /usr/lib/zoneminder/cgi-bin/nph-zms(+0x63475) [0x558aa7476475] | zm_signal.cpp | 100
2021-08-22 14:15:04 | zms_m17 |   | 26938 | ERR | Backtrace 1: /lib/x86_64-linux-gnu/libpthread.so.0(+0x12980) [0x7f62c705f980] | zm_signal.cpp | 100
2021-08-22 14:15:04 | zms_m17 |   | 26938 | ERR | Backtrace 0: /usr/lib/zoneminder/cgi-bin/nph-zms(+0xcc1c4) [0x558aa74df1c4] | zm_signal.cpp | 100
2021-08-22 14:15:04 | zms_m17 |   | 26938 | ERR | Signal address is 0x558aa7476475, from 0x558aa7476475 | zm_signal.cpp | 79
2021-08-22 14:15:04 | zms_m17 |   | 26938 | ERR | Got signal 8 (Floating point exception), crashing | zm_signal.cpp

<!--EndFragment-->
</body>
</html>

```

connortechnology, 2023-01-04:
It has been a long time since 1.36.5.  Can you update to .32 and report back whether it is still an issue?  I don't think it will be.

MoHf7f4, 2023-01-04:
I’m embarrassed this is still open.  Please close it.  There was a network
adapter problem that caused it.  Unrelated to ZM.

closed 2023-01-04 by connortechnology