ZoneMinder/zoneminder #3343: Event video does not auto-play on Safari

dvsingletary, 2021-08-28:
**Describe Your Environment**
- Zoneminder 1.36.5 (but I've seen this with 1.34 also, possibly earlier)
- Installed via RPM
- Centos 7
- Safari 14.1.2

**Describe the bug**
When I select an event to view the video for, the video does not auto-play as it does in Chrome and Firefox.  Instead there is an image of the first frame and a big triangular "play" button in the upper left corner. This makes gapless playback painful as each video has to be manually started.

**To Reproduce**
Steps to reproduce the behavior:
1. In Safari, log into Zoneminder and go to an event list and select an event to view it.
2.  Event loads, video playback does not start in Safari (but does if using Chrome or Firefox)

**Expected behavior**
Video playback should auto-start in Safari the same as Chrome/Firefox.

Note: I've tried editing the site settings in Safari to allow auto play (even with sound) and this has no effect.

dvsingletary, 2021-08-28:
I did an experiment to see if I could hack a fix to this - I'm not sure if it's the right way to do it, but I did find that simply adding the "autoplay" directive to the video tag in the event.php file of the classic skin resolved the problem.  I'm not sure by what mechanism ZoneMinder is triggering the auto-play, but whatever mechanism that is, it doesn't work in Safari (but the standard "autoplay" tag does work).

As a temporary fix, I modified:
/usr/share/zoneminder/www/skins/classic/views/event.php

line 198, where the <video id ...> tag is, and I added "autoplay" after "video".

connortechnology, 2021-08-29:
I merged your fix.  Will be included in 1.36.6

commit d102732a (2021-08-29): add autoplay tag.  Fixes #3343
closed 2021-08-29 by connortechnology