ZoneMinder/zoneminder #3344: zm_rtsp_server crash

HaoLVb, 2021-08-29:
Describe Your Environment
Version of ZoneMinder : 1.36.1
How you installed ZoneMinder : from source
Full name and version of OS : Ubuntu 20.04

Describe the bug:
I have a problem, zm_rtsp_server was crashed. After i checked it was caused by a cam causing this error. I don't know the reason but I think a cam can't crash the system
[image]

AndrewwHummer, 2021-09-15:
Did you verify the rtsp stream is readable by other applications like VLC Media Player?

The cam may very well crash the system if the stream is not in the correct format or an execption is not handled correctly by the code.
Garbage in Garbage out. Sometimes this crashes a system.

Check that your stream is really readable.

Cheers :)

connortechnology, 2021-09-15:
I have seen this crash as well, but it is very intermittent and I havn't caught it with debug logs turned on yet.

commit 1ee85caa (2021-09-15): Don't crash when unable to create source. erase will call the desctructor.  Fixes #3344
commit 2b18dd59 (2021-09-15): Don't crash when unable to create source. erase will call the desctructor.  Fixes #3344
closed 2021-09-15 by connortechnology