ZoneMinder/zoneminder #3356: zmrepo possibly not updating

dodexahedron, 2021-09-17:
Looks like the last build tag available on zmrepo is 97c10fc13.

One of the commits after that one is particularly exciting, related to SSE and motion detection. Definitely curious to run before/after benchmarking on that change, with a ton of high-res cameras doing lots of zoned motion detection. :)

knight-of-ni, 2021-09-19:
We have a lot of builds on zmrepo. Exactly which build and architecture are you referring to? buster? bionic? el8? fedora? something else?????

dodexahedron, 2021-09-19:
Sorry. RHEL 8.
Get Outlook for Android

knight-of-ni, 2021-09-19:
hmmmm, Looks okay to me. 

Master branch is at fba7dba3 and the associated package `zoneminder-1.37.1-1.18.20210916gitgfba7dba3d.el8.x86_64.rpm` shows in zmrepo:
https://github.com/ZoneMinder/zoneminder/commits/master

n4cn4c, 2021-09-29:
Same here for centos 8
the packages are indeed in the repository, but dnf dont want to update zoneminder.

dnf info zoneminder
source:  zoneminder-1.37.1-1.161.20210913gitg97c10fc13.el8.src.rpm

 dnf repoinfo zmrepo
Repo-baseurl :       : https://zmrepo.zoneminder.com/el/8/x86_64

dnf update
 no actions to perform

closed 2021-09-29 by no-response[bot]

n4cn4c, 2021-09-29:
Problem is in package version. 
current version is: zoneminder-1.37.1-1.**40**.20210928gitg509a5549b
package with build tag 97c10fc has version zoneminder-1.37.1-1.**161**.20210913gitg97c10fc13
so 161>40 and dnf thinks that version 161 is the latest.

fix
rpm -e --nodeps zoneminder zoneminder-common zoneminder-httpd
dnf install zoneminder zoneminder-common zoneminder-httpd