ZoneMinder/zoneminder #3359: Playback of events are too fast. Maybe same as Issue #676

tedukes, 2021-09-19:
**Describe Your Environment**
Running ZM 1.36.5, CentOS 7 from RPM Fusion repository

**Describe the bug**

https://github.com/ZoneMinder/zoneminder/issues/676

Event replay is too fast. Doing gapless events, the first replay is fine, then its a blur. Been having this issue since June. I posted in the forum but did not get a response. It has been going on since 2 releases before the current one.

It does it with Firefox and Edge on a Windows 10 PC as well as on the CentOS server with Firefox.

I have tried different playback formats, makes no difference.

HELP!!

connortechnology, 2021-11-29:
This was fixed in 1.36.11.

closed 2021-11-29 by connortechnology

tedukes, 2021-11-29:
Its not fixed. Even tried rolling my own.

tedukes, 2021-12-10:
Done with Zoneminder. Been using it for probably 10 yrs. Even donated $$ Go fuck yourself!! Fucking morons trying to do the work of intelligent people!! You're not fooling anyone, asshole!!