ZoneMinder/zoneminder #3362: Debian bullseye package libx264 and lib265 breaks ZM

cyrildtm, 2021-09-30:
**Describe Your Environment**
- Version of ZoneMinder 1.36.7
- How you installed ZoneMinder https://zmrepo.zoneminder.com/debian/release-1.36/buster/
- Full name and version of OS debian 11 bullseye
- Browser name and version Firefox 92.0.1

**If the issue concerns a camera**
- Make and Model Raspberry pi 4 (v4l2rtspserver)
- frame rate 30 fps
- resolution 1080p
- ZoneMinder Source Type: rtsp, ffmpeg, cuda

**Describe the bug**

I understand this is not officially supported by 1.36.7 (the bullseye folder in the repo is almost empty). But this may help with transitioning to the new release in the future. 

Around this morning the source stopped working. ZM log said zmc exit code 127, internet search said not enough memory, and I knew it was irrelevant. 
Tried running zmc in console, said libx264.so.155 and libx265.so.165 were missing. 
Bullseye at this moment has 160 and 192 for both libraries, respectively. 

I'm pretty amazed that everything has been working mostly(*) since I upgraded debian from buster about two weeks ago. 
I tried master 1.37.1, same errors. 
(*) I remember the monitor thumbnail in front page has been broken, but I don't remember since when. 

**Temporary fix**

Temporarily installing the older versions of the two packages solved the issue. I temporarily added a buster entry in source.list, installed the packages, and then disabled it. 

Copying the newer versions (160&192, I tried with ln -s) lead to `zmc` displaying help message in console, but still giving exit code 127 in Zoneminder.

cyrildtm, 2021-11-22:
Now that the latest versions are targeted at bullseye, this becomes a relevant issue. 
Here's what I found: 

```
$ ldd /bin/zmu
        linux-vdso.so.1 (0x00007ffcc954a000)
        libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f49bfb6b000)
        libjpeg.so.62 => /lib/x86_64-linux-gnu/libjpeg.so.62 (0x00007f49bfae7000)
        libcrypto.so.1.1 => /lib/x86_64-linux-gnu/libcrypto.so.1.1 (0x00007f49bf7f3000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f49bf7d1000)
        libpcre.so.3 => /lib/x86_64-linux-gnu/libpcre.so.3 (0x00007f49bf75e000)
        libmariadb.so.3 => /lib/x86_64-linux-gnu/libmariadb.so.3 (0x00007f49bf70e000)
        libavformat.so.58 => /usr/local/lib/libavformat.so.58 (0x00007f49bf4aa000)
        libavcodec.so.58 => /usr/local/lib/libavcodec.so.58 (0x00007f49be203000)
        libavutil.so.56 => /usr/local/lib/libavutil.so.56 (0x00007f49bdf55000)
        libswscale.so.5 => /usr/local/lib/libswscale.so.5 (0x00007f49bdec2000)
        libswresample.so.3 => /usr/local/lib/libswresample.so.3 (0x00007f49bdea4000)
        libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f49bde9e000)
        libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f49bdccf000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f49bdb8b000)
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f49bdb71000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f49bd9ac000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f49c20a8000)
        libssl.so.1.1 => /lib/x86_64-linux-gnu/libssl.so.1.1 (0x00007f49bd919000)
        libmp3lame.so.0 => /lib/x86_64-linux-gnu/libmp3lame.so.0 (0x00007f49bd89f000)
        libopenjp2.so.7 => /lib/x86_64-linux-gnu/libopenjp2.so.7 (0x00007f49bd842000)
        libx264.so.155 => /lib/x86_64-linux-gnu/libx264.so.155 (0x00007f49bd584000)
        libx265.so.165 => /lib/x86_64-linux-gnu/libx265.so.165 (0x00007f49bc637000)
        libnuma.so.1 => /lib/x86_64-linux-gnu/libnuma.so.1 (0x00007f49bc629000)

$ apt list zoneminder
Listing... Done
zoneminder/bullseye,now 1.36.10-bullseye1 amd64 [installed]
zoneminder/bullseye 1.36.0-bullseye1 armhf
```

connortechnology, 2021-11-22:
1.36.11 on zmrepo links to h264.160.  No problems.

cyrildtm, 2021-11-23:
> 1.36.11 on zmrepo links to h264.160. No problems.

I don't understand this: 

```
$ sudo apt update
Hit:1 http://security.debian.org/debian-security bullseye-security InRelease
Hit:2 http://deb.debian.org/debian bullseye InRelease
Ign:3 https://zmrepo.zoneminder.com/debian/release-1.36 bullseye/ InRelease
Hit:4 https://zmrepo.zoneminder.com/debian/release-1.36 bullseye/ Release
Ign:5 https://developer.download.nvidia.com/compute/cuda/repos/debian11/x86_64  InRelease
Hit:6 https://developer.download.nvidia.com/compute/cuda/repos/debian11/x86_64  Release
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.

$ zmc -v
1.36.10

$ apt list --installed zoneminder
Listing... Done
zoneminder/bullseye,now 1.36.10-bullseye1 amd64 [installed]
N: There is 1 additional version. Please use the '-a' switch to see it

$ apt list --installed zoneminder -a
Listing... Done
zoneminder/bullseye,now 1.36.10-bullseye1 amd64 [installed]
zoneminder/stable 1.34.23-1 amd64

```

Looking at [the repo](https://zmrepo.zoneminder.com/debian/release-1.36/bullseye/) it seems like there isn't a amd64 deb file for 1.36.11. 

I'll have to wait for a while and confirm when the latest version arrives.

connortechnology, 2021-11-23:
crap you are right, I never built a package for amd64.  Only arm.  Building now.

cyrildtm, 2021-11-24:
> crap you are right, I never built a package for amd64. Only arm. Building now.

Thanks! So I tried again, and it's still not right: 

```
$ zmc -v
1.36.11

$ sudo apt remove libx264-155 libx265-165

$ sudo service zoneminder restart

$ ldd /bin/zmu
        linux-vdso.so.1 (0x00007ffd3cfe2000)
        libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f205e625000)
        libjpeg.so.62 => /lib/x86_64-linux-gnu/libjpeg.so.62 (0x00007f205e5a1000)
        libcrypto.so.1.1 => /lib/x86_64-linux-gnu/libcrypto.so.1.1 (0x00007f205e2ad000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f205e28b000)
        libpcre.so.3 => /lib/x86_64-linux-gnu/libpcre.so.3 (0x00007f205e218000)
        libmariadb.so.3 => /lib/x86_64-linux-gnu/libmariadb.so.3 (0x00007f205e1c8000)
        libavformat.so.58 => /usr/local/lib/libavformat.so.58 (0x00007f205df64000)
        libavcodec.so.58 => /usr/local/lib/libavcodec.so.58 (0x00007f205ccbd000)
        libavutil.so.56 => /usr/local/lib/libavutil.so.56 (0x00007f205ca0f000)
        libswscale.so.5 => /usr/local/lib/libswscale.so.5 (0x00007f205c97c000)
        libswresample.so.3 => /usr/local/lib/libswresample.so.3 (0x00007f205c95e000)
        libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f205c958000)
        libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f205c789000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f205c645000)
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f205c62b000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f205c466000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f2060b62000)
        libssl.so.1.1 => /lib/x86_64-linux-gnu/libssl.so.1.1 (0x00007f205c3d3000)
        libmp3lame.so.0 => /lib/x86_64-linux-gnu/libmp3lame.so.0 (0x00007f205c359000)
        libopenjp2.so.7 => /lib/x86_64-linux-gnu/libopenjp2.so.7 (0x00007f205c2fc000)
        libx264.so.155 => not found
        libx265.so.165 => not found

$ zmc -v
zmc: error while loading shared libraries: libx264.so.155: cannot open shared object file: No such file or directory

$ sudo apt install libx264-155 libx265-165

$ zmc -v
1.36.11

$ sudo service zoneminder restart

$ ldd /bin/zmu
        linux-vdso.so.1 (0x00007ffcf830b000)
        libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f346204b000)
        libjpeg.so.62 => /lib/x86_64-linux-gnu/libjpeg.so.62 (0x00007f3461fc7000)
        libcrypto.so.1.1 => /lib/x86_64-linux-gnu/libcrypto.so.1.1 (0x00007f3461cd3000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f3461cb1000)
        libpcre.so.3 => /lib/x86_64-linux-gnu/libpcre.so.3 (0x00007f3461c3e000)
        libmariadb.so.3 => /lib/x86_64-linux-gnu/libmariadb.so.3 (0x00007f3461bee000)
        libavformat.so.58 => /usr/local/lib/libavformat.so.58 (0x00007f346198a000)
        libavcodec.so.58 => /usr/local/lib/libavcodec.so.58 (0x00007f34606e3000)
        libavutil.so.56 => /usr/local/lib/libavutil.so.56 (0x00007f3460435000)
        libswscale.so.5 => /usr/local/lib/libswscale.so.5 (0x00007f34603a2000)
        libswresample.so.3 => /usr/local/lib/libswresample.so.3 (0x00007f3460384000)
        libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f346037e000)
        libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f34601af000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f346006b000)
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f3460051000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f345fe8c000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f3464588000)
        libssl.so.1.1 => /lib/x86_64-linux-gnu/libssl.so.1.1 (0x00007f345fdf9000)
        libmp3lame.so.0 => /lib/x86_64-linux-gnu/libmp3lame.so.0 (0x00007f345fd7f000)
        libopenjp2.so.7 => /lib/x86_64-linux-gnu/libopenjp2.so.7 (0x00007f345fd22000)
        libx264.so.155 => /lib/x86_64-linux-gnu/libx264.so.155 (0x00007f345fa64000)
        libx265.so.165 => /lib/x86_64-linux-gnu/libx265.so.165 (0x00007f345eb17000)
        libnuma.so.1 => /lib/x86_64-linux-gnu/libnuma.so.1 (0x00007f345eb09000)

```

I'm not sure if a reboot is necessary, but it seems to me like the latest version (1.36.11) is still using x264-155 and x265-165.

MarkBarbieri, 2021-11-24:
For what it's worth, I'm not seeing this issue on Bullseye

I get **libx264.so.160 => /lib/x86_64-linux-gnu/libx264.so.160 (0x00007f52000cb000)**

Try installing package libx264-dev

```
mark@zoneminder:~$ ldd /bin/zmu
	linux-vdso.so.1 (0x00007fff38ea4000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f5205a57000)
	libjpeg.so.62 => /lib/x86_64-linux-gnu/libjpeg.so.62 (0x00007f52059d3000)
	libcrypto.so.1.1 => /lib/x86_64-linux-gnu/libcrypto.so.1.1 (0x00007f52056df000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f52056bd000)
	libpcre.so.3 => /lib/x86_64-linux-gnu/libpcre.so.3 (0x00007f520564a000)
	libmariadb.so.3 => /lib/x86_64-linux-gnu/libmariadb.so.3 (0x00007f52055fa000)
	libavformat.so.58 => /lib/x86_64-linux-gnu/libavformat.so.58 (0x00007f5205373000)
	libavcodec.so.58 => /lib/x86_64-linux-gnu/libavcodec.so.58 (0x00007f5203dbb000)
	libavutil.so.56 => /lib/x86_64-linux-gnu/libavutil.so.56 (0x00007f5203afa000)
	libswscale.so.5 => /lib/x86_64-linux-gnu/libswscale.so.5 (0x00007f5203a61000)
	libswresample.so.3 => /lib/x86_64-linux-gnu/libswresample.so.3 (0x00007f5203a3f000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f5203a39000)
	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f520386a000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f5203726000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f520370c000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f5203547000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f5207f99000)
	libssl.so.1.1 => /lib/x86_64-linux-gnu/libssl.so.1.1 (0x00007f52034b4000)
	libxml2.so.2 => /lib/x86_64-linux-gnu/libxml2.so.2 (0x00007f5203306000)
	libbz2.so.1.0 => /lib/x86_64-linux-gnu/libbz2.so.1.0 (0x00007f52032f1000)
	libgme.so.0 => /lib/x86_64-linux-gnu/libgme.so.0 (0x00007f52032a2000)
	libopenmpt.so.0 => /lib/x86_64-linux-gnu/libopenmpt.so.0 (0x00007f52030b8000)
	libchromaprint.so.1 => /lib/x86_64-linux-gnu/libchromaprint.so.1 (0x00007f52030a2000)
	libbluray.so.2 => /lib/x86_64-linux-gnu/libbluray.so.2 (0x00007f5203057000)
	libgnutls.so.30 => /lib/x86_64-linux-gnu/libgnutls.so.30 (0x00007f5202e57000)
	librabbitmq.so.4 => /lib/x86_64-linux-gnu/librabbitmq.so.4 (0x00007f5202e3d000)
	libsrt-gnutls.so.1.4 => /lib/x86_64-linux-gnu/libsrt-gnutls.so.1.4 (0x00007f5202d7c000)
	libssh-gcrypt.so.4 => /lib/x86_64-linux-gnu/libssh-gcrypt.so.4 (0x00007f5202cee000)
	libzmq.so.5 => /lib/x86_64-linux-gnu/libzmq.so.5 (0x00007f5202c31000)
	libvpx.so.6 => /lib/x86_64-linux-gnu/libvpx.so.6 (0x00007f5202a16000)
	libwebpmux.so.3 => /lib/x86_64-linux-gnu/libwebpmux.so.3 (0x00007f5202a0a000)
	libwebp.so.6 => /lib/x86_64-linux-gnu/libwebp.so.6 (0x00007f520299f000)
	liblzma.so.5 => /lib/x86_64-linux-gnu/liblzma.so.5 (0x00007f5202977000)
	libdav1d.so.4 => /lib/x86_64-linux-gnu/libdav1d.so.4 (0x00007f5202843000)
	librsvg-2.so.2 => /lib/x86_64-linux-gnu/librsvg-2.so.2 (0x00007f5201e45000)
	libgobject-2.0.so.0 => /lib/x86_64-linux-gnu/libgobject-2.0.so.0 (0x00007f5201deb000)
	libglib-2.0.so.0 => /lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007f5201cbc000)
	libcairo.so.2 => /lib/x86_64-linux-gnu/libcairo.so.2 (0x00007f5201b95000)
	libzvbi.so.0 => /lib/x86_64-linux-gnu/libzvbi.so.0 (0x00007f5201b05000)
	libsnappy.so.1 => /lib/x86_64-linux-gnu/libsnappy.so.1 (0x00007f5201afa000)
	libaom.so.0 => /lib/x86_64-linux-gnu/libaom.so.0 (0x00007f5201662000)
	libcodec2.so.0.9 => /lib/x86_64-linux-gnu/libcodec2.so.0.9 (0x00007f520087d000)
	libgsm.so.1 => /lib/x86_64-linux-gnu/libgsm.so.1 (0x00007f520086d000)
	libmp3lame.so.0 => /lib/x86_64-linux-gnu/libmp3lame.so.0 (0x00007f52007f3000)
	libopenjp2.so.7 => /lib/x86_64-linux-gnu/libopenjp2.so.7 (0x00007f5200796000)
	libopus.so.0 => /lib/x86_64-linux-gnu/libopus.so.0 (0x00007f520073b000)
	libshine.so.3 => /lib/x86_64-linux-gnu/libshine.so.3 (0x00007f5200530000)
	libspeex.so.1 => /lib/x86_64-linux-gnu/libspeex.so.1 (0x00007f5200515000)
	libtheoraenc.so.1 => /lib/x86_64-linux-gnu/libtheoraenc.so.1 (0x00007f52004d8000)
	libtheoradec.so.1 => /lib/x86_64-linux-gnu/libtheoradec.so.1 (0x00007f52004b6000)
	libtwolame.so.0 => /lib/x86_64-linux-gnu/libtwolame.so.0 (0x00007f5200491000)
	libvorbis.so.0 => /lib/x86_64-linux-gnu/libvorbis.so.0 (0x00007f5200464000)
	libvorbisenc.so.2 => /lib/x86_64-linux-gnu/libvorbisenc.so.2 (0x00007f52003b9000)
	libwavpack.so.1 => /lib/x86_64-linux-gnu/libwavpack.so.1 (0x00007f520038d000)
	libx264.so.160 => /lib/x86_64-linux-gnu/libx264.so.160 (0x00007f52000cb000)
```

cyrildtm, 2021-11-24:
> 

Thanks @MarkBarbieri ! That's really weird, you don't even have x265 dependency. 

And I do have the dev package installed: 

```
$ apt list --installed libx264*
Listing... Done
libx264-155/oldstable,now 2:0.155.2917+git0a84d98-2 amd64 [installed]
libx264-160/stable,now 2:0.160.3011+gitcde9a93-2.1 amd64 [installed,automatic]
libx264-dev/stable,now 2:0.160.3011+gitcde9a93-2.1 amd64 [installed]

```

Do you have CUDA? AFAIK nvidia moved to bullseye a short while ago, that was when I moved fully to bullseye (after seeing one post reporting zm works on bullseye, before official support was added)

MarkBarbieri, 2021-11-24:
I'm not using x265, so I expect it's fine without it.

No CUDA, all in software for me.  Not enough slots in the server for another card!

cyrildtm, 2021-11-24:
Mine is different for some reason. Without the versions being asked for, zmc will crash. I don't use h265 either, but the library has to be there or it will crash and cannot connect to camera (h264 stream). 
Is it possible that during the first time installing zoneminder, the libraries are selected? I never tried a clean install with different libraries.

cyrildtm, 2021-11-24:
Finally this rings a bell. I think I have been using my own ffmpeg build with CUDA. Let me find some time and build a new one using new platforms and see if it's resolved.

cyrildtm, 2021-11-24:
Yep confirmed, it's my ffmpeg build being outdated. 
I recompiled ffmpeg on bullseye (with libx264-160-dev and no libx265), and zmu links to 160 now. 
Thanks! @connortechnology and @MarkBarbieri

closed 2021-11-24 by cyrildtm