ZoneMinder/zoneminder #3368: streaming via nph-zms stops after 1-4secs on 1.36.7

PVi1, 2021-10-06:
- Version of ZoneMinder [1.36.7 ]
- How you installed ZoneMinder [from iconner ppa, upgraded from 1.34.21]
- Full name and version of OS - Ubuntu 18.04.6 LTS Bionic,  ZM multiserver setup with 2x server, one of them running DB, shared storage.
- Browser name and version Firefox 92

**Describe the bug**
Streaming via` $zmhostname/zm/cgi-bin/nph-zms?mode=jpeg&monitor=34&buffer=1000&user=usernamex&pass=passwordy&scale=30&maxfps=5` stops after 1-4 secs.

**To Reproduce**
Steps to reproduce the behavior:
1. Upgraded from 1.43.21 to 1.36.7
2. Open stream via nph-zms similar to the link above. **Do not user Zoneminder gui, because event monitor in zm restarts streams automatically, so the stream looks fine.**
3. Page will be loaded and stream ends in several seconds.

**Expected behavior**
Stream should continue uninterrupted for the longest possible time.

**Debug Logs**
```
...
10/06/21 11:58:39.286003 zms_m4[3066].DB1-zm_utils.cpp/204 [Detected a x86\x86-64 processor with SSE4.2]
10/06/21 11:58:39.286020 zms_m4[3066].DB4-zm_image.cpp/390 [Blend: Using SSE2 fast blend function]
10/06/21 11:58:39.286027 zms_m4[3066].DB4-zm_image.cpp/452 [Delta: Using SSSE3 delta functions]
10/06/21 11:58:39.286033 zms_m4[3066].DB4-zm_image.cpp/553 [Deinterlace: Using standard functions]
10/06/21 11:58:39.286038 zms_m4[3066].DB4-zm_image.cpp/566 [Image buffer copy: Using standard memcpy]
10/06/21 11:58:39.286391 zms_m4[3066].DB2-zm_stream.h/217 [Setting replay_rate 100]
10/06/21 11:58:39.287195 zms_m4[3066].DB5-zm_db.cpp/165 [Success]
10/06/21 11:58:39.287231 zms_m4[3066].DB1-zm_storage.cpp/91 [No id passed to Storage constructor.  Using default path /var/cache/zoneminder/events instead]
10/06/21 11:58:39.287250 zms_m4[3066].DB1-zm_monitor.cpp/482 [Have camera type Remote]
10/06/21 11:58:39.287255 zms_m4[3066].DB1-zm_monitor.cpp/2322 [Reloading linked monitors for monitor kam4_prijazd, '(null)']
10/06/21 11:58:39.287266 zms_m4[3066].DB1-zm_packetqueue.cpp/662 [Setting pre_event_video_packet_count to 16]
10/06/21 11:58:39.287271 zms_m4[3066].DB1-zm_packetqueue.cpp/656 [Setting max_video_packet_count to 0]
10/06/21 11:58:39.287276 zms_m4[3066].DB1-zm_monitor.cpp/644 [mem.size(8) SharedData=760 TriggerData=560 VideoStoreData=4128 timestamps=48 images=3x3072000 = 9216000 total=9221560]
10/06/21 11:58:39.287283 zms_m4[3066].DB1-zm_monitor.cpp/682 [Loaded monitor 4(kam4_prijazd), 0 zones]
10/06/21 11:58:39.287290 zms_m4[3066].DB3-zm_monitor.cpp/906 [Connecting to monitor.  Purpose is 0]
10/06/21 11:58:39.287316 zms_m4[3066].ERR-zm_monitor.cpp/916 [Can't open memory map file /dev/shm/zm.mmap.4: No such file or directory]
10/06/21 11:58:39.287322 zms_m4[3066].ERR-zm_stream.cpp/58 [Unable to connect to monitor id 4 for streaming]
10/06/21 11:58:39.287327 zms_m4[3066].DB1-zm_monitor.cpp/1062 [Already disconnected]
10/06/21 11:58:39.287332 zms_m4[3066].ERR-zms.cpp/268 [Unable set start stream for monitor 4]
10/06/21 11:58:39.287337 zms_m4[3066].DB2-zm_stream.cpp/270 [Sending 1280x800 * 50 text frame 'Unable to connect to monitor']
10/06/21 11:58:39.293352 undef[3066].DB1-zm_logger.cpp/232 [LogOpts: level=DB9 effective=DB9, screen=OFF, database=OFF, logfile=DB9->/storage/tmp/zm/zm_debug.log.03066, syslog=FAT]
10/06/21 11:58:39.293398 undef[3066].DB4-zm_packetqueue.cpp/70 [Done in destructor]
```
web_php.log is full of:

```
10/19/17 16:26:53.816094 web_php[8535].ERR [Socket /storage/tmp/zms-190638s.sock does not exist.  This file is created by zms, and since it does not exist, either zms did not run, or zms exited early.  Please check your zms logs and ensure that CGI is enabled in apache and check that the PATH_ZMS is set correctly.  Make sure that ZM is actually recording.  If you are trying to view a live stream and the capture process (zmc) is not running then zms will exit. Please go to http://zoneminder.readthedocs.io/en/latest/faq.html#why-can-t-i-see-streamed-images-when-i-can-see-stills-in-the-zone-window-etc for more information.] at /usr/share/zoneminder/www/includes/functions.php line 2033

```

[zm_debug.log](https://github.com/ZoneMinder/zoneminder/files/7293853/zm_debug.log)

connortechnology, 2021-10-06:
The issue is not in zms.  It is in zmc.  For some reason the capture process is dying.  We will need the logs from it to determine why.

PVi1, 2021-10-06:
> The issue is not in zms. It is in zmc. For some reason the capture process is dying. We will need the logs from it to determine why.
Hello  @connortechnology,
thank you for the quick reply.
I have edited the original post and attached zmc from that camera. 

Regards,

Peter

jaapf4fs, 2022-02-11:
It's most likely because the rate in the :
/zm/cgi-bin/nph-zms?mode=jpeg&frame=1&scale=100&**rate=1**&maxfps=5&replay=none&source=event&event=1729&connkey=248364&rand=1644588761
Is in Firefox 1 (rate=1)!

When you watch the same event in like Google-chrome the same page (event) does play because the rate is 100..
/zm/cgi-bin/nph-zms?mode=jpeg&frame=1&scale=100&**rate=100**&maxfps=5&replay=none&source=event&event=1729&connkey=334776&rand=1644589172

When you change the rate manually from 1 to 100 the video does play...

I hope someone knows where this comes from without me having a multiple days search : -)

Take care,
Jaap

connortechnology, 2022-02-11:
Ooh neat, good hunting.  I'll take a look today.

jaapf4fs, 2022-02-13:
I have searched in the Monitor table and I don't know how related it is but the Default rate of all monitors is 100.
But somehow  the player in the event webpage is showing 0.01 at the controls below the vid for jpeg events and 1 under the video events.
Mode: Replay Rate: 0.01x Progress: 0s Zoom: 1x 
Mode: Replay Rate: 1x Progress: 6s Zoom: 1x 
I think there is something wrong with the calculation of the frame rate it looks like it's twice divided by 100....????
Best regards,
Jaap