ZoneMinder/zoneminder #3375: nph-zms does not terminate and continues to use CPU

gcormier, 2021-10-18:
**Describe Your Environment**
- 1.36.8
- docker
- ubuntu 20.04.3LTS
- Google Chrome

**Describe the bug**
`nph-zms` does not terminate after watching a video event.

**To Reproduce**
Steps to reproduce the behavior:
1. Restart zoneminder and launch `htop` or other process monitor
2. Open zoneminder web console
3. Click on one of the events and then watch the video
4. Browse to another site or close the tab or go back
5. Process does not terminate
6. Repeate step 3 and observe multiple instances begin to stack up

**Expected behavior**
Process should terminate and no longer consume CPU.

[image]

freexmonster, 2021-11-01:
wtf cpu 100% v1.36.7 Debian 11 Hyper-V( no docker) [image]

Carbenium, 2021-11-01:
@freexmonster This report is specifically about `zms` and not `zmc`. Additionally, I'd say that CPU load is to be expected with the amount of cameras you try to run on 2 cores.

jangrewe, 2021-11-03:
I'm not using any Docker containers, but just bare-metal ZM & ZM-ES, and i experience the same issue. There's always 1-2 `nph-zms` running, consuming ~25% CPU (each). No web console or ZMNinja - or any other client - running.

maverick214, 2021-11-15:
Have the same, running ZM plugin in Truenas 12.0-U6 each time I view a thumbnail it opens a new ZMS process.  Turning off the option WEB_ANIMATE_THUMBS will stop it.

mcgman, 2021-11-18:
same issue here ZM plugin in Truenas.  Live video feed works fine in that clicking back or closing the tab kills the zms process.  However, when viewing an event video, zms is spawned but never exits.

connortechnology, 2022-06-01:
I believe this is fixed.  Please update to 1.36.17 or higher and report.

daveymg-nz, 2022-07-10:
Using Zoneminder-containers/zoneminder-base docker image with Zoneminder 1.36.19 I can report this issue still exists. Disabling WEB_ANIMATE_THUMBS does not appear to affect the issue. I can reproduce by replaying events and clicking on the "next event" button before the currently playing event has completed. Doing this repeatedly results in more and more nph-zms processes existing and gradual depletion of free ram.

pgrandin, 2022-08-01:
I can confirm this issue on `v1.36.21` from the docker image here: https://github.com/zoneminder-containers/zoneminder-base/pkgs/container/zoneminder-base/29421610?tag=amd64-nightly-822

crazy28, 2022-08-06:
Yea it’s happening to my setup.

debian 11 
Nginx 
Php fpm 7.4
Zoneminder v1.36.21

Not closing the zms after use

SmokeyBR, 2022-08-26:
> Is suspect this is specific to the dlandon docker. It does not happen here on bare ubuntu. You could try posting debug logs, but I suspect that the docker is using a different cgi method, like php-fpm and it is not noticing the dropped connection. Not sure how we can fix that ZM side.

this happens with me on ubuntu20.04 nothing under docker but i do use nginx with php-fpm, i noticed happening since 1.36 never figure it out how to reproduce thou, sometimes it happens sometimes doesn't, seems more likely to happen using zmninja app although cant say for sure

waynieack, 2022-11-09:
Version of ZoneMinder 1.37.25
    How you installed ZoneMinder PPA: deb-src http://ppa.launchpad.net/iconnor/zoneminder-master/ubuntu focal main
    Ubuntu 20.04.5 LTS
    apache2 2.4.41
    Not using php-fpm

I also see this issue, it happens when I replay recorded events as described by @daveymg-nz.

waynieack, 2022-12-23:
I finally did some digging on this and found that there's 2 issues here. 

1. When you watch a recorded video and click the next button, the nph-zms process for the last event is left running. 

     For this issue I noticed that the apache connection was listed as tcp6 for ipv4 connections in netstat:
    ` tcp6      0      0 192.168.99.35:80       192.168.99.179:60448   ESTABLISHED 37662/apache2`

    To resolve this, I disabled ipv6 in the apache config by changing all the "Listen 80" and "Listen 443" to "Listen 0.0.0.0:80" and "Listen 0.0.0.0:443", then restarting apache. So far, I don't see nph-zms processes continuing to run after going to the next event. No idea what the actual issue here is but I don't use ipv6 so I don't care to have it enabled. 
    
    
2. When you watch an event and delete it, the nph-zms process hangs and uses 100% cpu. I also see errors like this "Can't open /mnt/ext-video/events/5/2022-12-23/14326436/086-capture.jpg: No such file or directory" because it seems that the event is deleted while the nph-zms process is accessing it. 

  For this issue I added "streamReq({command: CMD_QUIT});" in "function manageDelConfirmModalBtns()" in file "www/skins/classic/views/js/event.js" which seems to stop nph-zms before deleting the event. 

This may not be the right way to fix this because I am getting an error like the below after the delete the event but at least the process isn't hung using 100% CPU anymore.
`Can't send /mnt/ext-video/events/7/2022-12-23/14326377/110-capture.jpg: Broken pipe`
  
**Below is the code update:**

**File:** www/skins/classic/views/js/event.js

867 function manageDelConfirmModalBtns() {
 868   document.getElementById("delConfirmBtn").addEventListener("click", function onDelConfirmClick(evt) {
 869     if (!canEdit.Events) {
 870       enoperm();
 871       return;
 872     }
 873
 874     pauseClicked();
 875     evt.preventDefault();
 876     **streamReq({command: CMD_QUIT});**
 877     $j.getJSON(thisUrl + '?request=event&action=delete&id='+eventData.Id)

connortechnology, 2022-12-25:
Some great insights here.  We definitely need to stop streaming before delete.  I will look at getting that code in.
It's kindof a bigger problem though, someone else could delete the event while we are viewing it.  We almost need to lock the db record while viewing in order to prevent any such action.  

Also, zms shouldn't consume 100%... if it can't find a jpeg, it should just sleep and try to send the next as per usual... and if we are replaying a single event.

closed 2023-06-22 by gcormier