ZoneMinder/zoneminder #3382: H265 issues

HaoLVb, 2021-11-09:
Describe Your Environment
Version of ZoneMinder : 1.36.10
How you installed ZoneMinder : from source
Full name and version of OS : Ubuntu 20.04

Describe the bug:
Cameras with H.265 codec often get blurry image even though the transport is TCP, this doesn't happen with the H.264 codec.
[image]

jwhite, 2022-01-24:
I'm seeing the same behavior with a EzViz C3W Pro.  Have 2 other cameras working well (h264)

I'm wondering if there is a trick to get the camera to ignore a MASSIVE movement, so as to not be triggered by the garbage frames.

connortechnology, 2022-01-24:
I think this issue is referring to zm rtsp server where we are acting as an rtsp proxy.

romale, 2023-08-31:
I've this issue too
h.265
zm v1.36.33

[image]