ZoneMinder/zoneminder #3384: zoneminder: depends on obsolete pcre3 library

onlyjob, 2021-11-19:
As reported in https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=999919, libPCRE3 is obsolete:

> Your package still depends on the old, obsolete PCRE3[0] libraries
(i.e. libpcre3-dev). This has been end of life for a while now, and
upstream do not intend to fix any further bugs in it. Accordingly, I
would like to remove the pcre3 libraries from Debian, preferably in
time for the release of Bookworm.

> The newer PCRE2 library was first released in 2015, and has been in
Debian since stretch. Upstream's documentation for PCRE2 is available
here: https://pcre.org/current/doc/html/

> Many large projects that use PCRE have made the switch now (e.g. git,
php); it does involve some work, but we are now at the stage where
PCRE3 should not be used, particularly if it might ever be exposed to
untrusted input.

> Historical reasons mean that old PCRE is packaged as pcre3 in Debian 

Please consider migrating to _libpcre2-dev_.

Please note that `libpolkit-gobject-1-dev` indirectly depends on obsolete `libpcre3-dev` through `libglib2.0-dev` on Debian so one have to build Zoneminder without Polkit to avoid installing both PCRE libraries simultaneously.

Currently Zoneminder fails to detect `libpcre2-dev` on build-time.

connortechnology, 2022-09-25:
I could have sworn I responded to this...

We have alternate code.  Let's just deprecate the use of the PCRE case.  pass -DZM_NO_PCRE=ON  to cmake.  We have been building our packages with this for some time.  It

david-geiger, 2023-04-18:
Hi,

Is a switch to pcre2 finally planned?

Regards,
David

connortechnology, 2023-04-18:
No plans. No one funding the work.

onlyjob, 2025-02-21:
A patch was contributed in Debian: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=999919#20

connortechnology, 2025-02-21:
@onlyjob we had merged that patch but it caused build failures.  I wasn't able to quickly determine why and the submitter didn't follow up.  Would be happy to try again if anyone is willing to look at it.

Also, PCRE can just be turned off.  We have non PCRE code that works just fine.

onlyjob, 2025-02-22:
I'm surprised to read about any problems. I have [applied this patch](https://salsa.debian.org/debian/zoneminder/-/commit/bb5bd53c556306a7c26a39852bef820a6d4db634) and uploaded it with current release 1.36.35 to Debian where it builds successfully, with that very patch [on all architectures](https://buildd.debian.org/status/package.php?p=zoneminder), as well as on CI infrastructure.

connortechnology, 2025-02-22:
The issue might have been 1.37 specific, or unrelated... thing is the author didn't respond further and my todo list is long.  Happy to take another look.  I'd like to release 1.36.36 by end of month so can include it possibly.

connortechnology, 2025-02-22:
Also, as that patch is against 1.36 it isn't the same patch that I was referring to, Similar I'm sure, but not the same.

onlyjob, 2025-02-24:
So would you be able to apply it and try again? Would you like a merge request?

connortechnology, 2025-02-24:
yes please

onlyjob, 2025-02-26:
See #4230. Adaptation for `master` was tedious because due to code style and formatting changes original patch did not apply... 
Fortunately only code formatting changed in `master` so I was able to adapt the patch.

commit 9634d663 (2025-02-26): Port to PCRE2 (Closes: #3384).

Bug-Debian: https://bugs.debian.org/999919
Bug: https://github.com/ZoneMinder/zoneminder/issues/3384
PR #4230 (2025-02-26): pcre3 --> pcre2 adaptation
This is minor adaptation of Yavor Doganov's patch from Debian, to accommodate code style (formatting) changes in `master`.  
(Closes: #3384)
closed 2025-02-26 by connortechnology