ZoneMinder/zoneminder #3390: 1.36.11 missing events and slow ui

MoHf7f4, 2021-11-28:
**Describe Your Environment**
Version of ZoneMinder 1.36.11 (wasn't happening in 1.36.10)
How you installed ZoneMinder PPA
Full name and version of OS Description:    Ubuntu 18.04.6 LTS
Browser name and version  96.0.4664.45

**Describe the bug**
Using previous or next, in view get error "Event was not found."
The list of events has no issues, clicking on them individually there are no issues.

Also, used to be able to click previous and next much faster.  Now there is a few second delay before it appears to be allowed.

**Expected behavior**
Get the next or previous event video

**Debug Logs**
```
<!--
getCmdResponse stream error: Socket /run/zm/zms-198158s.sock does not exist. This file is created by zms, and since it does not exist, either zms did not run, or zms exited early. Please check your zms logs and ensure that CGI is enabled in apache and check that the PATH_ZMS is set correctly. Make sure that ZM is actually recording. If you are trying to view a live stream and the capture process (zmc) is not running then zms will exit. Please go to http://zoneminder.readthedocs.io/en/latest/faq.html#why-can-t-i-see-streamed-images-when-i-can-see-stills-in-the-zone-window-etc for more information. - checkStreamForErrors()

<html>
<body>
<!--StartFragment-->

?view=event&eid=0&filter%5BQuery%5D%5Bterms%5D%5B0%5D%5Battr%5D=MonitorId&filter%5BQuery%5D%5Bterms%5D%5B0%5D%5Bop%5D=%3D&filter%5BQuery%5D%5Bterms%5D%5B0%5D%5Bval%5D=16&filter%5BQuery%5D%5Bsort_asc%5D=1&filter%5BQuery%5D%5Bsort_field%5D=StartDateTime&fil
--
-->

```

alabamatoy, 2021-12-14:
I was unable to repro this behavior, mine seems to work correctly.  eid should be the event number, not zero, I believe.

MoHf7f4, 2021-12-14:
Is there any way I can pull more from my side to help repro this?

alabamatoy, 2021-12-14:
Check webserver and DB logs for any correlating info.  Probably will require help from one of the more capable admins than me.

connortechnology, 2021-12-14:
I can't reproduce it either.
The buttons are populated using a background ajax call.  So if the db is slow or something it can take a while to send back the previous and next event ids.  If we right click -> Inspect -> Network you should be able to see this ajax called (will have an XHR type going to ?view=request&request=status&entity=nearevents. You can see what is returned in the response tab.  

In any event we have code that says to do something special when NextEvent==0.  Since this isn't firing and instead we are being directed to event 0, I think what is happening is that NextEvent == "0".  So in master I will add a parseInt and maybe that will fix it.

I'd be interested to know what was returned from that ajax call though.

MoHf7f4, 2021-12-14:
I'll see if I can capture that ajax call.

Much appreciated 👍

commit e8edeab6 (2021-12-15): parseInt prevEventId and nextEventID to be sure that our tests for != 0 work.  Also don't go to previous event if prevEventId ==0.Maybe Fixes #3390