ZoneMinder/zoneminder #3393: Linked monitors not recording when parent goes into alert

maddios, 2021-12-01:
**Describe Your Environment**
- Version of ZoneMinder: 1.36.10
- How you installed ZoneMinder: from-source
- Full name and version of OS: Ubuntu 18.04 LTS
- Server is a 24 core dell with 64GB ram, my whole setup runs at about 10 load average while using approx 30GB ram.

I'm running 5 Amcrest IP cameras each outputting 2 streams (low and high rez), I'm actually doing my triggering in reverse though, I have my high res streams running MoDect and the low res streams actually go to a ML object detection/motion tracking software I wrote which then outputs a stream which I'm also streaming into zoneminder, These 5 AI streams are as NoDect. Each of these 5 streams has its full res stream set as a Linked monitor.

**Describe the bug**
The issue I'm having is that MOST of the time my linked monitors don't record anything, I've tried restarting ZM and triggering an alert once ZM is back up and it'll sometimes trigger and sometimes not. As far as I can tell there's nothing in the log files to indicate any kind of issue even with debug mode enabled. I've check the zmc_m{id}.log files and the zmwatch.log files.

**To Reproduce**
I have no idea how to reproduce this other than it happens in my env all the time.

**Expected behavior**
I expect that every single time an alert is triggered all linked cameras must record it

**Debug Logs**

This is the initiating stream
```

12/01/21 18:42:55.137821 zmc_m2[33045].INF-zm_monitor.cpp/1675 [front door: 1000 - Capturing at 8.78 fps, capturing bandwidth 976795bytes/sec]
12/01/21 18:42:55.194924 zmc_m2[33159].INF-zm_monitor.cpp/1718 [front door: 1000 - Analysing at 8.70 fps from 999 - 989=10 / 1638402175.194921 - 1638402174.045736 = 1.149185]
12/01/21 18:43:35.909086 zmc_m2[33159].INF-zm_monitor.cpp/2132 [front door: 1405 - Gone into prealarm state]
12/01/21 18:43:36.111011 zmc_m2[33159].INF-zm_monitor.cpp/2132 [front door: 1407 - Gone into prealarm state]
12/01/21 18:43:37.428903 zmc_m2[33159].INF-zm_monitor.cpp/2132 [front door: 1421 - Gone into prealarm state]
12/01/21 18:44:06.765441 zmc_m2[33159].INF-zm_monitor.cpp/2132 [front door: 1713 - Gone into prealarm state]
12/01/21 18:44:07.736896 zmc_m2[33159].INF-zm_monitor.cpp/2077 [front door: 1723 - Gone into alarm state PreAlarmCount: 9 > AlarmFrameCount:10 Cause:Forced Web]
12/01/21 18:44:07.752811 zmc_m2[33159].INF-zm_monitor.cpp/2123 [front door: 1722 - Opening new event 693466, alarm start]
12/01/21 18:44:12.747340 zmc_m2[33159].INF-zm_monitor.cpp/2157 [front door: 1772 - Gone into alert state]
12/01/21 18:44:13.684522 zmc_m2[33159].INF-zm_monitor.cpp/2138 [front door: 1782 - Alarmed frame while in alert state. Consecutive alarmed frames left to return to alarm state: 009]
12/01/21 18:44:13.795669 zmc_m2[33159].INF-zm_monitor.cpp/2138 [front door: 1783 - Alarmed frame while in alert state. Consecutive alarmed frames left to return to alarm state: 008]
12/01/21 18:44:13.882419 zmc_m2[33159].INF-zm_monitor.cpp/2138 [front door: 1784 - Alarmed frame while in alert state. Consecutive alarmed frames left to return to alarm state: 007]
12/01/21 18:44:14.254587 zmc_m2[33159].INF-zm_monitor.cpp/2138 [front door: 1787 - Alarmed frame while in alert state. Consecutive alarmed frames left to return to alarm state: 009]
12/01/21 18:44:14.357455 zmc_m2[33159].INF-zm_monitor.cpp/2138 [front door: 1788 - Alarmed frame while in alert state. Consecutive alarmed frames left to return to alarm state: 008]
12/01/21 18:44:14.523994 zmc_m2[33159].INF-zm_monitor.cpp/2138 [front door: 1789 - Alarmed frame while in alert state. Consecutive alarmed frames left to return to alarm state: 007]
12/01/21 18:44:14.851556 zmc_m2[33159].INF-zm_monitor.cpp/2138 [front door: 1793 - Alarmed frame while in alert state. Consecutive alarmed frames left to return to alarm state: 009]
12/01/21 18:44:14.988704 zmc_m2[33159].INF-zm_monitor.cpp/2138 [front door: 1794 - Alarmed frame while in alert state. Consecutive alarmed frames left to return to alarm state: 008]
12/01/21 18:44:15.089600 zmc_m2[33159].INF-zm_monitor.cpp/2138 [front door: 1795 - Alarmed frame while in alert state. Consecutive alarmed frames left to return to alarm state: 007]
12/01/21 18:44:15.186858 zmc_m2[33159].INF-zm_monitor.cpp/2138 [front door: 1796 - Alarmed frame while in alert state. Consecutive alarmed frames left to return to alarm state: 006]
12/01/21 18:44:17.578609 zmc_m2[33159].INF-zm_monitor.cpp/2138 [front door: 1819 - Alarmed frame while in alert state. Consecutive alarmed frames left to return to alarm state: 009]
12/01/21 18:44:17.718690 zmc_m2[33159].INF-zm_monitor.cpp/2138 [front door: 1820 - Alarmed frame while in alert state. Consecutive alarmed frames left to return to alarm state: 008]
12/01/21 18:44:17.757718 zmc_m2[33159].INF-zm_monitor.cpp/2138 [front door: 1821 - Alarmed frame while in alert state. Consecutive alarmed frames left to return to alarm state: 007]
12/01/21 18:44:17.794101 zmc_m2[33159].INF-zm_monitor.cpp/2138 [front door: 1822 - Alarmed frame while in alert state. Consecutive alarmed frames left to return to alarm state: 006]
12/01/21 18:44:17.836006 zmc_m2[33159].INF-zm_monitor.cpp/2138 [front door: 1823 - Alarmed frame while in alert state. Consecutive alarmed frames left to return to alarm state: 005]
12/01/21 18:44:17.966155 zmc_m2[33159].INF-zm_monitor.cpp/2138 [front door: 1824 - Alarmed frame while in alert state. Consecutive alarmed frames left to return to alarm state: 004]
12/01/21 18:44:18.072897 zmc_m2[33159].INF-zm_monitor.cpp/2138 [front door: 1825 - Alarmed frame while in alert state. Consecutive alarmed frames left to return to alarm state: 003]
12/01/21 18:44:18.199612 zmc_m2[33159].INF-zm_monitor.cpp/2138 [front door: 1826 - Alarmed frame while in alert state. Consecutive alarmed frames left to return to alarm state: 002]
12/01/21 18:44:20.679779 zmc_m2[33159].INF-zm_monitor.cpp/2138 [front door: 1850 - Alarmed frame while in alert state. Consecutive alarmed frames left to return to alarm state: 009]
12/01/21 18:44:20.737023 zmc_m2[33159].INF-zm_monitor.cpp/2138 [front door: 1851 - Alarmed frame while in alert state. Consecutive alarmed frames left to return to alarm state: 008]
12/01/21 18:44:20.813814 zmc_m2[33159].INF-zm_monitor.cpp/2138 [front door: 1852 - Alarmed frame while in alert state. Consecutive alarmed frames left to return to alarm state: 007]
12/01/21 18:44:20.917181 zmc_m2[33159].INF-zm_monitor.cpp/2166 [front door: 1853 - Left alarm state (693466) - 173(80) images]
12/01/21 18:44:20.917209 zmc_m2[33159].INF-zm_monitor.cpp/2171 [front door: 1853 - Closing event 693466, alarm end]
12/01/21 18:44:21.020582 zmc_m2[33159].INF-zm_monitor.cpp/2132 [front door: 1854 - Gone into prealarm state]

```

Linked monitor did not log anything

maddios, 2021-12-02:
Actually, I forgot to mention that the triggering monitor has a total of 4 linked monitors, 3 other cameras link to it as well as the AI stream. Not sure if this causes some sort of mutex issue

connortechnology, 2021-12-06:
Whoops, accidentally closed this with an unrelated commit.

Um there are no mutexes involved.  we will need debug level 3 logs from all related zmc processes.

maddios, 2021-12-09:
the logs I included were level 9, is level 3 more verbose? or does level 3 specifically log the mutexes?

SteveGilvarry, 2021-12-09:
Only got INF records in the logs you posted, should see DBG lines in debug logs. Also normally set to log to separate files for each process. Level 3 is lower and less verbose than 9. So you might need it for both monitors using _zmc_mX options

This might help or just ask if no luck.
https://zoneminder.readthedocs.io/en/stable/userguide/options/options_logging.html

maddios, 2021-12-09:
Wait where are you seeing INF records in these, these are the zm_debug.log.{id} files. When I download the 2 files I just get a bunch of stuff like this:

```
12/01/21 19:14:49.607387 zmc_m2[37395].DB4-zm_packetqueue.cpp/431 [Locking in get_packet using it 0x560ef41db6c0 queue end? 0]
12/01/21 19:14:49.607392 zmc_m2[37395].DB4-zm_packetqueue.cpp/436 [Have Lock in get_packet]
12/01/21 19:14:49.607397 zmc_m2[37395].DB3-zm_packetqueue.cpp/450 [get_packet using it 0x560ef41db6c0 locking index 33]
12/01/21 19:14:49.607402 zmc_m2[37395].DB4-zm_packet.h/106 [TryLocking packet 33]
```

But sure, I'll get a level 3 debug snapshot when I get a chance, i can see how level 9 might be too verbose.

SteveGilvarry, 2021-12-09:
Sorry couldn't see the files from mobile app for GitHub. Was talking about the Debug logs you pasted into issue log.

maddios, 2021-12-11:
Here are new zm debug logs for the 2 monitors:

[zm_debug_parent.log](https://github.com/ZoneMinder/zoneminder/files/7696226/zm_debug_parent.log)
[zm_debug_linked.log](https://github.com/ZoneMinder/zoneminder/files/7696227/zm_debug_linked.log)

maddios, 2021-12-11:
is this a relevant thing? 

12/10/21 20:37:17.906460 zmc_m2[5368].DB1-zm_monitor.cpp/2336 [Reloading linked monitors for monitor front door, '(null)']

shouldn't null here be the 4 linked monitors?

connortechnology, 2021-12-11:
So the issue here is that the analysis thread was waiting on capture instead of looking for a linked monitor to be triggered.
Capture took .2 of a second which is a long time actually.
So linked monitor checking needs to be  it's own thread...  I can't imagine making this kind of change in 1.36.  Maybe with enough testing.  
That being said... the linked monitor DID notice the alarmed state of the primary...  so I'm gonna have to look closer at the code.
Keep to that code says I,
Yeargh!

maddios, 2021-12-11:
Thanks, I'm glad you can see the issue, it's ok if you can't do it in .36. I've been living with this "bug" for years, I just never really invested a ton of time into debugging it until now. Hopefully it can be addressed at some point in the future.

If you do release a fix I'll gladly test it.

connortechnology, 2021-12-15:
I think I can improve it a lot.  Ultimate fix in a few months. At the end of the day we will always be polling and will have to sleep unless we consume 100% cpu.  
I think long term we will use mqtt to queue a notice about the alarm status and so even though we might notice it late, we will still get it.

maddios, 2021-12-15:
yeah, even if it triggers within a second or two it should still be fine, at least for my setup. Can always increase the pre-alarm frame count.

commit 3b84d7d8 (2021-12-06): Add --daemon to command line documentation so it ends up in man page.  Fixes #3393.  Improve the description and move the pod documentation to the bottom after __END__. Improve executeCommand a bit