ZoneMinder/zoneminder #3395: Update the manual of zmfilter.pl

antivirtel, 2021-12-06:
**Describe Your Environment**
- Version of ZoneMinder [release version, development version, or commit] - 1.36.11-focal1 
- How you installed ZoneMinder [e.g. PPA, RPMFusion, from-source, etc] - PPA
- Full name and version of OS - Ubuntu 20.04.3 LTS Server

**Describe the bug**
I've noticed that some zmfilter.pl processes are running with `--debug`, but this option is not described in the manual. I'm also curious if there are other options, or log redirection for example. (Running a filter from cron, then get the output e-mailed, so I don't need to dig it in the logs). Thank you 

**To Reproduce**
Steps to reproduce the behavior:
Open manual for zmfilter.pl: `man zmfilter.pl` or `info zmfilter.pl`

**Expected behavior**
List and describe all available options.

antivirtel, 2021-12-06:
Thanks, I've fixed that in my post. The point of a manual is to avoid reading the code and have a quick list of all the options. Can you please at least add the `--daemon` option to the manual along with any other? Thank you

jcartner-young, 2021-12-07:
What an amazingly insightful and helpful response. I know why I don't join these cliques...
"If you want to find out more, read the source." - Wha...?!? Just no... No no no no no... *facepalm*
"The point of a manual is to avoid reading the code" - Yes, I agree!  Bullseye!  Exactly correct!
This prevailing, unhealthy culture of only validating fellow developers and treating others with disdain unless they're a developer in their clique is just pathetic.
I see this in so many projects... It's thoroughly off-putting.
The code is only 50% of the required output.  A project is only as useful as the documentation.
Linux will never take off when most proponents are as dismissive as this.
It is easy to see why so many FOSS projects are unfinished or non functional.  Willingness in communication is severely lacking.
This is the straw that broke the camel's back... I'm out.
Good luck with the project.  I hope you finish it (and produce a useful manual.)

On Mon, Dec 6, 2021 at 14:14, B. Roland  wrote:
 Thanks, I've fixed that in my post. The point of a manual is to avoid reading the code and have a quick list of all the options. Can you please at least add the --daemon option to the manual along with any other? Thank you

connortechnology, 2021-12-07:
In case it got lost due to my screwing up the commit logs, I added it and generally updated the man page.

antivirtel, 2021-12-07:
Great, thank you. I've seen you added `--daemon` in 3b84d7d82a2da112e5fbbcfdd8518455b3665fd7. - Is there any other missed options or it's complete now?

Ps. I hope @jcartner-young will have some patience next time, it's FOSS, not a company (which usually worse than FOSS)... Thank you for all your hard work @connortechnology!

connortechnology, 2021-12-07:
As far as I'm aware, it's complete.