ZoneMinder/zoneminder #3397: zmu is trying to connect to the camera in None state (with lots of messages in logs)

hubaksis, 2021-12-06:
**Describe Your Environment**
- Version of ZoneMinder: v1.36.11
- How you installed ZoneMinder: installed Ubuntu, then followed [this process](https://wiki.zoneminder.com/Ubuntu_Server_or_Desktop_Zoneminder_1.34.x) (top section, Install Zoneminder 1.34.x on Ubuntu 20.04, 18.04 with shell script)
- Full name and version of OS: Ubuntu 20.04.3 LTS (GNU/Linux 5.4.0-91-generic x86_64)

**Describe the bug**
Camera is set to `None` state. 
Despite that I can see lots of messages in the log: 
```
Can't connect to capture daemon: 6 Portable camera
Can't open memory map file /dev/shm/zm.mmap.6: No such file or directory
```
[image]

**To Reproduce**
This is happening to only one camera out of 3 which are currently in 'None' mode. 
When I switch it to Monitor - it works fine. When I disconnect the camera and turn it off - it still generates messages in log. 
Not sure when it started. Firstly I noticed a couple of month ago that the only camera in None (at that time) generate these logs after server reboot. Switching the camera to `Monitor` state and back to `None` state resolved the issue till the next reboot. 
Now switching to `Monitor` and back to `None` doesn't do the trick.
Happy to dig in if needed.

**Expected behavior**
zmu process should not be trying to connect to the camera in None state (which it is trying to do according to the logs).

**Debug Logs**
```

2021-12-07 11:38:40 | zmu |   | 658510 | ERR | Can't connect to capture daemon: 6 Portable camera | zmu.cpp | 505
2021-12-07 11:38:40 | zmu |   | 658510 | ERR | Can't open memory map file /dev/shm/zm.mmap.6: No such file or directory | zm_monitor.cpp | 927
2021-12-07 11:38:35 | zmu |   | 658485 | ERR | Can't connect to capture daemon: 6 Portable camera | zmu.cpp | 505
2021-12-07 11:38:35 | zmu |   | 658485 | ERR | Can't open memory map file /dev/shm/zm.mmap.6: No such file or directory | zm_monitor.cpp | 927
2021-12-07 11:38:29 | zmu |   | 658461 | ERR | Can't connect to capture daemon: 6 Portable camera | zmu.cpp | 505
2021-12-07 11:38:29 | zmu |   | 658461 | ERR | Can't open memory map file /dev/shm/zm.mmap.6: No such file or directory | zm_monitor.cpp | 927
2021-12-07 11:38:23 | zmu |   | 658437 | ERR | Can't connect to capture daemon: 6 Portable camera | zmu.cpp | 505
2021-12-07 11:38:23 | zmu |   | 658437 | ERR | Can't open memory map file /dev/shm/zm.mmap.6: No such file or directory | zm_monitor.cpp | 927
2021-12-07 11:38:17 | zmu |   | 658413 | ERR | Can't connect to capture daemon: 6 Portable camera | zmu.cpp | 505
2021-12-07 11:38:17 | zmu |   | 658413 | ERR | Can't open memory map file /dev/shm/zm.mmap.6: No such file or directory | zm_monitor.cpp | 927
2021-12-07 11:38:12 | zmu |   | 658389 | ERR | Can't connect to capture daemon: 6 Portable camera | zmu.cpp | 505
2021-12-07 11:38:12 | zmu |   | 658389 | ERR | Can't open memory map file /dev/shm/zm.mmap.6: No such file or directory | zm_monitor.cpp | 927
2021-12-07 11:38:06 | zmu |   | 658365 | ERR | Can't connect to capture daemon: 6 Portable camera | zmu.cpp | 505
2021-12-07 11:38:06 | zmu |   | 658365 | ERR | Can't open memory map file /dev/shm/zm.mmap.6: No such file or directory | zm_monitor.cpp | 927
2021-12-07 11:38:00 | zmu |   | 658341 | ERR | Can't connect to capture daemon: 6 Portable camera | zmu.cpp | 505
2021-12-07 11:38:00 | zmu |   | 658341 | ERR | Can't open memory map file /dev/shm/zm.mmap.6: No such file or directory | zm_monitor.cpp | 927

```

connortechnology, 2021-12-09:
Are you using zmNinja or the api in general?

Otherwise as near as I can think.... we only query monitor state while live viewing.  So if you had live view or montage up and changed to None in another tab..

hubaksis, 2021-12-14:
For now: stop using API to connect to the camera in 'None' state.

commit 61fe35a3 (2021-12-15): Don't log errors when querying a monitor in NONE Function.  Just report a 0 status.Fixes  #3397
closed 2021-12-14 by hubaksis