ZoneMinder/zoneminder #3398: Error after starting Zoneminder services

mateeees, 2021-12-07:
**Describe Your Environment**
- Version of ZoneMinder:stable,now 1.34.23-1 amd64
- Installed ZoneMinder with your own [guide](https://github.com/ZoneMinder/zoneminder/blob/master/docs/installationguide/debian.rst)
- I am using Debian GNU/Linux 11 (bullseye)

**Describe the bug**
To start with, I think that you have some major spaces in your  install guide. For example, after the installation, you don't mention starting and enabling the **zoneminder.service** . After installing on a completely fresh system with your steps, and starting the service, when I use `systemctl status zoneminder` it gives me this:

> ● zoneminder.service - ZoneMinder CCTV recording and surveillance system
>      Loaded: loaded (/lib/systemd/system/zoneminder.service; enabled; vendor preset: enabled)
>      Active: failed (Result: exit-code) since Tue 2021-12-07 16:36:20 EST; 11min ago
>        Docs: http://zoneminder.readthedocs.org/en/latest/
>     Process: 592 ExecStart=/usr/bin/zmpkg.pl start (code=exited, status=255/EXCEPTION)
>         CPU: 591ms

> Dec 07 16:36:20 detection-server zmpkg[592]: ERR [Error reconnecting to db: errstr:Access denied for user 'zmuser'@'localhost' error val:]
> Dec 07 16:36:20 detection-server zmpkg.pl[592]: Can't call method "prepare_cached" on an undefined value at /usr/share/perl5/ZoneMinder/Config.pm line 96.
> Dec 07 16:36:20 detection-server zmpkg.pl[592]: BEGIN failed--compilation aborted at /usr/share/perl5/ZoneMinder/Config.pm line 147.
> Dec 07 16:36:20 detection-server zmpkg.pl[592]: Compilation failed in require at /usr/share/perl5/ZoneMinder.pm line 33.
> Dec 07 16:36:20 detection-server zmpkg.pl[592]: BEGIN failed--compilation aborted at /usr/share/perl5/ZoneMinder.pm line 33.
> Dec 07 16:36:20 detection-server zmpkg.pl[592]: Compilation failed in require at /usr/bin/zmpkg.pl line 34.
> Dec 07 16:36:20 detection-server zmpkg.pl[592]: BEGIN failed--compilation aborted at /usr/bin/zmpkg.pl line 34.
> Dec 07 16:36:20 detection-server systemd[1]: zoneminder.service: Control process exited, code=exited, status=255/EXCEPTION
> Dec 07 16:36:20 detection-server systemd[1]: zoneminder.service: Failed with result 'exit-code'.
> Dec 07 16:36:20 detection-server systemd[1]: Failed to start ZoneMinder CCTV recording and surveillance system.

~
**I think this sort of speak bug is very similar to [this one](https://github.com/zoneMinder/zoneminder/issues/2799)** 
(that was closed without an answer that solves the problem, I know it was a bot, but you didn't answer....)

**To Reproduce**
Steps to reproduce the behavior:
1. follow the install guide(with additional commands as I said)
2. enable the service
3. start the service
4. See an error

**Expected behavior**
It should just start and I should be able to use the browser GUI.

**Debug Logs**

> Dec 07 16:47:45 detection-server sudo[729]: detection : TTY=pts/0 ; PWD=/home/detection ; USER=root ; COMMAND=/usr/bin/systemctl status zoneminder.service
Dec 07 16:47:45 detection-server sudo[729]: pam_unix(sudo:session): session opened for user root(uid=0) by detection(uid=1000)
Dec 07 16:54:52 detection-server sudo[729]: pam_unix(sudo:session): session closed for user root
Dec 07 16:54:58 detection-server sudo[756]: detection : TTY=pts/0 ; PWD=/home/detection ; USER=root ; COMMAND=/usr/bin/systemctl start zoneminder.service
Dec 07 16:54:58 detection-server sudo[756]: pam_unix(sudo:session): session opened for user root(uid=0) by detection(uid=1000)
Dec 07 16:54:58 detection-server sudo[756]: pam_unix(sudo:session): session closed for user root

using `journalctl -xe`

I have done everything in my power to fix this problem, but can't do it. 

Thanks in advance. :D

connortechnology, 2021-12-07:
We don't document that because the install process does it for you.

Your problem is that the credentials to connect to mysql are wrong. Probably because you changed the root password in mysql so the install wasn't able to set it up for you.

You might also note that our instructions don't include a section for bullseye, so there that too.  However the steps are the same.

Even further, the docs don't mention it because we do not package 1.34 for bullseye because 1.34 is OLD.

So lots going on here.  

My recommendation is edit /etc/apt/sources.list.d/zoneminder.list and replace the 1.34 with 1.36.  Then sudo apt update && sudo apt install zoneminder

connortechnology, 2021-12-08:
I updated the wiki a bit, you might want to try following it.
https://wiki.zoneminder.com/Debian_11_Bullseye_with_Zoneminder_1.36.x

mateeees, 2021-12-08:
ok, I will try that. I will let you know, thanks.

mateeees, 2021-12-08:
> We don't document that because the install process does it for you.
>
> Your problem is that the credentials to connect to mysql are wrong. Probably because you changed the root password in mysql so [...]

> We don't document that because the install process does it for you.
>
> Your problem is that the credentials to connect to mysql are wrong. Probably because you changed the root password in mysql so [...]

Also, I don't know if we are talking about the same thing, but in the guide I mentioned in my original post, there is a tutorial for Debian 11(bullseye)

mateeees, 2021-12-08:
Yeah, reinstalled it using your new wiki post, all working, thanks a lot.

proximus26usa, 2022-09-07:
I have really try everything and yet, i cant figure out why I am getting this error:

Sep 07 01:58:45 thanatos-linux zmpkg[7222]: ERR [Error reconnecting to db: errstr:Can't connect to local MySQL s>
Sep 07 01:58:45 thanatos-linux systemd[1]: zoneminder.service: Control process exited, code=exited, status=255/E>

It is clear that I can't connect to DB, MySQL, however I was able resolve this issue, and was okay, then I start getting other alarms and this was back... Frustrating ...

proximus26usa, 2022-09-07:
Now even stranger things happening:
ep 07 02:01:29 thanatos-linux systemd[1]: zoneminder.service: Control process exited, code=exited, status=255/EXCEPTION
Sep 07 02:01:29 thanatos-linux systemd[1]: zoneminder.service: Failed with result 'exit-code'.
Sep 07 02:01:29 thanatos-linux systemd[1]: Failed to start ZoneMinder CCTV recording and surveillance system.

connortechnology, 2025-08-05:
Look in your logs.  Probably the issue is with mysql.