ZoneMinder/zoneminder #3405: zm_rtsp_server pending

HaoLVb, 2022-01-05:
Describe Your Environment
Version of ZoneMinder : 1.37.6
How you installed ZoneMinder : from deb
Full name and version of OS : Ubuntu 20.04

Describe the bug:
I can't watch live view even though the service zm_rtsp_server is still running. 
I am pending in here and connection timed out
[image]
But when I restart rtsp_server: sudo -u www-data zmdc.pl restart zm_rtsp_server, I can watch live view

connortechnology, 2022-01-05:
Will need zm_rtsp_server logs from when it wasn't responding.

I recently merged an upstream patch for a case where a long running rtsp_server could run out of file handles.  Is this a long running zm_rtsp_server that displays the problem?

connortechnology, 2022-01-31:
rtsp server code received more updates in current master.  Please test.

connortechnology, 2022-05-03:
Has there been any improvement?