ZoneMinder/zoneminder #3407: Playback issues with 1.36

g1bs0nsg, 2022-01-10:
**Describe Your Environment**
- Version of ZoneMinder: 1.36.12
- How you installed ZoneMinder: From zmrepo.zoneminder.com
- Full name and version of OS: Debian 11 Bullseye, 4 CPU cores, 4 GB of RAM
- Browser name and version: Chrome, Firefox, Windows, Linux, MacOS - Happens on all machines on my network that I attempt playback from. 

**Describe the bug**
Was previously running 1.34 from the Debian repo with no issues on Buster.  Upgraded system to Bullseye, still no issues.  Changed over to zmrepo and installed 1.36.11.  Everything was working fine, able to stream cameras, etc.  Went to review my captures at the end of the week and ran into a problem.   When I attempt to replay video using Gapless events, it will playback 2-3 events and then video freezes.  Upgraded to 1.36.12, but problem remains. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to: Events for any monitor
2. Click on: Any event, then set replay to Gapless Events, Scale Actual, Codec Auto
4. When video playback stops, check web_php.log and see the following two error messages:
`01/10/22 11:38:48.675972 web_php[23736].ERR [192.168.0.81] [Timed out waiting for msg /run/zm/zms-729128s.sock] at /usr/share/zoneminder/www/ajax/stream.php line 95
01/10/22 11:38:48.716992 web_php[23736].ERR [192.168.0.81] [No data to read from socket] at /usr/share/zoneminder/www/includes/functions.php line 1902
`
Doesn't matter what the rate is set to, but changing the rate from 1x to 16x doesn't seem to make much difference in the actual playback speed.  It seems kind of choppy, and then sometimes it looks like it's fast-forwarding through the frames ultra fast. 

In /var/log/apache2/error.log seeing the following when this occurs:
`[Mon Jan 10 11:49:28.751898 2022] [cgi:warn] [pid 23735] [client 192.168.0.81:53571] AH01220: Timeout waiting for output from CGI script /usr/lib/zoneminder/cgi-bin/nph-zms, referer: http://zoneminder.sky.net/zm/index.php?view=event&eid=47126&filter%5BQuery%5D%5Bterms%5D%5B0%5D%5Battr%5D=MonitorId&filter%5BQuery%5D%5Bterms%5D%5B0%5D%5Bop%5D=%3D&filter%5BQuery%5D%5Bterms%5D%5B0%5D%5Bval%5D=4&filter%5BQuery%5D%5Bsort_asc%5D=1&filter%5BQuery%5D%5Bsort_field%5D=StartDateTime&filter%5BQuery%5D%5Blimit%5D=100&sort_field=StartDateTime&sort_asc=1
[Mon Jan 10 11:51:34.272874 2022] [cgi:error] [pid 25506] [client 192.168.0.81:53682] AH01215: corrupted size vs. prev_size while consolidating: /usr/lib/zoneminder/cgi-bin/nph-zms, referer: http://zoneminder.sky.net/zm/index.php?view=event&eid=46990&filter%5BQuery%5D%5Bterms%5D%5B0%5D%5Battr%5D=MonitorId&filter%5BQuery%5D%5Bterms%5D%5B0%5D%5Bop%5D=%3D&filter%5BQuery%5D%5Bterms%5D%5B0%5D%5Bval%5D=4&filter%5BQuery%5D%5Bsort_asc%5D=1&filter%5BQuery%5D%5Bsort_field%5D=StartDateTime&filter%5BQuery%5D%5Blimit%5D=100&sort_field=StartDateTime&sort_asc=1&page=1
`

If I play these events 1 at a time using the skip buttons, they all play perfectly fine.  Likewise if I set it to All Events and hit the skip buttons during blank spaces everything plays back fine as well.  Although the rate settings still seem to not make much difference in terms of playback speed. 

Also the count of zms.lock and zms.sock files increases every time this error is logged.  I have 5 cameras total, and only 3 of them are set to record events.  When I restart zoneminder I will have 5 locks and 5 socks, and it will stay that way for days until I try to playback video.  Currently after attempting to run gapless events 4 to 5 times I have 19 lock files, and 9 sock files in /run/zm

I have monitored the system with top during playback and don't see anything out of the ordinary..... no where close to taxing the CPU/RAM at all. 

**Expected behavior**
All events to playback seamlessly as they did prior to upgrade.   On 1.34 I was able to set playback rate to 32x and stream through all events on each monitor without issue.

sweetsounds48v, 2022-01-17:
I believe I'm experiencing the same issue.

Zoneminder 1.36.12
Running on Ubuntu 18.04 with apache2
4 core CPU, 8GB of RAM

Mostly tested with Firefox and Chrome on Linux, and Android. I've tried both HTTP, and self signed HTTPS. I'm also running behind an NGINX proxy server, I've tested both direct and proxy connections. Getting consistent errors under all of these conditions.

Everything seems to work fine except playback controls and cue frames appearing on the playback scrub bar. All of my issues are exactly the same as OP has described. Video will load for event playback, but navigation buttons seem greyed out. Changing the playback rate to 1x will force the video to play, and clicking on the scrub bar while the video is playing will skip to various points in the video. There are no cue marks on the video either (browser is showing error), and when playback begins the progress indicator below the video changes to Progress: NaN:NaN:NaNs.

Zoneminder Log Output:
`
2022-01-17 10:31:31 | web_js | ERR | getCmdResponse  stream error: Socket /run/zm/zms-109251s.sock does not exist.  This  file is created by zms, and since it does not exist, either zms did not  run, or zms exited early.  Please check your zms logs and ensure that  CGI is enabled in apache and check that the PATH_ZMS is set correctly.   Make sure that ZM is actually recording.  If you are trying to view a  live stream and the capture process (zmc) is not running then zms will  exit. Please go to  http://zoneminder.readthedocs.io/en/latest/faq.html#why-can-t-i-see-streamed-images-when-i-can-see-stills-in-the-zone-window-etc  for more information. - checkStreamForErrors() | ?view=event&eid=0&filter%5BQuery%5D%5Bterms%5D%5B0%5D%5Battr%5D=StartDateTime&filter%5BQuery%5D%5Bterms%5D%5B0%5D%5Bop%5D=%3E%3D&filter%5BQuery%5D%5Bterms%5D%5B0%5D%5Bval%5D=-1+hour&filter%5BQuery%5D%5Bterms%5D%5B1%5D%5Bcnj%5D=and&filter%5BQuery%5D%5Bterm | -
-- | -- | -- | -- | -- | --
2022-01-17 10:31:31 | web_php | ERR | Socket  /run/zm/zms-109251s.sock does not exist.  This file is created by zms,  and since it does not exist, either zms did not run, or zms exited  early.  Please check your zms logs and ensure that CGI is enabled in  apache and check that the PATH_ZMS is set correctly.  Make sure that ZM  is actually recording.  If you are trying to view a live stream and the  capture process (zmc) is not running then zms will exit. Please go to  http://zoneminder.readthedocs.io/en/latest/faq.html#why-can-t-i-see-streamed-images-when-i-can-see-stills-in-the-zone-window-etc  for more information. | /usr/share/zoneminder/www/includes/functions.php | 1902
2022-01-17 10:31:30 | zms_e2 | ERR | Unable to send raw frame 1: Broken pipe 0 | zm_eventstream.cpp | 1096
2022-01-17 10:31:29 | web_js | ERR | TypeError: bottomEl.offset() is undefined | zm/cache/skins_classic_js_skin-classic-1639175886.js | 589
2022-01-17 10:31:27 | zms_e2 | ERR | Can't sendto on sd 6: No such file or directory | zm_eventstream.cpp | 579
2022-01-17 10:31:26 | web_php | ERR | No data to read from socket | /usr/share/zoneminder/www/includes/functions.php | 1902
2022-01-17 10:31:26 | web_php | ERR | Timed out waiting for msg /run/zm/zms-458259s.sock | /usr/share/zoneminder/www/ajax/stream.php | 95
`

Browser Console Output:
`
No cue frames for event skins_classic_views_js_event-classic-1639175886.js:98:13

Request Failed: timeout, timeout skins_classic_js_skin-classic-1639175886.js:773:11

Ajax request failed.  No responseText.  jqxhr follows: skins_classic_js_skin-classic-1639175886.js:775:13

Object { readyState: 0, getResponseHeader: getResponseHeader(e), getAllResponseHeaders: getAllResponseHeaders(), setRequestHeader: setRequestHeader(e, t), overrideMimeType: overrideMimeType(e), statusCode: statusCode(e), abort: abort(e), state: state(), always: always(), catch: catch(e)
, … }
skins_classic_js_skin-classic-1639175886.js:776:13
`

Here's the Apache2 error:
`
[php7:warn] [pid 1569] [client xxx.xxx.xxx.xxx:36546] PHP Warning:  A non-numeric value encountered in /usr/share/zoneminder/www/skins/classic/views/event.php on line 274, referer: https://<domain>.com/zm/index.php?view=events&page=1&filter%5BQuery%5D%5Bterms%5D%5B0%5D%5Battr%5D=StartDateTime&filter%5BQuery%5D%5Bterms%5D%5B0%5D%5Bop%5D=%3E%3D&filter%5BQuery%5D%5Bterms%5D%5B0%5D%5Bval%5D=-1+hour&filter%5BQuery%5D%5Bterms%5D%5B1%5D%5Bcnj%5D=and&filter%5BQuery%5D%5Bterms%5D%5B1%5D%5Battr%5D=MonitorId&filter%5BQuery%5D%5Bterms%5D%5B1%5D%5Bop%5D=%3D&filter%5BQuery%5D%5Bterms%5D%5B1%5D%5Bval%5D=1&filter%5BQuery%5D%5Bsort_asc%5D=1&filter%5BQuery%5D%5Bsort_field%5D=StartDateTime&filter%5BQuery%5D%5Blimit%5D=100
`

I can log in via SSH and verify that zms is running (lots of other things wouldn't work if it wasn't) and I can see the .sock file being created in /run/zm. I've double checked the permissions of the file and they're set to 755 for user www-data:www-data. As far as I can see everything should be working but it's not.

connortechnology, 2022-04-11:
No you are having a different issue.  Not very many browsers can play h265.  If you are using passthrough, then you will have to tell the camera to send h264.  Or select mjpeg as the viewing codec.  Or if encoding, select h264 instead of h265.

wnadim92, 2022-07-01:
think I have same issue.  
My setup is {"version":"1.36.19","apiversion":"2.0"}

The Ubuntu is 
snadim92@ubuntu-hp:~$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 22.04 LTS
Release:        22.04
Codename:       jammy

Unable to export video

<html xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/wnadi/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/wnadi/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

Date/Time | Component | Server | PID | Level | Message | File | Line
-- | -- | -- | -- | -- | -- | -- | --
6/30/2022 23:08 | web_php |   | 24449 | ERR | Video Generation Failed | /usr/share/zoneminder/www/includes/functions.php | 1907
6/30/2022 23:08 | zmvideo |   | 24562 | ERR | Unable to generate video, check   /var/cache/zoneminder/events/3/2022-06-30/104/ffmpeg.log for details | zmvideo.pl | '-
6/30/2022 23:08 | web_js |   | 24449 | PNC | Unknown error type: [8192]   Function strftime() is deprecated#0   /usr/share/zoneminder/www/includes/logger.php(529): ZM\Panic('...')#1   /usr/share/zoneminder/www/includes/logger.php(366): ZM\ErrorHandler(8192,   '...', '...', 366)#2 /usr/share/zoneminder/www/ajax/log.php(66):   ZM\Logger->logPrint(-2, '...', '...', NULL)#3   /usr/share/zoneminder/www/ajax/log.php(33): createRequest()#4   /usr/share/zoneminder/www/index.php(277): require_once('...') | /usr/share/zoneminder/www/includes/logger.php | 529
6/30/2022 23:08 | web_js |   | 24449 | PNC | Unknown error type: [8192]   Function strftime() is deprecated#0   /usr/share/zoneminder/www/includes/logger.php(529): ZM\Panic('...')#1   /usr/share/zoneminder/www/includes/logger.php(366): ZM\ErrorHandler(8192,   '...', '...', 366)#2 /usr/share/zoneminder/www/ajax/log.php(66):   ZM\Logger->logPrint(-2, '...', '...', NULL)#3   /usr/share/zoneminder/www/ajax/log.php(33): createRequest()#4   /usr/share/zoneminder/www/index.php(277): require_once('...') | /usr/share/zoneminder/www/includes/logger.php | 529

</body>

</html>

schuellerf, 2022-09-13:
I had similar issues running on Ubuntu 22.04.
"strftime() is deprecated" led me to the conclusion that zoneminder might not support PHP 8.x ?
so I followed https://medium.com/@skounis/downgrade-php8-1-to-php8-0-or-php7-4-on-ubuntu-22-04-2fab4a6a3be3
and hack-downgraded to php7.4. On my system noone else needs PHP so I guess that's fine.

essentially I was doing
```
sudo apt install software-properties-common ca-certificates lsb-release apt-transport-https
LC_ALL=C.UTF-8 sudo add-apt-repository ppa:ondrej/php
sudo apt install libapache2-mod-php7.4 php7.4-cli php7.4-common php7.4-gd php7.4-mysql php7.4-opcache php7.4-readline
sudo update-alternatives --config php  # <--- select php7.4 here!
php -v # double check that really PHP7.4.x is not active

sudo a2dismod php8.1
sudo a2enmod php7.4
sudo systemctl reload apache2
```

additionally (maybe I messed up somehow) I had to start zoneminder again and fix this font path:
```
sudo systemctl start zoneminder # that should't really be necessary
sudo systemctl enable zoneminder
sudo ln -s /usr/share/zoneminder/www/fonts /usr/share/zoneminder/fonts
```
I didn't find any official place which PHP versions are supported... maybe this issue (or a new one) should care about PHP8.1 compatibility? (just my wild guess)

connortechnology, 2022-09-13:
We have strayed very far from the original issue.  php8 has been dealt with.  Upgrade to 1.36.25.

schuellerf, 2022-09-13:
> We have strayed very far from the original issue. php8 has been dealt with. Upgrade to 1.36.25.

Thank's for the hint @connortechnology - sorry for distraction - the problem of @wnadim92 and me seem to be *unrelated to the current issue* and rather already fixed in [1.36.20](https://github.com/ZoneMinder/zoneminder/releases/tag/1.36.20) or later which does not seem to be packaged with Ubuntu 22.04 as of now, so we'd rather head over to …hmm just found it… _your_ PPA and upgrade 🥳👌
https://launchpad.net/~iconnor/+archive/ubuntu/zoneminder-1.36
(works for me now - thank's again!)

ITFekkak, 2023-05-08:
hello everyone , I have a question, is it possible to browse the recording that is in the hard drive of the DVR hikvision or Dahua via Zoneminder?
Thanks soo much!

surfzoid, 2024-03-28:
> hello everyone , I have a question, is it possible to browse the recording that is in the hard drive of the DVR hikvision or Dahua via Zoneminder? Thanks soo much!

I don't think, but under linux you can use QtVsPlayer and HikNetExtractor for Hik.