ZoneMinder/zoneminder #3411: Problem with linked monitors, alarm state never end on linked monitor

gonzalezcalleja, 2022-01-21:
**Describe Your Environment**
- Version of ZoneMinder [release version, development version, or commit]
v1.37.9 4eb4e0e

- How you installed ZoneMinder [e.g. PPA, RPMFusion, from-source, etc]
from-source

- Full name and version of OS
Ubuntu 18.04

- Browser name and version (if this is an issue with the web interface)
Chrome 97.0.4692.71 

**Describe the bug**
I have 2 monitors:
- Monitor 1 in MoDect mode
- Monitor 2 in Record mode and linked to Monitor 1

When Monitor 1 trigger and alarm, Monitor 2 status changed to Alarm but never changes from this state (infinite alarm state).

If I change Monitor 2 to "NoDect" mode instead of "Record" works as expected (Monitor 2 Alarm status go off at the same time that the Monitor 1 that triggered the Alarm).

**Expected behavior**
That the Monitor 2 status changes in "Record" mode in the same way that it does in "MoDect" mode.

ovargasp, 2022-01-21:
Hi Gonzalezcalleja that's actually the expected behavior of linked monitors... if you put a monitor on "record" it will be recording continuously and creating fixed size events not linked to an external trigger, in order to record on an external or linked event the second monitor must be on NoDect mode.

Check the ZM documentation for information

https://zoneminder.readthedocs.io/en/stable/userguide/definemonitor.html#source-tab

check:

Monitor Tab
- Function
- Linked Monitors
 
regards

Osvaldo

gonzalezcalleja, 2022-01-21:
Hi @ovargasp, thanks for your fast answer.

Regarding to the video recording and the fixed size events is the expected behaviour but not for the "infinite alert status" that is the reason I opened this bug.

My expected  behaviour should be that the status of the monitor and the linked monitor that triggered the alarm should be synchronised to "ON" and "OFF" alarm status -like in MoDect->NoDect modes-, now in MoDect->Record modes, only changes to "ON" but never to OFF so this causes that the Monitor goes into an infinite status of alarm.

Regards

changes the status to alert "ON" when the linked monitor trigger it should be "OFF when the alert status in linked monitor finished (in the same way that in MoDect-NoDect modes) regardless of status

closed 2022-01-22 by connortechnology

gonzalezcalleja, 2022-01-22:
Thanks @connortechnology  for the fast patch! I have tested right now and alert status are synchronized!

commit 9443aaa2 (2022-01-22): Do state changes when in RECORD mode as well as it doesn't do motion detection Fixes #3411