ZoneMinder/zoneminder #3412: Control Type: NETCAT ONVIF PTZ doesn't stop

jef737, 2022-01-22:
- ZoneMinder release V1.34.26
- Installed from PPA
- Ubuntu 18.04.1 LTS

As mentioned in the topic https://forums.zoneminder.com/viewtopic.php?f=40&t=30509#p120200 the camera does'nt stop when using PTZ control
To solve the problem change in /usr/share/perl5/ZoneMinder/Control/Netcam.pm has been made as follow:
  replace the line 211 starting with my $msg ='<s:Envelope xml........
  by the line 233 and change Y="0.5" by Y="0"
Then in the control tab of the monitor configuration set Auto Stop Timeout to 0.5

connortechnology, 2022-01-22:
@knight-of-ni do you still have a camera that this works on? Can you test this replacement of line 211? Is line 206 now in 1.36.

knight-of-ni, 2022-01-22:
Nope Sorry. I do not have this hardware any longer.

jef737, 2022-01-23:
I intend to move my installation on new hardware and upgrade to new release, so i could test this in next weeks, and give you the result

jef737, 2022-01-30:
I started to do a fresh install of ZoneMinder 1.36 on Ubuntu 20.04 LTS, and test PTZ control. Finally it works without changing the line 206. 
Nevertheless, i had to change something in the authentificationHeader routine, as i did in the previous release. As it has no relationship with this issue let me know if i must open a new one? If not i will give you details about what i did in that one.

gyzod, 2022-02-01:
Hello, I have the same problem with 2 of my cameras on master branch 1.37.10.  How can I test the workaround ? 

Also, I don't know if it can be related but the 2 cameras that do not stop have a PTZ profile the attribut : default ptz speed: (pan=0.5,tilt=0.5,zoom=0.5) .  The other camera that works have a PTZ profile of default ptz speed: (pan=1,tilt=1,zoom=1).

gyzod, 2022-02-02:
Ok, I was finally able to fix the problem using the @jef737 recipe.  All my cameras works using a SOAP move message with x=0 and y=0.  I confirm that it works. @connortechnology could it be fixed in the master branch ?

connortechnology, 2022-02-03:
So.. line 206 in 1.34 reboots the camera.  Not exactly the preferred way of stopping ptz.  In terms of API reset is not supposed to reboot the camera it is supposed go back to home position. . So that's not good.  

Can I assume that you mean you achieve the STOP by sending a move command with x=0, y=0?  I guess that would make some stupid sense.

gyzod, 2022-02-03:
line 206 in master (1.37) is the autostop SOAP message which, if we replace for a move command with x and y=0, works.

connortechnology, 2022-02-03:
Got it.  So why don't we just send both?  I'll commit.  Anyone have other issues that need to go in, no need to create a separate thread, just post what you know.

gyzod, 2022-02-03:
Another thing, I have another netcat onvif camera that works using the current line 206 message.  When I apply the fix (x=0,y=0), is also works. So from what I see, it doesn't break already working cameras.

connortechnology, 2022-02-03:
I just pushed a commit that does both, so hopefully we have covered all cameras.

jef737, 2022-02-04:
So in authentificationHeader routine i changed the random variable nonce calculation line  121 in release 1.36 which doesn't work for me. May be the code is not optimized but it works:
`  my @n;
  my $i = 0;
  while ($i < 20)
	{
	$n[$i] = chr(int(rand(254)));
	$i = $i + 1;
	}
  my $nonce = join("",@n);`