ZoneMinder/zoneminder #3416: Chrome cant find the next event while playback. "gapless" is enabled with 24/7 record.

aaronsta1, 2022-01-26:
**Describe Your Environment**
- Version of ZoneMinder [release version, development version, or commit]
- 1.34.26
- How you installed ZoneMinder [e.g. PPA, RPMFusion, from-source, etc]
- PPA
- Full name and version of OS
- ubuntu 20.04 focal
- Browser name and version (if this is an issue with the web interface)
Chrome Version 97.0.4692.99 (Official Build) (64-bit)

**Describe the bug**
when watching an event it will not go to the next or previous video.. it says no more events and video not found.

**To Reproduce**
click on an event.. hit the next button. 

**Expected behavior**
its supposed to load the next event.

**Debug Logs**
```
i dont know what logs to look at for this situation.. 
Microsoft edge browser works just fine as far as i can tell its just chrome.

```

connortechnology, 2022-01-31:
First step is update to 1.36.  1.34 is no longer supported. 

Alternate step would be to check your javascript console, likely there is an error reported there.  Right click anywhere , choose inspect, go to Console tab.

Regardless 1.34 won't be receiving any further updates so unless you plan to edit your code yourself, update to 1.36.

aaronsta1, 2022-01-31:
i didnt know there was a 1.36, ill update.
its been working this whole time just fine, this is a new thing. im guessing its something with the new chrome version.

aaronsta1, 2022-02-02:
upgrading to v1.36 seemed to have fixed the chrome issue..

closed 2022-02-02 by aaronsta1