ZoneMinder/zoneminder #3419: FTBFS with ffmpeg 5.0: `no member named ‘codec_info_nb_frames’`

onlyjob, 2022-02-01:
As [reported in Debian](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1004641), ZoneMinder FTBFS with Ffmpeg 5.0 as follows:

```
| [ 72%] Building CXX object src/CMakeFiles/zm.dir/zm_ffmpeg.cpp.o
| cd /<<PKGBUILDDIR>>/dbuild/src && ccache /usr/lib/ccache/c++ -DHAVE_SENDFILE4_SUPPORT=1 -I/<<PKGBUILDDIR>>/dbuild/src -I/<<PKGBUILDDIR>>/src -I/usr/include/mysql -I/<<PKGBUILDDIR>>/dbuild -I/<<PKGBUILDDIR>>/dep/libbcrypt/include/bcrypt -I/<<PKGBUILDDIR>>/dep/jwt-cpp/include -I/<<PKGBUILDDIR>>/dep/RtspServer/src -I/<<PKGBUILDDIR>>/dep/RtspServer/src/xop -I/<<PKGBUILDDIR>>/dep/RtspServer/src/3rdpart -I/<<PKGBUILDDIR>>/dep/span-lite/include -g -O2 -ffile-prefix-map=/<<PKGBUILDDIR>>=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -O2 -Wall -Wconditionally-supported -Wextra -Wformat-security -Wno-cast-function-type -Wno-unused-parameter -Woverloaded-virtual -Wvla -MD -MT src/CMakeFiles/zm.dir/zm_ffmpeg.cpp.o -MF CMakeFiles/zm.dir/zm_ffmpeg.cpp.o.d -o CMakeFiles/zm.dir/zm_ffmpeg.cpp.o -c /<<PKGBUILDDIR>>/src/zm_ffmpeg.cpp
| Manifying 22 pod documents
| [ 73%] Building CXX object src/CMakeFiles/zm.dir/zm_ffmpeg_camera.cpp.o
| cd /<<PKGBUILDDIR>>/dbuild/src && ccache /usr/lib/ccache/c++ -DHAVE_SENDFILE4_SUPPORT=1 -I/<<PKGBUILDDIR>>/dbuild/src -I/<<PKGBUILDDIR>>/src -I/usr/include/mysql -I/<<PKGBUILDDIR>>/dbuild -I/<<PKGBUILDDIR>>/dep/libbcrypt/include/bcrypt -I/<<PKGBUILDDIR>>/dep/jwt-cpp/include -I/<<PKGBUILDDIR>>/dep/RtspServer/src -I/<<PKGBUILDDIR>>/dep/RtspServer/src/xop -I/<<PKGBUILDDIR>>/dep/RtspServer/src/3rdpart -I/<<PKGBUILDDIR>>/dep/span-lite/include -g -O2 -ffile-prefix-map=/<<PKGBUILDDIR>>=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -O2 -Wall -Wconditionally-supported -Wextra -Wformat-security -Wno-cast-function-type -Wno-unused-parameter -Woverloaded-virtual -Wvla -MD -MT src/CMakeFiles/zm.dir/zm_ffmpeg_camera.cpp.o -MF CMakeFiles/zm.dir/zm_ffmpeg_camera.cpp.o.d -o CMakeFiles/zm.dir/zm_ffmpeg_camera.cpp.o -c /<<PKGBUILDDIR>>/src/zm_ffmpeg_camera.cpp
| In file included from /<<PKGBUILDDIR>>/src/zm_ffmpeg.cpp:22:
| /<<PKGBUILDDIR>>/src/zm_ffmpeg.cpp: In function ‘void zm_dump_stream_format(AVFormatContext*, int, int, int)’:
| /<<PKGBUILDDIR>>/src/zm_ffmpeg.cpp:299:11: error: ‘AVStream’ {aka ‘struct AVStream’} has no member named ‘codec_info_nb_frames’
|   299 |       st->codec_info_nb_frames, codec->frame_size,
|       |           ^~~~~~~~~~~~~~~~~~~~
| /<<PKGBUILDDIR>>/src/zm_logger.h:200:72: note: in definition of macro ‘logPrintf’
|   200 |       Logger::fetch()->logPrint(false, __FILE__, __LINE__, logLevel, ##params); \
|       |                                                                        ^~~~~~
| /<<PKGBUILDDIR>>/src/zm_ffmpeg.cpp:298:3: note: in expansion of macro ‘Debug’
|   298 |   Debug(1, "frames:%d, frame_size:%d stream timebase: %d/%d",
|       |   ^~~~~
| make[4]: *** [src/CMakeFiles/zm.dir/build.make:471: src/CMakeFiles/zm.dir/zm_ffmpeg.cpp.o] Error 1
```

connortechnology, 2022-02-01:
Fixed by 34d7f192b4d10da46294185c97db73c5371c2c7c

connortechnology, 2022-02-01:
Will be included in 1.36.13 soon.

closed 2022-02-01 by connortechnology

onlyjob, 2022-02-02:
Awesome. Thank you very much.