ZoneMinder/zoneminder #3427: White screen and Undefined constant "ZM\ZM_LOG_LEVEL_DATABASE" after recent update

calvss, 2022-02-09:
**Describe Your Environment**
- 1.36.12
- Installed via apt install
- Debian Bookworm
- Firefox 97.0

**Describe the bug**
I'm getting a white screen on the web interface after a recent update, and enabling display_errors in php gets me this error message:
`Fatal error: Uncaught Error: Undefined constant "ZM\ZM_LOG_LEVEL_DATABASE" in /usr/share/zoneminder/www/includes/logger.php:102 Stack trace: #0 /usr/share/zoneminder/www/includes/logger.php(197): ZM\Logger->initialise() #1 /usr/share/zoneminder/www/includes/config.php(141): ZM\Logger::fetch() #2 /usr/share/zoneminder/www/index.php(46): require_once('...') #3 {main} thrown in /usr/share/zoneminder/www/includes/logger.php on line 102`

**To Reproduce**
Steps to reproduce the behavior:
1. Have a working setup before 1.36.12
2. `# apt update && apt upgrade`
3. Open browser
4. White screen

**Expected behavior**
Not a white screen

**Debug Logs**
`systemctl status zoneminder` doesn't print any errors

bachmarc, 2022-04-20:
Just to double it, I have same issue with fresh Debian Testing build of this week.
No DB "zm" was created during apt install and no user "zmuser"
I did manually, service starts fine afterwards but calling the main page ends in 500 and log shows: PHP Fatal error:  Uncaught Error: Undefined constant "ZM\\ZM_LOG_LEVEL_DATABASE" in /usr/share/zoneminder/www/includes/logger.php:102\nStack trace:\n#0 /usr/share/zoneminder/www/includes/logger.php(197): ZM\\Logger->initialise()\n#1 /usr/share/zoneminder/www/includes/config.php(141): ZM\\Logger::fetch()\n#2 /usr/share/zoneminder/www/index.php(46): require_once('...')\n#3 {main}\n  thrown in /usr/share/zoneminder/www/includes/logger.php on line 102

onlyjob, 2022-04-20:
What gave you the idea that database should be created for you automatically?? MariaDB might be running on a different server and it is up to admin to know/plan the setup and to implement it. Not everything is worth automating and instructions on how to initialise database are simple enough. You should have checked the README and the documentation. Some documentation is provided by the package and other is available here: https://zoneminder.readthedocs.io/en/stable/

onlyjob, 2022-04-22:
> There is no current installation guide for Debian 11.

So you've never checked README (and other files) in `/usr/share/doc/zoneminder` ??

Please let me know if the following README is missing anything (or in need of any corrections): https://sources.debian.org/src/zoneminder/1.36.14%2Bdfsg1-1/debian/README.Debian/
Your improvement suggestions will be most welcome.
(`README.Debian` is maintained here: https://salsa.debian.org/debian/zoneminder/-/blob/master/debian/README.Debian)

connortechnology, 2022-04-22:
The wiki is a mess.  Every debian based install guide should start with a pointer to the README.

bachmarc, 2022-04-25:
> So you've never checked README (and other files) in `/usr/share/doc/zoneminder` ??
>
> Please let me know if the following README is missing anything (or in need of any corrections): [...]

HI,
to be frank: no
Usually I read through the installation guide to estimate the effort and challenges... 
I found no current but several very similar older guides. One mentioned DB installation, one said it is done "magically" starting with a certain release, one pointed to a sql dump.

So I decided to run apt-get install zoneminder and see what happens.
It ran, installed some dependencies (DB ...) and exited with a clean 0.
I think to remember that some debian packages inform about open tasks, I do not recall that zoneminder did it.

No think to have created the DB correctly, I read the README , the service is starting BUT hostnome/zm gives a 500...
I am not sure what went wrong or how to fix it. So reading the README was not an eye opener.. somehow it was like a mixture of the other guides read.

When I get it up and runnig I will consolidate my findings in guide 5 /readme

Marc

nabbi, 2022-06-08:
I also seen this symptom if the zm_create.sql import failed at reading the triggers.sql, resulting in a partial configuration.
DROP the table and start over.

rbur004, 2022-08-29:
with mariadb you don't have a root password anymore, so it is 
mysql < /usr/share/zoneminder/db/zm_create.sql
mysql -e "grant lock tables,alter,drop,select,insert,update,delete,create,index,alter routine,create routine, trigger,execute,references on zm.* to 'zmuser'@localhost identified by 'zmpass';"

abcdash, 2024-03-01:
white screen problem after upgrade from 1.36.xx to 1.37.55 (with add-apt-repository ppa:iconnor/zoneminder-master)
My solution:
install (allow) PHP intl library (try to find in phpinfo() page first)

sudo apt install php-intl
sudo phpenmod intl
service apache2 restart

connortechnology, 2024-03-02:
Closing this issue as there is no further action to take other than improving wiki/documentation content.

closed 2024-03-02 by connortechnology