ZoneMinder/zoneminder #3428: ZM 1.36 locks up and throws kernal errors - zmaudit.pl: iget: checksum invalid. ZM 1.36 on Ubuntu 18.04 VM.

Jsalas424, 2022-02-11:
**Describe Your Environment**
ZM 1.36 installed via PPA
Ubuntu 18.04.6 LTS

**Describe the bug**

**To Reproduce**

**Expected behavior**

**Debug Logs**
```
2022-02-11T04:20:12-05:00 zonemindergpu kernel - - -  [24773.395961] EXT4-fs error (device sdb1): ext4_lookup:1590: inode #5679841: comm zmaudit.pl: iget: checksum invalid
...
2022-02-11T04:51:31-05:00 zonemindergpu kernel - - -  [26652.159322] CPU: 1 PID: 82 Comm: kswapd0 Tainted: P           O     4.15.0-167-generic #175-Ubuntu
...
2022-02-11T04:51:31-05:00 zonemindergpu kernel - - -  [26652.158900] BUG: unable to handle kernel paging request at 0000000000002008
2022-02-11T04:51:31-05:00 zonemindergpu kernel - - -  [26652.160015] RIP: dentry_unlink_inode+0x43/0xe0 RSP: ffffb44382013bf0

```
 So i ran fsck but didn't find anything interesting:
```
jon@zonemindergpu:~$ sudo fsck -f /dev/sdb1
fsck from util-linux 2.31.1
e2fsck 1.44.1 (24-Mar-2018)
Pass 1: Checking inodes, blocks, and sizes
Pass 2: Checking directory structure
Pass 3: Checking directory connectivity
Pass 4: Checking reference counts
Pass 5: Checking group summary information
/dev/sdb1: 3398986/13107200 files (0.0% non-contiguous), 44220381/52428539 blocks
```
Any ideas what's going on here? Note: /dev/sdb1 holds all the videos and events for ZM, but this is independent of the OS disk . This is a VM, and I have no qualms with nuking the storage area disk if need be.

connortechnology, 2022-02-11:
So first off: There is no way that ZM should be able to cause kernel errors or fs corruption.  
second: What kind of VM?  I have seen issues with virtualbox doing block device errors if Host IO is a bit slow.
third: zmaudit.pl should not generally be running. it is too CPU/RAM/IO intensive.  Should only be run manually to clean up after a crash. Turn it off.

Jsalas424, 2022-02-11:
Thanks for the quick reply @connortechnology ! 

This is a QEMU VM hosted on Proxmox. It is configured with a UEFI BIOS and has a graphics card passed through. See attached pic:

[image]

I'll turn off zmaudit.pl, thanks for that tip.

closed 2022-04-05 by connortechnology