ZoneMinder/zoneminder #3429: Browser streaming of short events is broken, zms crashes with signal 6

mddepew, 2022-02-12:
**Describe Your Environment**
- Version of ZoneMinder _v1.37.11_
- How you installed ZoneMinder _Ubuntu PPA_
- Full name and version of OS _Ubuntu 20.04 LTS_
- Browser name and version _[Version 1.35.101 Chromium: 98.0.4758.87 (Official Build) (64-bit)](https://brave.com/latest/)_

**If the issue concerns a camera**
- Make and Model _Reolink E1 Pro 4MP_
- frame rate _15fps_
- resolution _640x360_
- ZoneMinder Source Type: _Remote (RTSP/Unicast)_

**Describe the bug**
Short events (<10s) recorded at 640x360 15fps will not stream in the browser when viewing events. The frames are present and video export does work. A javascript error is reports that it cannot connect to the zms socket. The ZMS logs indicate that the zms process crashed with signal 6.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a Remote monitor with a Reolink E1 Pro 4MP using the 640x360 RTSP stream
2. Use modect mode
3. Trigger an event with a duration of less than 10 seconds
4. Attempt to view the event in the browser

**Expected behavior**
Short events (<10s at 640x360 15fps) stream, as longer events currently do.

**Debug Logs**
```
from browser:
/run/zm/zms-485233s.sock does not exist.  This file is created by zms, and since it does not exist, either zms did not run, or zms exited early.  Please check your zms logs and ensure that CGI is enabled in apache and check that the PATH_ZMS is set correctly.  Make sure that ZM is actually recording.  If you are trying to view a live stream and the capture process (zmc) is not running then zms will exit. Please go to http://zoneminder.readthedocs.io/en/latest/faq.html#why-can-t-i-see-streamed-images-when-i-can-see-stills-in-the-zone-window-etc for more information.

from zms_e898.log
02/11/22 20:24:49.241448 zms_e898[53440].DB3-zm_eventstream.cpp/273 [Frame 119 timestamp (1644605329.680000 s), offset (9.680000 s), delta(0.000000 s), in_db(1)]
02/11/22 20:24:49.241451 zms_e898[53440].DB3-zm_eventstream.cpp/273 [Frame 120 timestamp (1644605329.780000 s), offset (9.780000 s), delta(0.100000 s), in_db(1)]
02/11/22 20:24:49.241457 zms_e898[53440].DB3-zm_eventstream.cpp/273 [Frame 121 timestamp (1644605329.880000 s), offset (9.880000 s), delta(0.100000 s), in_db(1)]
02/11/22 20:24:49.241462 zms_e898[53440].DB3-zm_eventstream.cpp/273 [Frame 122 timestamp (1644605329.980000 s), offset (9.980000 s), delta(0.100000 s), in_db(1)]
02/11/22 20:24:49.241466 zms_e898[53440].DB3-zm_eventstream.cpp/273 [Frame 123 timestamp (1644605330.090000 s), offset (10.090000 s), delta(0.110000 s), in_db(1)]
02/11/22 20:24:49.241502 zms_e898[53440].ERR-zm_signal.cpp/50 [Got signal 6 (Aborted), crashing]

```

connortechnology, 2022-02-15:
Logs don't contain enough info.

Also, remote monitor type is known to have issues.  Change to ffmpeg.

mddepew, 2022-02-15:
What other logs would be helpful to provide?

I tried using ffmpeg originally, but had a lot of issues with smearing. Switching to remote resolved these issues.
Segments recorded with ffmpeg also caused a signal 6 in the zms log.

connortechnology, 2022-09-21:
The smearing with ffmpeg is caused by using UDP. You can fix it by either specifying reorder_queue_size in Options or switching to TCP.  

Many changes have been made since February, are you still affected?  

We may need debug level 5 logs