ZoneMinder/zoneminder #3432: Audit Events Report filter SQL-ERR

jjakob, 2022-02-14:
**Describe Your Environment**
zoneminder 1.36.12-focal1
ppa.launchpad.net/iconnor/zoneminder-1.36
Ubuntu 20.04.3 LTS
mariadb-server 1:10.3.32-0ubuntu0.20.04.1

**Describe the bug**
Can't filter in Audit Events Report, get a SQL-ERR, even if I leave the filter untouched

**To Reproduce**
1. Click on "Audit Events Report" in top menu
2. Click on Events number for a camera
3. Click on "Filter" icon (4th from the left)
4. Click on "List Matches"

**Expected behavior**
Should show a list of matched events

**Debug Logs**
```
14/02/22 11:52:09.437942 web_php[43835].ERR [1.2.3.4] [SQL-ERR 'SQLSTATE[HY000]: General error: 4078 Illegal parameter data types int and row for operation '='', statement was 'SELECT E.*, M.Name AS Monitor FROM `Events` AS E INNER JOIN Monitors AS M ON E.MonitorId = M.Id WHERE ( ( E.StartDateTime >= '2022-02-14 00:47:41'
 and E.StartDateTime <= '2022-02-14 11:47:41' )
 and E.MonitorId = ('1','3')
) ORDER BY E.StartDateTime DESC' params:] at /usr/share/zoneminder/www/includes/database.php line 161
14/02/22 11:52:09.441864 web_php[43835].ERR [1.2.3.4] [SQL-ERR '00000

', statement was 'SELECT E.*, M.Name AS Monitor FROM `Events` AS E INNER JOIN Monitors AS M ON E.MonitorId = M.Id WHERE ( ( E.StartDateTime >= '2022-02-14 00:47:41'
 and E.StartDateTime <= '2022-02-14 11:47:41' )
 and E.MonitorId = ('1','3')
) ORDER BY E.StartDateTime DESC'] at /usr/share/zoneminder/www/includes/database.php line 119
14/02/22 11:52:09.446205 web_php[43835].ERR [1.2.3.4] [SQL-ERR '00000

', statement was 'SELECT E.*, M.Name AS Monitor FROM `Events` AS E INNER JOIN Monitors AS M ON E.MonitorId = M.Id WHERE ( ( E.StartDateTime >= '2022-02-14 00:47:41'
 and E.StartDateTime <= '2022-02-14 11:47:41' )
 and E.MonitorId = ('1','3')
) ORDER BY E.StartDateTime DESC'] at /usr/share/zoneminder/www/includes/functions.php line 1902
```

jjakob, 2022-02-14:
if I change `Monitor Id equal to 1` (instead of `1,3`) it works, so there is a syntax error in the monitor id selection

lthrhx, 2022-02-14:
To who ever can update this code, the SQL operator should be "IN" or "in", not "equal" or "=". 

Example: 
`and E.MonitorId IN ('1','3')`

The rest of the syntax is correct.

jjakob, 2022-02-15:
I don't think that's correct as I found a simpler way to correct it is by changing the condition to `in set` and the filter will work without any changes to the code. Of course, it's only temporary. So I think there is a different issue with the code that selects "is equal" instead of "in set" as the default condition for Monitor Id.

jjakob, 2022-02-15:
When in the events view, clicking the filterBtn sends us to the filter view with this query term (url-decoded):
`&filter[Query][terms][2][attr]=MonitorId&filter[Query][terms][2][op]=IN&filter[Query][terms][2][val]=1,3&filter[Query][sort_asc]=0`
but the filter view displays this as "equals", not "in set", because it doesn't know `IN` from opTypes, only `=[]`.
If I add opTypes elements for  `IN` and `NOT IN` this fixes the issue, when clicking filterBtn the correct filter operator is selected in the filter view, and after that everything else works correctly at least after my quick testing.
The diff is:
```
--- a/web/skins/classic/views/filter.php  2022-02-15 14:07:55.327267321 +0100
+++ b/web/skins/classic/views/filter.php    2022-02-15 14:05:35.805753393 +0100
@@ -130,6 +130,8 @@
     '!~'  => translate('OpNotMatches'),
     '=[]' => translate('OpIn'),
     '![]' => translate('OpNotIn'),
+    'IN'  => translate('OpIn'),
+    'NOT IN' => translate('OpNotIn'),
     'IS'  => translate('OpIs'),
     'IS NOT'  => translate('OpIsNot'),
     'LIKE' => translate('OpLike'),
```

lthrhx, 2022-02-15:
Excellent! Thank you.

commit 59039de9 (2022-02-15): Use =[] instead of IN as operator. Fixes #3432
closed 2022-02-15 by connortechnology