ZoneMinder/zoneminder #3436: ssmtp msmtp sending emails form ZM daemon.

ptroms, 2022-02-15:
I have got a small problem regarding sending emails form ZM via ssmtp, or msmtp so far. According to the documentation and pretty straightforward guide all emails are send by command via ssmtp, or msmtp passed in flying colours, as a root, www-data user and so on. Syslog and mail-log do indicate the right action. When it comes to receiving emails form ZM daemon - no luck. The only indication of some action taken by ZM daemon is below:

```
15/02/22 18:10:47.219807 zmeventnotification[39671].INF [main:1048] [|----> FORK:F (1), eid:245508 use hooks/start hook not being used, going to directly send out a notification if checks pass]
15/02/22 18:11:15.229064 zmeventnotification[39671].INF [main:1048] [|----> FORK:F (1), eid:245508 Event 208 for Monitor 1 has finished]
15/02/22 18:11:17.232707 zmeventnotification[39671].INF [main:1048] [|----> FORK:F (1), eid:245508 end hooks/use hooks not being used, going to directly send out a notification if checks pass]
```

Syslog, and mail.log - no record at all. ZM daemon (Options -> Email) is configured as guided. I tried msmtp and ssmtp as well. Is a way around it?

baudneo, 2022-02-16:
Those logs are for ZMES the event server add-on. What version of ZM are you running and how do you have things set up?

ptroms, 2022-02-16:
That s right. It is the only place in logs where mail is mention. 
system: 
`5.10.0-11-amd64`
zoneminder:
`v1.36.12`

baudneo, 2022-02-16:
The 'notificatiom' that zmeventnotofication is talking about is push messages via FCM a completely different thing. I don't use the mail utility in zm, there must be some logs for it somewhere though. You could try grepping the zm log directory for 'mail' or your email username or something.

ptroms, 2022-02-16:
I am afraid i couldn't find anything related to email.  Alarm section form log below:
```
02/16/22 00:59:03.978279 zmc_m1[43990].INF-zm_monitor.cpp/2076 [F: 19729 - Gone into alarm state PreAlarmCount: 2 > AlarmFrameCount:3 Cause:Forced Web]
02/16/22 08:29:54.480404 zmc_m1[43990].INF-zm_videostore.cpp/514 [some options not used, turn on debugging for a list.]
02/16/22 00:59:04.420829 zmc_m1[43990].INF-zm_monitor.cpp/2123 [F: 19728 - Opening new event 221, alarm start]
02/16/22 00:59:07.156219 zmc_m1[43951].INF-zm_monitor.cpp/1674 [F: 19800 - Capturing at 22.17 fps, capturing bandwidth 143067bytes/sec]
02/16/22 00:59:07.188146 zmc_m1[43990].INF-zm_monitor.cpp/1715 [F: 19800 - Analysing at 22.13 fps from 19799 - 19699=100 / 1644973147.188143 - 1644973142.668811 = 4.519332]
02/16/22 00:59:11.251843 zmc_m1[43990].INF-zm_monitor.cpp/2157 [F: 19888 - Gone into alert state]
02/16/22 00:59:11.718536 zmc_m1[43951].INF-zm_monitor.cpp/1674 [F: 19900 - Capturing at 21.92 fps, capturing bandwidth 135202bytes/sec]
02/16/22 00:59:11.751270 zmc_m1[43990].INF-zm_monitor.cpp/1715 [F: 19900 - Analysing at 21.91 fps from 19899 - 19799=100 / 1644973151.751267 - 1644973147.188143 = 4.563124]
02/16/22 00:59:16.283858 zmc_m1[43951].INF-zm_monitor.cpp/1674 [F: 20000 - Capturing at 21.90 fps, capturing bandwidth 135029bytes/sec]
02/16/22 00:59:16.317840 zmc_m1[43990].INF-zm_monitor.cpp/1715 [F: 20000 - Analysing at 21.90 fps from 19999 - 19899=100 / 1644973156.317837 - 1644973151.751267 = 4.566570]
02/16/22 00:59:18.379835 zmc_m1[43990].INF-zm_monitor.cpp/2165 [F: 20045 - Left alarm state (221) - 370(162) images]
02/16/22 00:59:18.379907 zmc_m1[43990].INF-zm_monitor.cpp/2170 [F: 20045 - Closing event 221, alarm end]
```

baudneo, 2022-02-16:
Ill setup the email functionality quickly and see if I can help out

baudneo, 2022-02-16:
In the ZM filters I do not see an option to 'Send email', from what I understand that's what is needed in order to receive an email. A ZM filter needs to match and the command to send emails needs to be set up.

baudneo, 2022-02-16:
Is there a reason you want to email instead of zmninja/pushover/gotify notifications using the Event Server? You can not run object detection and instead be notified on every single event on specified monitors.

ptroms, 2022-02-16:
If is hard to disagree with you regarding accessing ZM form mobile. The only one disadvantage of zmNinja is consuming data. It is a problem when I am far away form civilization. Email would be more convenient in that scenario. 
Regarding debug in syslog produce tons of data but I am afraid beside does ones that I was sending/testing manually no signs of mails from ZM.

baudneo, 2022-02-16:
The reason ZM is not sending emails from what I can tell is that that functionality was controlled by using 'filters'. I am running 1.37.11 and I do not see a filter command to send emails so I am assuming that functionality was removed or deprecated, I cant say for sure though. Here is a screenshot of the filters section.
[image]

There is an option to run a command there though that you could set up with a custom shell script to send an email, I just don't know how you would extract event info. The other option is to use Event Server push API option and create a script that sends an email instead of using a push notification service, you would get the event #, monitor name and the 'notes' to send in an email.

connortechnology, 2022-02-16:
You have to turn it on Options->Email OPT_EMAIL

baudneo, 2022-02-16:
Ah, I hadnt turned all that on yet. @ptroms - here is a screenshot of the filters page one OPT_EMAIL is enabled.
[image]

You just need to setup a filter to match for emails to be sent.

ptroms, 2022-02-17:
Let me show what I have here:
Email conf:
[image]
Filters conf:
[image]
Syslog debug mode shows no sings of mail. When I entering the filter configuration over http the only thing that comes up by command:
`tail -n 30000 /var/log/syslog | grep mail`
is:
```
Feb 17 09:09:24 zm web_php[74417]:             [AutoEmail] => 0
Feb 17 09:09:24 zm web_php[74417]:             [EmailTo] => 
Feb 17 09:09:24 zm web_php[74417]:             [EmailSubject] => 
Feb 17 09:09:24 zm web_php[74417]:             [EmailBody] => 
Feb 17 09:09:24 zm web_php[74417]:             [AutoEmail] => 0
Feb 17 09:09:24 zm web_php[74417]:             [EmailTo] => 
Feb 17 09:09:24 zm web_php[74417]:             [EmailSubject] => 
Feb 17 09:09:24 zm web_php[74417]:             [EmailBody] => 
Feb 17 09:09:40 zm web_php[77921]:             [AutoEmail] => 0
Feb 17 09:09:40 zm web_php[77921]:             [EmailTo] => 
Feb 17 09:09:40 zm web_php[77921]:             [EmailSubject] => 
Feb 17 09:09:40 zm web_php[77921]:             [EmailBody] => 
Feb 17 09:09:40 zm web_php[77921]:     [AutoEmail] => 1
Feb 17 09:09:40 zm web_php[77921]:     [EmailTo] => 
Feb 17 09:09:40 zm web_php[77921]:     [EmailSubject] => 
Feb 17 09:09:40 zm web_php[77921]:     [EmailBody] => 
Feb 17 09:09:40 zm web_php[77921]:             [AutoEmail] => 0
Feb 17 09:09:40 zm web_php[77921]:             [EmailTo] => 
Feb 17 09:09:40 zm web_php[77921]:             [EmailSubject] => 
Feb 17 09:09:40 zm web_php[77921]:             [EmailBody] => 
Feb 17 09:09:40 zm web_php[77921]:     [AutoEmail] => 1
Feb 17 09:09:40 zm web_php[77921]:     [EmailTo] => 
Feb 17 09:09:40 zm web_php[77921]:     [EmailSubject] => 
Feb 17 09:09:40 zm web_php[77921]:     [EmailBody] => 
```

baudneo, 2022-02-17:
Set the filter to Id equal to <monitor ID> and then go and force an event on that monitor. I think live view has force alalrm button dosabled but you can use zmu from the command line -> `sudo zmu -m[MONITOR ID] -a -U [USERNAME] -P [PASSWORD]` and to cancel the alarm use  `sudo zmu -m[MONITOR ID] -ac -U [USERNAME] -P [PASSWORD]` Or try executing the filter from the filter page.

I don't use filters so I cant really help on that part, I am assuming having 'StartDate' 'equals to' and then the editable field set to nothing is invalid or may not trigger. IDK for sure though.

ptroms, 2022-02-17:
Right.
I have sliced and diced baudneo's suggestion's commands, but as he mentioned it triggers on-off the alarms.
I am sure there is a simple catch.

jjakob, 2022-02-17:
These are my settings, but I'm probably going to switch to zmeventnotification to get notifications in real time ratner than delayed 1 minute in the worst case. It may be possible to change "end date/time" to "start date/time" to receive the event even sooner, "end date/time" means it won't send the mail until the event has already finished.
[image]
[image]

jjakob, 2022-02-17:
You also need to set "run filter in background" and "run filter concurrently".

connortechnology, 2022-02-17:
So those End Time clauses are pretty superfluous. Also you likely want Cause CONTAINS, not equal to.

connortechnology, 2022-02-17:
Oh and DON'T use the concurrent option.  It was ill-conceived and not needed.

jjakob, 2022-02-17:
> So those End Time clauses are pretty superfluous. Also you likely want Cause CONTAINS, not equal to.

Yep, you can omit the last 2 "End Time" clauses, I have them because I want to ignore events during work hours.
Good to know about Cause contains, thanks.

ptroms, 2022-02-17:
Great point. The filter works for sure, however no alert emails (so far),

ptroms, 2022-02-19:
Finally works.
As usual it was a tiny bit. In options "email to", "email subject" and "email body" needs to be filled
[image]
.
And then "voila" here we are. 
Also I have a two additional questions regarding this functionality. 
1. Is there any change to email does alarms in two email addresses? I tried duplicate the filter with a different email address but the system randomly send emails only to one of them.
2. I tried to create another filter that get current photograph of all of the cameras and email it as a daily report. Here what I have, but it is not working.
[image]
The classic question would be. How to do it?

baudneo, 2022-02-19:
I would use an email filtering rule to automatically forward a copy of the email to the 2nd address. If you are using outlook or Gmail you need to login to your account on desktop and find where the email filtering/forwarding rules are.

ptroms, 2022-02-19:
There is many, many ways of doing it. As baudneo pointed out I think is one of the most popular one. 
Good stuff.

jjakob, 2022-02-25:
1. You could maybe specify multiple e-mail addresses in the standard e-mail syntax by separating them with commas. The `<>` are not mandatory, they are only if you want to include a human-readable name in front of the e-mail address (e.g. `"John Doe" <johndoe@foo.bar>`).
2. Your filter is probably not working because of "and Start Time equal to 08:50:00" which will only select events with a start time of exactly 8:50. Each event has a start and end time. There is no "Current Time" which you'd need to schedule a filter to run only at a specific time. I don't know how you'd do that, maybe it's possible, but I can't see it. If so, it would be nice if a "Current Time" (also Date/Time) were added to enable this.

P.S.: If you also want to filter out events from a particular time range, you probably want a range like I did in my example above, e.g. `and (Start Time greater than 8:50)` for 8:50-23:59; or `and (Start Time greater than 8:50 and Start Time less than 20:00)` for 8:50-20:00. My example above has an "or" between the two conditions because I want to include midnight, which means I need 2 conditions for one time range, because 00:00 is less than 23:59, so it needs a separate condition.

SteveGilvarry, 2026-06-12:
Triaging the issue backlog. This was a filter/email configuration question and the reporter solved it in-thread by completing the email settings. Closing.