ZoneMinder/zoneminder #3444: Error when deleting more than 200 events

DemoFreak, 2022-02-22:
**Describe Your Environment**
- Version 1.36.12
- installed by RPM package
- Opensuse 15.3

**Describe the bug**
If "ALL" is selected in the event list, the events are not deleted as long as there are more than 200 events in the list.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to an event list with more than 200 entries
2. Select "ALL" in the bottom drop-down box
3. Select all events
4. Press the delete button and confirm the security prompt
5. No event will be deleted

**Expected behavior**
All events will be deleted

connortechnology, 2022-03-30:
I havn;t been able to reproduce this. I may have already fixed it.. I am about to release 1.36.13.  I suggest we evaluate that and see if it still needs fixing.

closed 2022-06-06 by connortechnology

connortechnology, 2022-06-06:
Will be included in 1.36.20

commit 5068cf5e (2022-06-06): Add a progress ticker, and delete events in chunks of 10. Fixes #3444
commit 27057157 (2022-06-07): Add a progress ticker, and delete events in chunks of 10. Fixes #3444