ZoneMinder/zoneminder #3445: Build error with ffmpeg 5.0

blackPantherOS, 2022-02-23:
- Version of ZoneMinder [release version, development version, or commit]
ZM 1.37.11
- How you installed ZoneMinder [e.g. PPA, RPMFusion, from-source, etc]
latest git
- Full name and version of OS
blackPanther OS

**Describe the bug**
build error with ffmeg 5.0
gcc 11.2.1
cmake 3.22.0

```
zoneminder-1.37.11/src/zm_ffmpeg_camera.cpp: In member function ‘int FfmpegCamera::OpenFfmpeg()’:
zoneminder-1.37.11/src/zm_ffmpeg_camera.cpp:358:52: error: invalid conversion from ‘const AVCodec*’ to ‘AVCodec*’ [-fpermissive]
  358 |     if ((mVideoCodec = avcodec_find_decoder_by_name("h264_mmal")) == nullptr) {
      |                        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~
      |                                                    |
      |                                                    const AVCodec*
zoneminder-1.37.11/src/zm_ffmpeg_camera.cpp:366:39: error: invalid conversion from ‘const AVCodec*’ to ‘AVCodec*’ [-fpermissive]
  366 |     mVideoCodec = avcodec_find_decoder(mVideoStream->codecpar->codec_id);
      |                   ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                       |
      |                                       const AVCodec*
zoneminder-1.37.11/src/zm_ffmpeg_camera.cpp:498:45: error: invalid conversion from ‘const AVCodec*’ to ‘AVCodec*’ [-fpermissive]
  498 |     if (!(mAudioCodec = avcodec_find_decoder(mAudioStream->codecpar->codec_id))) {
      |                         ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                             |
      |                                             const AVCodec*
make[2]: Leaving directory 'zoneminder-1.37.11/build'
make[2]: *** [src/CMakeFiles/zm.dir/build.make:551: src/CMakeFiles/zm.dir/zm_ffmpeg_camera.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:516: src/CMakeFiles/zm.dir/all] Error 2
make[1]: Leaving directory '/usr/src/RPM/BUILD/zoneminder-1.37.11/build'
make: *** [Makefile:139: all] Error 2

```

exuvo, 2022-03-11:
I fixed the const errors in https://github.com/exuvo/zoneminder/commit/110dcac4fd0c7399d5796e99311d88ba9670414d but i don't know how to handle AVPicture being removed in src/zm_mpeg.cpp:483 nor AVStream no longer having cur_dts.

exuvo, 2022-03-11:
Okay found the commit for cur_dts https://github.com/FFmpeg/FFmpeg/commit/591b88e6787c4e678237f02a50421d101abd25c2 apparently it is private now, not sure what we should be using instead.

exuvo, 2022-03-11:
http://ffmpeg.org/pipermail/ffmpeg-devel/2021-June/280859.html seems to be an example of what to do instead, i'll see if i can figure out how to apply it to zoneminder.

exuvo, 2022-03-11:
Well my branch compiles and seems to work but i just removed the code referencing cur_dts which i don't think is an acceptable solution.

sergiomb2, 2022-04-14:
> Well my branch compiles and seems to work but i just removed the code referencing cur_dts which i don't think is an acceptable solution.

Hi, thank you for your patch , please can you point me where is the commit that " just removed the code referencing cur_dts" . 
Best regards

exuvo, 2022-04-14:
This one https://github.com/exuvo/zoneminder/commit/c3b216d90e1c6ae9e15efdd29f05eb38c8fcde24

closed 2022-05-18 by connortechnology